esphome/feature-requests #2322: Remote transmitter as remote entity  in Home Assistant

samuelolteanu, 2023-07-23:
**Describe the problem you have/What new integration you would like**

I want esphome (ir) transmitter to automatically have a remote entity with all its services available from Home assistant, just like Broadlink Universal Remotes have. If receiver is set up, learn command services should be present as well..

**Please describe your use case for this integration and alternatives you've tried:**

I need to use SmartIR to control my air conditioner from Home Assistant with esphome transmitter but i'm having trouble using "remote_transmitter.transmit_raw" as a service. The AC is not supported by esphome ir climate or Arduino-HeatpumpIR. 

**Additional context**

rnkhq, 2023-08-21:
Why these IR things are so difficult till now? Like the tuya app it's too easy to setup a remote. want the same features in ha too. Could you figure out how to expose a remote entity from the ir blaster through esphome?

grigi, 2023-10-06:
As far as I know, those devices tend to store the learned IR/RF commands on the local device, but with ESPHome setting up local persistence is... complicated. (at least to me)
A solution would be to have a Home Assistant service that can learn IR/RF commands and then stores it on Home Assistant.

Once you have the keycode data in Home Assistant, getting a ESPHome device to send it is quite simple.

So, it's not that its HARD per se, just someone has to do the legwork.

felipecrs, 2024-11-06:
Another good reason for a Remote Entity:

https://community.home-assistant.io/t/create-remote-template-entity/757425

felipecrs, 2024-11-06:
In my opinion this is the only thing missing before ESPHome with IR or RF modules are a better 
 (and much cheaper) replacement to the Broadlink ones.