esphome/feature-requests #2324: BinarySensor needs a way of reporting invalid or undefined state.

clydebarrow, 2023-07-24:
**Describe the problem you have/What new integration you would like**

A `BinarySensor` needs a way of telling the front end that its value is now unknown. This is possible for `Sensor` by publishing `NaN` but currently only `true` or `false` can be published to a binary sensor.

**Please describe your use case for this integration and alternatives you've tried:**

Reporting devices need a way of telling the front end that they are currently unable to determine the state of a given sensor.  This could be e.g. a result of a loss of communication with a remote source connected via an unreliable channel such as a UART. 

Otherwise the last reported state persists, which could be misleading.

The least disturbance to the current API might be to add a new function to `binary_sensor::BinarySensor` to publish an undefined state.

**Additional context**

nagyrobi, 2025-05-29:
@bdraco

bdraco, 2025-05-29:
We already have `missing_state` in the proto def https://github.com/esphome/aioesphomeapi/blob/62e55daed57cd0ed31835f5009e13e45ddbdbcc9/aioesphomeapi/api.proto#L293

Are you looking for something different?

nagyrobi, 2025-05-29:
Is there a way to set this, eg. from a lambda?

clydebarrow, 2025-05-30:
> We already have `missing_state` in the proto def [esphome/aioesphomeapi@`62e55da`/aioesphomeapi/api.proto#L293](https://github.com/esphome/aioesphomeapi/blob/62e55daed57cd0ed31835f5009e13e45ddbdbcc9/aioesphomeapi/api.proto#L293)

So that api flag gets set if the binary sensor `has_state()` method returns false; but that method returns a protected property which is initialised to false, set to true on first publish, and never reset, nor is there any way of resetting it.

So there should be a new method in `binary_sensor` like `invalidate()` that resets the `has_state_` flag. And an action callable from yaml to invoke it.

closed 2025-06-11 by clydebarrow