esphome/feature-requests #2327: Integrate developments in libretiny fork of esphome that allow using esphome on BK7231 and RTL8710B chips

abelbour, 2023-07-25:
**Describe the problem you have/What new integration you would like**
Integrate developments in libretiny fork of esphome that allow using esphome on BK7231 and RTL8710B chips

**Please describe your use case for this integration and alternatives you've tried:**
Newer smart bulbs by Tuya use these MCUs

**Additional context**

https://github.com/kuba2k2/libretiny
https://github.com/kuba2k2/libretiny-esphome

gb0101010101, 2023-07-26:
+1 please merge LibreTiny.
So many devices are switching to the Beken chips it would be great to support them in the main ESPHome development.

RickDickinson, 2023-08-24:
Yes. Please merge this into the main release

paravoid, 2023-09-30:
I believe 2023.09 fully integrated LibreTiny and this can be resolved :) I flashed a BK7231T the other day using nothing but stock ESPHome.

closed 2023-09-30 by nagyrobi