esphome/feature-requests #2331: battery support

iron-exx, 2023-07-28:
Can you add the battery states to your script? The most important one for me would be 'statement_of_charge_soc' so that I can always monitor the battery level. Currently, I'm using the Growatt cloud for that, but it synchronizes very irregularly. The rest is already working great with ESPHome! Being able to see the battery status would complete the whole setup.

ssieb, 2023-07-28:
You need to include some more context.  What are you referring to?

iron-exx, 2023-07-28:
I have a Home Assistant, and I use the Growatt cloud integration, which also shows me the battery charge level. However, since the cloud integration synchronizes very irregularly, I connected your script "https://esphome.io/components/sensor/growatt_solar.html" to an ESP8266 Wemos D1 Mini with an RS485 converter to my Growatt inverter, and it works great. However, I only receive data from the inverter, and not from the battery. Is it possible to extend the script to also output the battery charge level, so that I can display it in Home Assistant?

ssieb, 2023-07-28:
That's certainly possible, but will need a bit of work on that component to add it.

iron-exx, 2023-07-28:
What could I do to help? I think this would be interesting for many people, as most PV systems are now built with storage.

ssieb, 2023-07-28:
Unless you want to write the code and make a PR for the component, there's not really anything you can do.

iron-exx, 2023-07-28:
Unfortunately, that exceeds my abilities -.-