esphome/feature-requests #2332: Support for M5Stack AtomS3

wildekek, 2023-07-29:
**Describe the problem you have/What new integration you would like**

I would like the [m5stack-atoms3](https://docs.platformio.org/en/latest/boards/espressif32/m5stack-atoms3.html))be supported as a board, so I can easily create a config for it in ESPHome.

**Please describe your use case for this integration and alternatives you've tried:**

I have tried using esp32-s3-devkitc-1, but I'm running into warnings and issues that I think would be moot if we can support the proper board config.

**Additional context**

From [platform-espressif32 6.1.0](https://github.com/platformio/platform-espressif32/releases/) there is support for [M5Stack AtomS3](https://docs.platformio.org/en/latest/boards/espressif32/m5stack-atoms3.html) and we'll get support for many other boards too if ESPHome is updated to this version.

randybb, 2023-07-29:
We are on platformio 5.3.0, so nope. Maybe next year once we move to 6.... Anyway, how it will help you that you can select that board? There is no special pin mapping, board templates are basically the same https://github.com/platformio/platform-espressif32/blob/master/boards/esp32-s3-devkitc-1.json <> https://github.com/platformio/platform-espressif32/blob/master/boards/m5stack-atoms3.json , so there is no benefit of selecting "the right" board.
I have that AtomS5 (well, the other with a display) on my desk and surprisingly I am running it with esp32-s3-devkitc-1 without any issue.

wildekek, 2023-07-29:
The issues I experienced were unrelated and were gone after clearing the build cache. However! When the docs say ‘select a board from the following list’ and they don’t work, that’s a waste of time. 
Either they should link to the actual list, or better, a shim list could be in place with compatible boards.

wildekek, 2023-07-30:
Here's my config for other people that have issues:
https://github.com/wildekek/ESPHome-M5Stack-Atom-S3-Lite/blob/main/M5Stack-AtomS3-Lite.yaml

Anonym-tsk, 2023-08-21:
Working config:
```yaml
esphome:
  name: atom-s3
  friendly_name: atom-s3
  name_add_mac_suffix: false

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: esp-idf
    version: 4.4.5
    platform_version: 5.4.0
```

Another working config with `m5stack-atoms3` board and newest libraries, but RMT LED doesn't work:
```yaml
esphome:
  name: atom-s3
  friendly_name: atom-s3
  name_add_mac_suffix: false

esp32:
  board: m5stack-atoms3
  variant: esp32s3
  framework:
    type: esp-idf
    version: 5.0.2
    platform_version: 6.3.2
```

johntdyer, 2023-08-22:
Tried this, didnt work for me...

```
Linking .pioenvs/atom-s3/firmware.elf
RAM:   [=         ]   9.6% (used 31420 bytes from 327680 bytes)
Flash: [====      ]  42.9% (used 786577 bytes from 1835008 bytes)
Building .pioenvs/atom-s3/firmware.bin
Creating esp32s3 image...
Successfully created esp32s3 image.
esp32_create_combined_bin([".pioenvs/atom-s3/firmware.bin"], [".pioenvs/atom-s3/firmware.elf"])
Wrote 0xd0200 bytes to file /Users/jdyer/.esphome/build/atom-s3/.pioenvs/atom-s3/firmware-factory.bin, ready to flash to offset 0x0
======================================================================================================== [SUCCESS] Took 58.90 seconds ========================================================================================================
INFO Successfully compiled program.
Found multiple options for uploading, please choose one:
  [1] /dev/cu.usbmodem1101 (USB JTAG/serial debug unit)
  [2] Over The Air (atom-s3.local)
(number): 1
esptool.py v4.6.2
Serial port /dev/cu.usbmodem1101
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:9e:2e:94
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 8MB
Flash will be erased from 0x00010000 to 0x000d0fff...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00009000 to 0x0000afff...
Compressed 786944 bytes to 517484...
Wrote 786944 bytes (517484 compressed) at 0x00010000 in 5.4 seconds (effective 1174.8 kbit/s)...
Hash of data verified.
Compressed 20528 bytes to 13054...
Wrote 20528 bytes (13054 compressed) at 0x00000000 in 0.4 seconds (effective 465.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 134...
Wrote 3072 bytes (134 compressed) at 0x00008000 in 0.1 seconds (effective 390.2 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Wrote 8192 bytes (31 compressed) at 0x00009000 in 0.1 seconds (effective 668.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/cu.usbmodem1101 with baud rate 115200
[13:35:41]I (139) esp_image: segment 1: paddr=00035610 vaddr=3fc95d00 size=03da4h ( 15780) load
[13:35:41]I (143) esp_image: segment 2: paddr=000393bc vaddr=40374000 size=06c5ch ( 27740) load
[13:35:41]I (151) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=8516ch (545132) map
[13:35:41]I (251) esp_image: segment 4: paddr=000c5194 vaddr=4037ac5c size=0b040h ( 45120) load
[13:35:41]I (269) boot: Loaded app from partition at offset 0x10000
[13:35:41]I (297) boot: Set actual ota_seq=1 in otadata[0]
[13:35:41]I (297) boot: Disabling RNG early entropy source...
[13:35:41]I (308) cpu_start: Pro cpu up.
[13:35:41]I (308) cpu_start: Starting app cpu, entry point is 0x40375bfc
[13:35:41]I (0) cpu_start: App cpu up.
[13:35:41]I (322) cpu_start: Pro cpu start user code
[13:35:41]I (322) cpu_start: cpu freq: 160000000 Hz
[13:35:41]I (323) cpu_start: Application information:
[13:35:41]I (325) cpu_start: Project name:     atom-s3
[13:35:41]I (330) cpu_start: App version:      2023.8.2
[13:35:41]I (335) cpu_start: Compile time:     Aug 22 2023 13:34:48
[13:35:41]I (342) cpu_start: ELF file SHA256:  15d9f173b8c4ecc1...
[13:35:41]I (348) cpu_start: ESP-IDF:          5.0.2
[13:35:41]I (352) cpu_start: Min chip rev:     v0.0
[13:35:41]I (357) cpu_start: Max chip rev:     v0.99
[13:35:41]I (362) cpu_start: Chip rev:         v0.1
[13:35:41]I (367) heap_init: Initializing. RAM available for dynamic allocation:
[13:35:41]I (374) heap_init: At 3FC9D7C0 len 0004BF50 (303 KiB): DRAM
[13:35:41]I (380) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
[13:35:41]I (387) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
[13:35:41]I (393) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
[13:35:41]I (400) spi_flash: detected chip: gd
[13:35:41]I (404) spi_flash: flash io: dio
[13:35:41]I (408) cpu_start: Starting scheduler on PRO CPU.
[13:35:41]I (0) cpu_start: Starting scheduler on APP CPU.
[13:35:41]E (174) uart: uart_param_config(726): Invalid src_clk

```

and here is my config

```
esphome:
  name: atom-s3
  friendly_name: atom-s3
  name_add_mac_suffix: false

esp32:
  board: m5stack-atoms3
  variant: esp32s3
  framework:
    type: esp-idf
    version: 5.0.2
    platform_version: 6.3.2

# Enable logging
logger:
api:
ota:

wifi:
  ssid: "xxxxxx.xxxxxx"
  password: "xxxxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: " Fallback Hotspot"
    password: "doA2mjPnoVLi"

captive_portal:

```

wildekek, 2023-08-22:
People, no need to post broken configs, this one works fine:

> Here's my config for other people that have issues: https://github.com/wildekek/ESPHome-M5Stack-Atom-S3-Lite/blob/main/M5Stack-AtomS3-Lite.yaml

randybb, 2023-08-22:
There is nothing special on this config - just the default one for s3 and settings for the peripherals/components. But you can make lazy people happy (who don't want to read documentation) and post it to https://devices.esphome.io/

Btw, a board can be selected by unchecking "Use recommended settings" in the wizard, but as I have mentioned - this board is not on the list and in many cases there is no need to select a board - the default one is enough (because platformio needs that), only selecting the right mcu (device) type matters.

morhimi, 2023-09-10:
Just got this one working on the [AtomS3](https://docs.m5stack.com/en/core/AtomS3) (with the lcd), running on the `esp-idf` framework
```
substitutions:
  name: "atoms3"
  device_name: "AtomS3"

esphome:
  name: atoms3
  friendly_name: AtomS3
  on_boot:
    - priority: 800
      then:
        - lambda: |-
            id(disp).enable();
            id(disp).transfer_byte(0x11);
            id(disp).disable();

esp32:
  board: m5stack-atoms3
  variant: esp32s3
  framework:
    type: esp-idf
    version: 5.1.1
    platform_version: 6.4.0

wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
 
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${device_name}
    password: !secret wifi_fallback_password
    ap_timeout: 15s
 
captive_portal:
 
# Enable logging
logger:
 level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "USE_YOUR_OWN"

ota:
  password: "USE_YOUR_OWN"

esp32_ble_tracker:

bluetooth_proxy:
  active: true

#G4=IR
# remote_transmitter:
#   pin: GPIO4
#   carrier_duty_percent: 50%

i2c:
  - id: bus_a
    sda: GPIO38
    scl: GPIO39

spi:
  clk_pin: GPIO17
  mosi_pin: GPIO21

color:
  - id: my_red
    red: 100%
    green: 0%
    blue: 0%
  - id: my_yellow
    red: 100%
    green: 100%
    blue: 0%
  - id: my_green
    red: 0%
    green: 100%
    blue: 0%
  - id: my_blue
    red: 0%
    green: 0%
    blue: 100%
  - id: my_gray
    red: 50%
    green: 50%
    blue: 50%

font:
  - file: "gfonts://Roboto"
    id: roboto_32
    size: 32
  - file: "gfonts://Roboto"
    id: roboto_24
    size: 24
  - file: "gfonts://Roboto"
    id: roboto_12
    size: 12

time:
  - platform: homeassistant
    id: esptime

display:
  - platform: st7789v # https://esphome.io/components/display/st7789v.html
    id: disp
    model: Custom
    backlight_pin: GPIO16
    cs_pin: GPIO15
    dc_pin: GPIO33
    reset_pin: GPIO34
    height: 128
    width: 128
    offset_height: 2
    offset_width: 1
    eightbitcolor: true
    update_interval: 3s
    lambda: |-
      it.rectangle(0,  0, it.get_width(), it.get_height(), id(my_blue));
      it.rectangle(0, 20, it.get_width(), it.get_height(), id(my_blue));   // header bar

      it.strftime((128 / 2), (128 / 3) * 1 + 5, id(roboto_24), id(my_gray), TextAlign::CENTER, "%Y-%m-%d", id(esptime).now());
      it.strftime((128 / 2), (128 / 3) * 2 + 5, id(roboto_32), id(my_gray), TextAlign::CENTER, "%H:%M:%S", id(esptime).now());
      it.print(5, 5, id(roboto_12), id(my_yellow), TextAlign::TOP_LEFT, "ESPHome");

      // Comment out the above lines to see the image without text overlaid
      // it.image(0, 0, id(my_image));

      if (id(system_status).state) {
        it.print(115, 5, id(roboto_12), id(my_green), TextAlign::TOP_RIGHT, "Online");
      }
      else {
        it.print(115, 5, id(roboto_12), id(my_red), TextAlign::TOP_RIGHT, "Offline");
      }
    

#G41=Button
binary_sensor:
  - platform: status
    name: "Node Status"
    id: system_status
  - platform: gpio
    name: Button
    pin:
      number: GPIO41
      inverted: true
      mode:
        input: true
        pullup: true
    filters:
      - delayed_off: 10ms
    on_press:
      then:
        - logger.log: Button Pressed
```

popstas, 2023-10-24:
Thank you, @morhimi , your config works well: display, button and sensors!

One question: how to get accel_x value in lambda? I can see values in Home assistant, I can get instance of class [esphome::mpu6886::MPU6886Component](https://esphome.io/api/classesphome_1_1mpu6886_1_1_m_p_u6886_component.html), but I have no idea how to get value.

vtolstov, 2024-01-12:
so does anybody have working m5stack atom s3 light config that works with bluetooth proxy?
my config based on arduino not worked with proxy (but m5stack booted)
```
#https://docs.m5stack.com/en/core/AtomS3%20Lite
substitutions:
  device_name: "esphome-ceaa8c"
  device_friendly_name: "esphome-ceaa8c"
  device_description: "esphome-ceaa8c"
  time_timezone: "Europe/Moscow"
       
esphome:
  name: ${device_name}
  friendly_name: $device_friendly_name
  comment: "${device_description}"
  name_add_mac_suffix: false

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: arduino
    #version: 4.4.5
    platform_version: 5.4.0
    
#esp32:
#  board: m5stack-atoms3
  #board: esp32-s3-devkitc-1
#  variant: esp32s3
#  framework:
#    type: esp-idf
#    platform_version: 6.1.0
#    version: 5.0.1
    #platform_version: 5.4.0
#    sdkconfig_options:
#      COMPILER_OPTIMIZATION_SIZE: y
#      CONFIG_COMPILER_DISABLE_GCC12_WARNINGS: y
#      CONFIG_COMPILER_DISABLE_GCC13_WARNINGS: y
#      CONFIG_RMT_SUPPRESS_DEPRECATE_WARN: y
      

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

spi:
  clk_pin: GPIO17
  mosi_pin: GPIO21

bluetooth_proxy:
  active: true

wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
  fast_connect: false

captive_portal:

logger:
  baud_rate: 115200
  level: DEBUG

api:
  encryption:
    key: !secret hass_key
  reboot_timeout: 2min
 
ota:
   password: !secret ota_password 

# Output that blinks the LED
output:
  - platform: template
    id: status_output
    type: binary
    write_action:
    - if:
        condition:
          lambda: return (state);
        then:
          - light.turn_on:
              id: led
              brightness: 70%
              red: 100%
              green: 0%
              blue: 0%
        else:
          - light.turn_off: led

#G4=IR
remote_transmitter:
  pin: GPIO4
  carrier_duty_percent: 50%

light:
  # Enable status LED
  - platform: status_led
    output: status_output
    name: "Status Led"
    internal: True

  #G35=RGB WS2812C-2020
  - platform: esp32_rmt_led_strip
    rgb_order: GRB
    pin: 35
    num_leds: 4
    # RMT 0 channels will be occupied by IR
    rmt_channel: 2
    chipset: ws2812
    id: led
    name: "Led"

#G41=Button
binary_sensor:
  - platform: gpio
    name: Button
    pin:
      number: GPIO41
      inverted: true
      mode:
        input: true
        pullup: true
    filters:
      - delayed_off: 10ms
    on_press:
      then:
        - logger.log: Button Pressed
```

with esp-idf it cont boot or failes to configure firmware as have errors

morhimi, 2024-01-12:
> so does anybody have working m5stack atom s3 light config that works with bluetooth proxy?
> my config based on arduino not worked with proxy (but m5stack booted)

@vtolstov this one works for me on M5Stack Atom S3 Lite

```
#https://docs.m5stack.com/en/core/AtomS3%20Lite
substitutions:
  device_name: "atoms3-lite-1"
  device_friendly_name: "AtomS3 Lite"
  device_description: "M5Stack AtomS3 Lite"
  time_timezone: "Asia/Jerusalem"
       
esphome:
  name: ${device_name}
  friendly_name: $device_friendly_name
  comment: "${device_description}"
  name_add_mac_suffix: false

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino
    version: 2.0.14
    platform_version: 6.5.0
  variant: esp32s3

wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
 
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${device_name}
    password: !secret wifi_fallback_password
    ap_timeout: 15s
 
captive_portal:
 
# Enable logging
logger:
  baud_rate: 0
  level: VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: "yourkey"

ota:
  password: "yourpwd"

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:
  active: true

# Output that blinks the LED
output:
  - platform: template
    id: status_output
    type: binary
    write_action:
    - if:
        condition:
          lambda: return (state);
        then:
          - light.turn_on:
              id: led
              brightness: 70%
              red: 100%
              green: 0%
              blue: 0%
        else:
          - light.turn_off: led

#G4=IR
remote_transmitter:
  pin: GPIO4
  carrier_duty_percent: 50%

light:
  # Enable status LED
  - platform: status_led
    output: status_output
    name: "Status Led"
    internal: True

  #G35=RGB WS2812C-2020
  - platform: esp32_rmt_led_strip
    rgb_order: GRB
    pin: 35
    num_leds: 4
    # RMT 0 channels will be occupied by IR
    rmt_channel: 1
    chipset: ws2812
    id: led
    name: "Led"

#G41=Button
binary_sensor:
  - platform: gpio
    name: Button
    pin:
      number: GPIO41
      inverted: true
      mode:
        input: true
        pullup: true
    filters:
      - delayed_off: 10ms
    on_press:
      then:
        - logger.log: Button Pressed
```

vtolstov, 2024-01-12:
@morhimi thanks! it booted and connected!

rlust, 2024-02-02:
Any ideas on how to use the grove connected 4 relay board?
Shopping cart
Product image	Description	Quantity	Price
4-Relay Unit - m5stack-store
1
4-Relay Unit
1	$12.95

rlust, 2024-02-02:
> Just got this one working on the [AtomS3](https://docs.m5stack.com/en/core/AtomS3) (with the lcd), running on the `esp-idf` framework. It's a mix of the AtomS3 Lite from @wildekek with some modifications
> 
> ```
> substitutions:
>   name: "atoms3"
>   device_name: "AtomS3"
> 
> esphome:
>   name: atoms3
>   friendly_name: AtomS3
>   on_boot:
>     - priority: 800
>       then:
>         - lambda: |-
>             id(disp).enable();
>             id(disp).transfer_byte(0x11);
>             id(disp).disable();
> 
> esp32:
>   board: m5stack-atoms3
>   variant: esp32s3
>   framework:
>     type: esp-idf
>     version: 5.1.1
>     platform_version: 6.4.0
> 
> wifi:
>   networks:
>     - ssid: !secret wifi_ssid
>       password: !secret wifi_password
>  
>   # Enable fallback hotspot (captive portal) in case wifi connection fails
>   ap:
>     ssid: ${device_name}
>     password: !secret wifi_fallback_password
>     ap_timeout: 15s
>  
> captive_portal:
>  
> # Enable logging
> logger:
>  level: DEBUG
> 
> # Enable Home Assistant API
> api:
>   encryption:
>     key: "USE_YOUR_OWN"
> 
> ota:
>   password: "USE_YOUR_OWN"
> 
> 
> esp32_ble_tracker:
> 
> bluetooth_proxy:
>   active: true
> 
> 
> #G4=IR
> # remote_transmitter:
> #   pin: GPIO4
> #   carrier_duty_percent: 50%
> 
> 
> i2c:
>   - id: bus_a
>     sda: GPIO38
>     scl: GPIO39
> 
> sensor:
>   - platform: mpu6886
>     address: 0x68
>     accel_x:
>       name: "$device_name MPU6886 Accel X"
>     accel_y:
>       name: "$device_name MPU6886 Accel Y"
>     accel_z:
>       name: "$device_name MPU6886 Accel z"
>     gyro_x:
>       name: "$device_name MPU6886 Gyro X"
>     gyro_y:
>       name: "$device_name MPU6886 Gyro Y"
>     gyro_z:
>       name: "$device_name MPU6886 Gyro z"
>     temperature:
>       name: "$device_name MPU6886 Temperature"
> 
> spi:
>   clk_pin: GPIO17
>   mosi_pin: GPIO21
> 
> color:
>   - id: my_red
>     red: 100%
>     green: 0%
>     blue: 0%
>   - id: my_yellow
>     red: 100%
>     green: 100%
>     blue: 0%
>   - id: my_green
>     red: 0%
>     green: 100%
>     blue: 0%
>   - id: my_blue
>     red: 0%
>     green: 0%
>     blue: 100%
>   - id: my_gray
>     red: 50%
>     green: 50%
>     blue: 50%
> 
> font:
>   - file: "gfonts://Roboto"
>     id: roboto_32
>     size: 32
>   - file: "gfonts://Roboto"
>     id: roboto_24
>     size: 24
>   - file: "gfonts://Roboto"
>     id: roboto_12
>     size: 12
> 
> time:
>   - platform: homeassistant
>     id: esptime
> 
> display:
>   - platform: st7789v # https://esphome.io/components/display/st7789v.html
>     id: disp
>     model: Custom
>     backlight_pin: GPIO16
>     cs_pin: GPIO15
>     dc_pin: GPIO33
>     reset_pin: GPIO34
>     height: 128
>     width: 128
>     offset_height: 2
>     offset_width: 1
>     eightbitcolor: true
>     update_interval: 3s
>     lambda: |-
>       it.rectangle(0,  0, it.get_width(), it.get_height(), id(my_blue));
>       it.rectangle(0, 20, it.get_width(), it.get_height(), id(my_blue));   // header bar
> 
>       it.strftime((128 / 2), (128 / 3) * 1 + 5, id(roboto_24), id(my_gray), TextAlign::CENTER, "%Y-%m-%d", id(esptime).now());
>       it.strftime((128 / 2), (128 / 3) * 2 + 5, id(roboto_32), id(my_gray), TextAlign::CENTER, "%H:%M:%S", id(esptime).now());
>       it.print(5, 5, id(roboto_12), id(my_yellow), TextAlign::TOP_LEFT, "ESPHome");
> 
>       // Comment out the above lines to see the image without text overlaid
>       // it.image(0, 0, id(my_image));
> 
>       if (id(system_status).state) {
>         it.print(115, 5, id(roboto_12), id(my_green), TextAlign::TOP_RIGHT, "Online");
>       }
>       else {
>         it.print(115, 5, id(roboto_12), id(my_red), TextAlign::TOP_RIGHT, "Offline");
>       }
>     
> 
> #G41=Button
> binary_sensor:
>   - platform: status
>     name: "Node Status"
>     id: system_status
>   - platform: gpio
>     name: Button
>     pin:
>       number: GPIO41
>       inverted: true
>       mode:
>         input: true
>         pullup: true
>     filters:
>       - delayed_off: 10ms
>     on_press:
>       then:
>         - logger.log: Button Pressed
> ```

Great work here on the Atom S3 config! Well done just what I was looking for.
 Just need to get the 4 relay board working!

k1asa, 2024-02-03:
> Warning
> This component has been made redundant since the ST7789V is now supported by the [ILI9XXX component](https://esphome.io/components/display/ili9xxx#ili9xxx). It is recommended that you use the ili9xxx component as it will be maintained, whereas this component may not be, or may be removed completely in the future.

from https://esphome.io/components/display/st7789v

Here its work well on my AtomS3.

```
# SPI pin
spi:
  clk_pin: 17
  mosi_pin: 21

# GPIO pin of the display backlight
switch:
  - platform: gpio
    pin: 16
    name: "Backlight"
    id: backlight
    restore_mode: ALWAYS_ON

display:
  - platform: ili9xxx
    model: ST7789V
    cs_pin: 15
    dc_pin: 33
    reset_pin: 34
    rotation: 180
    invert_colors: true
    update_interval: 1s
    dimensions:
      height: 128
      width: 128
      offset_height: 1
      offset_width: 2
    lambda: |-
      auto red = Color(255, 0, 0);
      auto green = Color(0, 255, 0);
      auto blue = Color(0, 0, 255);
      auto white = Color(255, 255, 255);
      // Draw a line from [0,0] to [100,50]
      it.line(0, 0, 0, it.get_height(), red);
      it.line(0, 0, it.get_width(), 0, green);
      it.line(0, it.get_height()-1, it.get_width(), it.get_height()-1, blue);
      it.line(it.get_width()-1, 0, it.get_width()-1, it.get_height(), white);
```

optimus1509, 2024-02-06:
Can anyone help a novice with a working atom s3 lite as bluetooth proxy to just add config for the LED to light up on activity? Most of the config here seems to be for ones with the display.

morhimi, 2024-02-06:
> Can anyone help a novice with a working atom s3 lite as bluetooth proxy to just add config for the LED to light up on activity? Most of the config here seems to be for ones with the display.

See here https://github.com/esphome/feature-requests/issues/2332#issuecomment-1888347255

optimus1509, 2024-02-06:
> > Can anyone help a novice with a working atom s3 lite as bluetooth proxy to just add config for the LED to light up on activity? Most of the config here seems to be for ones with the display.
> 
> See here [#2332 (comment)](https://github.com/esphome/feature-requests/issues/2332#issuecomment-1888347255)

Does this also work with esp-idf or are there some changes required?

dbrgn, 2024-06-12:
> We are on platformio 5.3.0, so nope. Maybe next year once we move to 6....

It seems that esphome is now using platformio 6.1, so maybe official support for the Atom S3 could be added?

morhimi, 2024-10-22:
I updated full config while implementing comments from @scarolan & @k1asa 
It's now working with ESP-IDF v5.3.1

### AtomS3 yaml:
```
substitutions:
  name: "atoms3"
  device_name: "AtomS3"

esphome:
  name: atoms3
  friendly_name: AtomS3
  platformio_options:
    build_flags:
      - "-w" # https://docs.platformio.org/en/latest/projectconf/sections/env/options/build/build_flags.html

esp32:
  board: m5stack-atoms3
  variant: esp32s3
  framework:
    type: esp-idf
    version: 5.3.1
    platform_version: 6.9.0
    # Custom sdkconfig options
    sdkconfig_options: # https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-compiler-disable-gcc12-warnings
      CONFIG_SPIFFS_DBG: n

wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
  enable_btm: true
  enable_rrm: true
  reboot_timeout: 7min
  id: wifi_id

 
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${device_name}
    password: !secret wifi_fallback_password
    ap_timeout: 15s
 
captive_portal:
 
# Enable logging
logger:
#  baud_rate: 0
 level: VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: "YourKey"

ota:
  platform: esphome
  password: "YourPassowrd"

esp32_ble_tracker:

bluetooth_proxy:
  active: true

i2c:
  - id: bus_a
    sda: GPIO38
    scl: GPIO39

sensor:
  - platform: mpu6886
    address: 0x68
    accel_x:
      name: "$device_name MPU6886 Accel X"
    accel_y:
      name: "$device_name MPU6886 Accel Y"
    accel_z:
      name: "$device_name MPU6886 Accel z"
    gyro_x:
      name: "$device_name MPU6886 Gyro X"
    gyro_y:
      name: "$device_name MPU6886 Gyro Y"
    gyro_z:
      name: "$device_name MPU6886 Gyro z"
    temperature:
      name: "$device_name MPU6886 Temperature"
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 120s

spi:
  clk_pin: GPIO17
  mosi_pin: GPIO21

color:
  - id: my_red
    red: 100%
    green: 0%
    blue: 0%
  - id: my_yellow
    red: 100%
    green: 100%
    blue: 0%
  - id: my_green
    red: 0%
    green: 100%
    blue: 0%
  - id: my_blue
    red: 0%
    green: 0%
    blue: 100%
  - id: my_gray
    red: 50%
    green: 50%
    blue: 50%

font:
  - file: "gfonts://Roboto"
    id: roboto_32
    size: 32
  - file: "gfonts://Roboto"
    id: roboto_24
    size: 24
  - file: "gfonts://Roboto"
    id: roboto_12
    size: 12

time:
  - platform: homeassistant
    id: esptime

# GPIO pin of the display backlight
switch:
  - platform: gpio
    pin: 16
    name: "Backlight"
    id: backlight
    restore_mode: ALWAYS_ON

display:
  - platform: ili9xxx
    model: ST7789V
    id: disp
    cs_pin: 15
    dc_pin: 33
    reset_pin: 34
    rotation: 180
    invert_colors: true
    update_interval: 1s
    dimensions:
      height: 128
      width: 128
      offset_height: 1
      offset_width: 2
    lambda: |-
      it.rectangle(0,  0, it.get_width(), it.get_height(), id(my_blue));
      it.rectangle(0, 20, it.get_width(), it.get_height(), id(my_blue));   // header bar

      it.strftime((128 / 2), (128 / 3) * 1 + 5, id(roboto_24), id(my_gray), TextAlign::CENTER, "%Y-%m-%d", id(esptime).now());
      it.strftime((128 / 2), (128 / 3) * 2 + 5, id(roboto_32), id(my_gray), TextAlign::CENTER, "%H:%M:%S", id(esptime).now());
      it.print(5, 5, id(roboto_12), id(my_yellow), TextAlign::TOP_LEFT, "ESPHome");

      // Comment out the above lines to see the image without text overlaid
      // it.image(0, 0, id(my_image));

      // if (id(system_status).state) {.   // depends in binary sensor :( 
      if (id(wifi_id).is_connected()) {
        it.print(115, 5, id(roboto_12), id(my_green), TextAlign::TOP_RIGHT, "Online");
      }
      else {
        it.print(115, 5, id(roboto_12), id(my_red), TextAlign::TOP_RIGHT, "Offline");
      }

#G41=Button
binary_sensor:
  - platform: status
    name: "Node Status"
    id: system_status
  - platform: gpio
    name: Button
    pin:
      number: GPIO41
      inverted: true
      mode:
        input: true
        pullup: true
    filters:
      - delayed_off: 10ms
    on_press:
      then:
        - logger.log: Button Pressed

text_sensor:
  - platform: wifi_info
    ip_address:
      name: ESP IP Address
    ssid:
      name: ESP Connected SSID
    bssid:
      name: ESP Connected BSSID
    mac_address:
      name: ESP Mac Wifi Address
    scan_results:
      name: ESP Latest Scan Results
    dns_address:
      name: ESP DNS Address

```

RuudNetideas, 2024-11-14:
Anyone got the config working for the AtomS3R (with TFT)? PIN info is available on M5Docs but no luck until now. Example config below has been taken from last poster, morhimi but modified for AtomS3R.

```
substitutions:
  ipnum: 192.168.2.94
  device_name: "AtomS3R"

esphome:
  name: "94e-atom-s3r"
  friendly_name: AtomS3
  platformio_options:
    build_flags:
      - "-w" # https://docs.platformio.org/en/latest/projectconf/sections/env/options/build/build_flags.html
  
esp32:
  board: m5stack-atoms3
  variant: esp32s3
  framework:
    type: esp-idf
    version: 5.3.1
    platform_version: 6.9.0
    # Custom sdkconfig options
    sdkconfig_options: # https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-compiler-disable-gcc12-warnings
      CONFIG_SPIFFS_DBG: n

# Include WIFI details
<<: !include details/wifi.yaml
 
# Enable logging
logger:
 level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "[Your Value]"

ota:
  - platform: esphome
    password: "[Your Value]"

esp32_ble_tracker:

bluetooth_proxy:
  active: true

spi:
  clk_pin: GPIO15 #AtomS3: GPIO17
  mosi_pin: GPIO21 #AtomS3: GPIO21

color:
  - id: my_yellow
    red: 100%
    green: 100%
    blue: 0%
  
font:
  - file: "gfonts://Roboto"
    id: roboto_24
    size: 24
  
time:
  - platform: homeassistant
    id: esptime

# GPIO pin of the display backlight

display:
  - platform: st7789v
    model: CUSTOM
    id: disp
    cs_pin: GPIO14 #AtomS3: GPIO15
    dc_pin: GPIO42 #AtomS3: GPIO33
    reset_pin: GPIO48 #AtomS3: GPIO34
    rotation: 180
    update_interval: 1s
    height: 128
    width: 128
    offset_height: 1
    offset_width: 2
    lambda: |-
      it.rectangle(0,  0, it.get_width(), it.get_height(), id(my_yellow));      
      it.strftime((128 / 2), (128 / 3) * 1 + 5, id(roboto_24), id(my_yellow), TextAlign::CENTER, "%Y-%m-%d", id(esptime).now());
      it.strftime((128 / 2), (128 / 3) * 2 + 5, id(roboto_24), id(my_yellow), TextAlign::CENTER, "%H:%M:%S", id(esptime).now());

binary_sensor:
  - platform: gpio
    name: Button
    pin:
      number: GPIO41
      inverted: true
      mode:
        input: true
        pullup: true
    filters:
      - delayed_off: 10ms
    on_press:
      then:
        - logger.log: Button Pressed

text_sensor:
  - platform: wifi_info
    ip_address:
      name: ESP IP Address
```

thanasis00, 2025-03-11:
@RuudNetideas I received my AtomS3R and I have been fiddling with it to make the display work, and managed it today. This is my config:

```
substitutions:
  name: "atoms3r"
  device_name: "AtomS3R"

esphome:
  name: ${name}
  friendly_name: ${device_name}
  platformio_options:
    build_flags:
      - "-w" # https://docs.platformio.org/en/latest/projectconf/sections/env/options/build/build_flags.html

esp32:
  board: m5stack-atoms3
  variant: esp32s3
  framework:
    type: esp-idf
    version: 5.3.1
    platform_version: 6.9.0
    # Custom sdkconfig options
    sdkconfig_options: # https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-compiler-disable-gcc12-warnings
      CONFIG_SPIFFS_DBG: n

wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
  enable_btm: true
  enable_rrm: true
  reboot_timeout: 7min
  id: wifi_id
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: MANUAL_IP    # Set this to the IP address of the router. Often ends with .1
    gateway: MANUAL_GATEWAY
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0

 
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${device_name}
    password: !secret wifi_fallback_password
    ap_timeout: 15s
 
captive_portal:
 
# Enable logging
logger:
#  baud_rate: 0
 level: VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_encryption_key

ota:
  platform: esphome
  password: !secret ota_password

esp32_ble_tracker:

bluetooth_proxy:
  active: true

i2c:
  - id: bus_a
    sda: GPIO38
    scl: GPIO39

sensor:
  - platform: mpu6886
    address: 0x68
    accel_x:
      name: "$device_name MPU6886 Accel X"
    accel_y:
      name: "$device_name MPU6886 Accel Y"
    accel_z:
      name: "$device_name MPU6886 Accel z"
    gyro_x:
      name: "$device_name MPU6886 Gyro X"
    gyro_y:
      name: "$device_name MPU6886 Gyro Y"
    gyro_z:
      name: "$device_name MPU6886 Gyro z"
    temperature:
      name: "$device_name MPU6886 Temperature"
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 120s

spi:
  clk_pin: GPIO15
  mosi_pin: GPIO21

color:
  - id: my_red
    red: 100%
    green: 0%
    blue: 0%
  - id: my_yellow
    red: 100%
    green: 100%
    blue: 0%
  - id: my_green
    red: 0%
    green: 100%
    blue: 0%
  - id: my_blue
    red: 0%
    green: 0%
    blue: 100%
  - id: my_gray
    red: 50%
    green: 50%
    blue: 50%

font:
  - file: "gfonts://Roboto"
    id: roboto_32
    size: 32
  - file: "gfonts://Roboto"
    id: roboto_24
    size: 24
  - file: "gfonts://Roboto"
    id: roboto_12
    size: 12

time:
  - platform: homeassistant
    id: esptime

# GPIO pin of the display backlight
switch:
  - platform: gpio
    pin: 16
    name: "Backlight"
    id: backlight
    restore_mode: ALWAYS_ON

display:
  - platform: ili9xxx
    model: ST7789V
    id: disp
    cs_pin: GPIO14
    dc_pin: GPIO42
    reset_pin: GPIO48
    rotation: 180
    invert_colors: true
    update_interval: 1s
    dimensions:
      height: 128
      width: 128
      offset_height: 1
      offset_width: 2
    lambda: |-
      it.rectangle(0,  0, it.get_width(), it.get_height(), id(my_blue));
      it.rectangle(0, 20, it.get_width(), it.get_height(), id(my_blue));   // header bar

      it.strftime((128 / 2), (128 / 3) * 1 + 5, id(roboto_24), id(my_gray), TextAlign::CENTER, "%Y-%m-%d", id(esptime).now());
      it.strftime((128 / 2), (128 / 3) * 2 + 5, id(roboto_32), id(my_gray), TextAlign::CENTER, "%H:%M:%S", id(esptime).now());
      it.print(5, 5, id(roboto_12), id(my_yellow), TextAlign::TOP_LEFT, "ESPHome");

      // Comment out the above lines to see the image without text overlaid
      // it.image(0, 0, id(my_image));

      // if (id(system_status).state) {.   // depends in binary sensor :( 
      if (id(wifi_id).is_connected()) {
        it.print(115, 5, id(roboto_12), id(my_green), TextAlign::TOP_RIGHT, "Online");
      }
      else {
        it.print(115, 5, id(roboto_12), id(my_red), TextAlign::TOP_RIGHT, "Offline");
      }

#G41=Button
binary_sensor:
  - platform: status
    name: "Node Status"
    id: system_status
  - platform: gpio
    name: Button
    pin:
      number: GPIO41
      inverted: true
      mode:
        input: true
        pullup: true
    filters:
      - delayed_off: 10ms
    on_press:
      then:
        - logger.log: Button Pressed

text_sensor:
  - platform: wifi_info
    ip_address:
      name: ESP IP Address
    ssid:
      name: ESP Connected SSID
    bssid:
      name: ESP Connected BSSID
    mac_address:
      name: ESP Mac Wifi Address
    scan_results:
      name: ESP Latest Scan Results
    dns_address:
      name: ESP DNS Address
```

simenopsahl, 2025-03-16:
@thanasis00 Did you mean Atom S3? That config does not work on the S3R. It has an i2c backlight (LP5562) and a different accelerometer sensor (BMI270 & BMM150).

This works for the S3R, still missing the accelerometer/gyro

```
external_components:
  - source:
      type: git
      url: https://github.com/ssieb/esphome
      ref: lp5562
    components: [ lp5562 ]

# https://www.reddit.com/r/Esphome/comments/1geelgf/what_would_it_take_to_port_led_driver_lp5562/
lp5562:
  # SCL GPIO0 & SDA GPIO45
  - i2c_id: i2c_internal
 
output:
  - platform: lp5562
    id: backlight_red
    channel: 0
  - platform: lp5562
    id: backlight_green
    channel: 1
  - platform: lp5562
    id: backlight_blue
    channel: 2
  - platform: lp5562
    id: backlight_white
    channel: 3

light:
  - platform: monochromatic
    id: backlight_light
    output: backlight_white
    default_transition_length: 100ms
    restore_mode: ALWAYS_ON
    name: "Backlight"
    entity_category: config
```

RuudNetideas, 2025-03-18:
@simenopsahl: Thanks for your example code.

I finally got the thing working. I 'missed' the backlight settings. 
I enclose my working yaml code for future use.

`substitutions:
  ipnum: [DEVICE_IP]
  name: "atoms3r"
  device_name: "AtomS3R"

external_components:
  - source:
      type: git
      url: https://github.com/ssieb/esphome
      ref: lp5562
    components: [ lp5562 ]

esphome:
  comment: "M5Stack Atom S3R"
  name: ${name}
  friendly_name: ${device_name}
  platformio_options:
    board_build.arduino.memory_type: qio_opi
    build_flags:
      - "-w" # https://docs.platformio.org/en/latest/projectconf/sections/env/options/build/build_flags.html
      - "-DBOARD_HAS_PSRAM"
      - "-DARDUINO_USB_CDC_ON_BOOT=1"
      - "-mfix-esp32-psram-cache-issue"

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf
    sdkconfig_options: # https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-compiler-disable-gcc12-warnings
      CONFIG_SPIFFS_DBG: n

psram:
  mode: octal
  speed: 80MHz

i2c:
  - id: i2c_internal
    sda: GPIO45
    scl: GPIO0

# https://www.reddit.com/r/Esphome/comments/1geelgf/what_would_it_take_to_port_led_driver_lp5562/
lp5562:
  # SCL GPIO0 & SDA GPIO45
  - i2c_id: i2c_internal
 
output:
  - platform: lp5562
    id: backlight_red
    channel: 0
  - platform: lp5562
    id: backlight_green
    channel: 1
  - platform: lp5562
    id: backlight_blue
    channel: 2
  - platform: lp5562
    id: backlight_white
    channel: 3

# Include WIFI details
<<: !include details/wifi.yaml
 
# Enable logging
logger:
#  baud_rate: 0
 level: DEBUG

api:
  encryption:
    key: "[YOUR KEY]"

ota:
  - platform: esphome
    password: "[YOUR PASSWORD]"

esp32_ble_tracker:

bluetooth_proxy:
  active: true

sensor:
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 120s

light:
  # White
  - platform: monochromatic
    id: backlight_light
    output: backlight_white
    default_transition_length: 100ms
    restore_mode: ALWAYS_ON
    name: "Backlight"
    entity_category: config

spi:
  clk_pin: GPIO15
  mosi_pin: GPIO21

color:
  - id: my_red
    red: 100%
    green: 0%
    blue: 0%
  - id: my_yellow
    red: 100%
    green: 100%
    blue: 0%
  - id: my_green
    red: 0%
    green: 100%
    blue: 0%
  - id: my_blue
    red: 0%
    green: 0%
    blue: 100%
  - id: my_gray
    red: 50%
    green: 50%
    blue: 50%

font:
  - file: "gfonts://Roboto"
    id: roboto_24
    size: 24

time:
  - platform: homeassistant
    id: esptime

display:
  - platform: ili9xxx
    model: st7789v
    id: display_ips
    cs_pin: GPIO14
    dc_pin: GPIO42
    reset_pin: GPIO48
    invert_colors: true
    update_interval: 1s
    #show_test_card: true
    dimensions:
      height: 128
      width: 128
      offset_height: 1
      offset_width: 2
    rotation: 270
    lambda: |-
      it.rectangle(0,  0, it.get_width(), it.get_height(), id(my_yellow));      
      it.strftime((128 / 2), (128 / 3) * 1 + 5, id(roboto_24), id(my_red), TextAlign::CENTER, "%Y-%m-%d", id(esptime).now());
      it.strftime((128 / 2), (128 / 3) * 2 + 5, id(roboto_24), id(my_green), TextAlign::CENTER, "%H:%M:%S", id(esptime).now())

#G41=Button
binary_sensor:
  - platform: gpio
    name: Button
    pin:
      number: GPIO41
      inverted: true
      mode:
        input: true
        pullup: true
    filters:
      - delayed_off: 10ms
    on_press:
      then:
        - logger.log: Button Pressed

text_sensor:
  - platform: wifi_info
    ip_address:
      name: ESP IP Address
    ssid:
      name: ESP Connected SSID
`