esphome/feature-requests #2335: Add identify option to status_led

svh1985, 2023-08-01:
**Describe the problem you have/What new integration you would like**

I would like to request an option which is common in HPE/DELL Servers to identify a harddrive. It's called "identify" and it basically very quickly blinks the Status LED for a minute or two. This way it's easy to identify the physical device from the UI.

**Please describe your use case for this integration and alternatives you've tried:**

This is handy to visually find the ESP when having a lot of ESPHome devices and it provides a way to identify a device without the need to switch on/off the connected device.

**Additional context**

nielsnl68, 2023-08-01:
I belief you can do that very easily your self by adding a Binary Sensor that light the status led and allow it to blink.

ChristophCaina, 2023-11-24:
here's an example on how you could expose a button to HomeAssistant, that triggers an effect to an LED

```
- platform: template
    name: "Identify"
    on_press:
      then:
        - light.turn_on:
            id: status_led
            effect: Identify
        - delay: 10s
        - light.turn_off: status_led
 ```
 
 when pressed, it will start the effect on the led for 10 seconds and then switch it off.