esphome/feature-requests #2336: Output Switch - Trigger in case command received, but state not changed

Masterz69, 2023-08-02:
**Describe the problem you have/What new integration you would like**

Have implemented inching/pulse by using script - see example below.
Looking how to actually restart script (thus prolong delay) while switch is ON and another command ON sent from HA.

When executing ```switch.turn_on``` service on HA see in log, f.e.:
```[23:04:29][D][switch:012]: 'Garland Hall' Turning ON.```

But trigger ```on_turn_on``` not fired as state NOT changed, but still ON.

```yaml
switch:
  - platform: output
    output: out_relay_1 
    id: sw_relay_1
    on_turn_on:
      - script.execute: delay_relay_1
    on_turn_off:
      - script.stop: delay_relay_1

script:
  - id: delay_relay_${id}
    mode: restart
    then:
      - delay: 5
      - switch.turn_off: sw_relay_1
```

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

ssieb, 2023-08-03:
This isn't a feature request, it's a support request and github issues are not the place for that.  Come ask on the esphome discord server or one of the forums.

closed 2023-08-03 by ssieb

Masterz69, 2023-08-03:
IMHO looks as feature - fire trigger ```on_turn_on``` as soon event ```Turning ON``` is logged.

ssieb, 2023-08-03:
If that's the feature request, then it's extremely unlikely to happen...  The state isn't changing, so the trigger should not be fired.

Masterz69, 2023-08-03:
What about triggers like ```on_action_on``` & ```on_action_off``` (f.e.) to be notified/triggered when actions ON/OFF received.
IMHO it will quite useful in mine described use case.

ssieb, 2023-08-03:
You have a very unusual use case that would be better solved in other ways.