esphome/feature-requests #2338: Triggers on component actions

Masterz69, 2023-08-03:
**Describe the problem you have/What new integration you would like**

Idea is to have triggers on component registered actions.
Which will fire on command/action received by component.

F.e. for ```switch``` see 3 registered actions:
```
@automation.register_action("switch.toggle", ToggleAction, SWITCH_ACTION_SCHEMA)
@automation.register_action("switch.turn_off", TurnOffAction, SWITCH_ACTION_SCHEMA)
@automation.register_action("switch.turn_on", TurnOnAction, SWITCH_ACTION_SCHEMA)
```
So, idea is to have triggers (naming just example) ```action_turn_on```, ```action_turn_off```, ```action_toggle```.

**Please describe your use case for this integration and alternatives you've tried:**

Use case.
Entering to room light is turned on by wall switch, where inching is coded to switch off light after some time (save the nature).
Or by door which send action ```turn_on``` to wall switch through HA or ```http_request``` directly to switch.

Also motion/presence sensor is installed to detect presence and prolong/restart inching time for light.
Which also send action ```turn_on``` while person still moves in room - by HA or ```http_request``` directly.

Currently action ```Turning ON``` is logged by ```switch```, but as state not changed - ```on_turn_on``` not firing. 
As per design, thanx for explanation to @ssieb.

```action_turn_on``` will fire on received command/action through HA API or by direct ```http_request``` from another ESPHome device - regardless on actual state & can be used to restart script with delay. 
Which is not possible to achieve by ```on_turn_on``` trigger.

```yaml
switch:
  - platform: output
    output: out_relay
    id: sw_relay
    action_turn_on: 
      - script.execute: delay_relay
    on_turn_off:
      - script.stop: delay_relay

script:
  - id: delay_relay
    mode: restart
    then:
      - delay: 5
      - switch.turn_off: sw_relay
```

IMHO it will add a lot of flexibility and usability.

**Additional context**

nielsnl68, 2023-08-03:
Hi, @Masterz69, I'm not sure of i understand on which components you want this new triggers.
Looking into the `switch` source code, i already see the triggers you shown above.

Masterz69, 2023-08-03:
> I'm not sure of i understand on which components

First of all - ```switch```, probably mostly used.

Current triggers ```on_turn_%``` fires only when state is changing.
If ```switch``` state is ```ON``` and arrives command/action ```turn_on```:
* action ```Turning ON``` appears in log, BUT trigger ```on_turn_on``` NOT firing - as no state change happens.

Idea is to have triggers which fires on command/action, regardless of state change.

Looks I failed to explain use case.
When multiple ```turn_on``` will be received from HA or other devices while state is already ```ON``` and would be nice to react on received actions.

nielsnl68, 2023-08-03:
Thanks now your request makes more sense to me.

I like your idea.