esphome/feature-requests #2341: introduce `set_invert(bool)` method for `esphome::ssd1306_base::SSD1306`

jostsalathe, 2023-08-03:
**Describe the problem you have/What new integration you would like**

I would love to use the invert functionality of my SSD1306 display from the rendering lambda.

**Please describe your use case for this integration and alternatives you've tried:**

My SSD1306 display is attached to an ESP8266 and BME280 driven room climate sensor.
I want to use inverted pixels on the SSD1306 to indicate a warning when the rel. humidity exceeds a given threshold.

**Additional context**

Implementing this should be relatively easy, if I am not mistaken, by having that new `set_invert(bool)` method directly call `this->command(SSD1306_COMMAND_NORMAL_DISPLAY | invert);` like in the setup method. It would probably be sensible to also do `this->invert_ = invert;` to keep track of state, I guess.

If I find the time to make myself familiar with the process of pushing stuff upstream to here, I'd be happy to just implement it myself and post a merge request. But as of now I have no specific experience with developing and testing new code for ESPHome.

Generally awesome project, by the way! <3

max246, 2023-08-21:
At the moment we have `init_invert` which is configured on the setup of the component but you would like to have it as an action to be triggered ?

jostsalathe, 2023-08-21:
I think so. I would expect `id.set_invert` to be accessible from the display rendering lambda the same way `id.set_contrast` is.

Here is a snippet from my target configuration:
```yaml
display:
  - platform: ssd1306_i2c
    model: SSD1306_128X32
    id: ssd
    update_interval: never
    rotation: ${display_rotation}
    lambda: |-
      it.print(4, 0, id(rob10), "°C temp");
      it.printf(4, 9, id(rob24), "%4.1f", id(temp).state);
      it.print(96, 0, id(rob10), "\% hum");
      it.printf(76, 9, id(rob24), "%5.1f", id(hum).state);
      // dim display at night
      ((esphome::ssd1306_base::SSD1306 &) it).set_contrast(id(sun_elev).state / 100.0);
      if (id(hum).state >= ${warn_threshold}) {
        it.print(48, 0, id(rob12), "!WARN!");
        //((esphome::ssd1306_base::SSD1306 &) it).set_invert(true); //waiting for https://github.com/esphome/feature-requests/issues/2341 to be resolved
      } else {
        //((esphome::ssd1306_base::SSD1306 &) it).set_invert(false); //waiting for https://github.com/esphome/feature-requests/issues/2341 to be resolved
      }
```

nagyrobi, 2023-08-21:
> At the moment we have `init_invert` which is configured on the setup of the component but you would like to have it as an action to be triggered ?

It would be useful to invert oled pixels periodically to compensate for burn-in.

max246, 2023-08-22:
@jostsalathe I made a PR which should be working, if you want to test it, just checkout my branch to create the firmware

jostsalathe, 2023-08-22:
Thanks a lot! I just tested it and it works like a charm! =D

closed 2023-08-22 by jesserockz