esphome/feature-requests #2343: Allow not immediately rebooting after OTA update

palves, 2023-08-06:
**Describe the problem you have/What new integration you would like**

Please allow updating firmware OTA, and then not automatically reboot the updated device.  I would like to control when to reboot the device.

**Please describe your use case for this integration and alternatives you've tried:**

The mains plug of the server machine where I have the ESPHome docker container running is connected to an Athom smart plug running an ESPHome firmware, so that I can remotelly power cycle the server if it crashes for some reason.  Updating the firmware of the smart plug automatically reboots it, causing the server to be turned off and on without chance of doing a proper shutdown/restart of the server.  If I were able to upload the new firmware to the plug device without causing it to reboot immediately, I could do:

- update the plug device's firmware OTA
- power down the server (remotely)
- open the plug device's web interface and reboot it (remotely)
- turn on the server again (remotely, by e.g., setting it bios to to power on on ac power, or with wake-on-lan)

I would be able to do the above all without physically touching the server or the Athom smart plug.

**Additional context**

Without this ability, I have to power down the server machine, physically remove the Athom plug so the server is connected to mains directly, power on the smart plug in a separate socket, boot server, update smart plug firmware, power down server again, put Athom smart plug back in place controlling the server, boot server.

nagyrobi, 2023-08-06:
I don't think that's possible.

But your approach to use an Athom plug with a relay constantly engaged to power a server is not a very good idea for several reasons:
- the smart plug could reboot itself by any time for various other reasons too, like WiFi loss/interference
- the relay in the smartplug is not designed to be almost constantly engaged. It dissipates heat and after quite some time these cheap relays tend to remain physically stuck in ON state, and can never be turned off.

Instead, you should simply connect an AC relay having NC (normally closed) contacts to your smartplug, and wire your server through that. With this, you invert the logic of the circuit: it keeps the server powered on with the relays being in OFF (unengaged) state. Beside not having to worry about heat and sticy relays, you can reboot/ota the ESP as many times you'd like. You can power-cycle the server when you actually turn the plug ON.

bkaufx, 2023-08-14:
Have you tried disabling early_pin_init?  It should stop the plug from toggling on reboot.

https://esphome.io/components/esp8266.html?highlight=early_pin_init

palves, 2023-08-18:
> I don't think that's possible.

I'm curious why.  Admitedly I haven't looked at the code, but I would assume that the update process is somewhat like:

#1 - new firmware is uploaded to empty flash space (does not overwrite current firmware)
#2 - device is setup to jump to new firmware next time (thus current firmware becomes empty space on next boot)
#3 - reboot

... and so I was thinking that it would be possible for the OTA process to just not do step #3 automatically (optionally).

> Instead, you should simply connect an AC relay having NC (normally closed) contacts to your smartplug, and wire your server through that. (snip)

Thanks.  Makes a lot of sense.  One other reason I have it setup like it is, is that I wanted to track power usage for a while, which the smart plug provides.  Otherwise, it was super easy to do and I needed it quickly as I was traveling very soon.  (and that's why I didn't reply earlier. :-) )

palves, 2023-08-18:
@bkaufx 

> Have you tried disabling early_pin_init? It should stop the plug from toggling on reboot.
> 
> https://esphome.io/components/esp8266.html?highlight=early_pin_init

Interesting!  I will give it a try as soon as I am able.  If it works, it'll be even better.

palves, 2024-02-26:
It took a long while, but I finally had a chance to try early_pin_init, and it works perfectly.  Thus, closing.  Thanks much for the suggestion!

closed 2024-02-26 by palves