esphome/feature-requests #2344: Refuse to arm home or away if at least one zone is faulted

hwstar, 2023-08-11:
**Describe the problem you have/What new integration you would like**

Add a configuration option to prevent the alarm panel from being armed if there is at least one sensor showing a fault.
The option would not allow an alarm away arm to happen if any of the sensors are faulted. For arm home, the option would only prevent arming for those sensors without a bypass_armed_home key set for the zone.

**Please describe your use case for this integration and alternatives you've tried:**

The way it works currently is that it is possible to arm the alarm with a faulted zone. This causes the alarm to go into pending mode, then trigger the alarm. If a user walks or drives away after doing this, the alarm will go off. The user gets confirmation that the alarm is armed, but may not be aware that a zone is faulted. 

My current workaround is to add 2 separate lists of tests as lambdas in the YAML code to test each group of arm home and arm away sensors, and provide ready/not ready binary sensors for use by homeassistant cards. This additional logic does not completely solve the problem though, as these binary sensors only provide status information, and do not prevent the user from arming the system in a faulted state.

Not having the system arm in a faulted state for arm away and arm home is the use case. It will prevent someone from arming the system, leaving the premises, and then have to return due to an alarm caused by the faulted zone.

**Additional context**

nagyrobi, 2023-08-11:
What do you mean by fault?
Triggered?

hwstar, 2023-08-11:
No. When a zone (binary sensor input) faults, it means a door is open, a PIR was tripped. In other words the zone is not in the normal or clear state.

augustodinizl, 2023-08-18:
> No. When a zone (binary sensor input) faults, it means a door is open, a PIR was tripped. In other words the zone is not in the normal or clear state.

dude, it's function for the alarm central, not of the keyboard (input device)

hwstar, 2023-08-18:
> dude, it's function for the alarm central, not of the keyboard (input device)

That's not the case with the Honeywell panel I replaced. It would not arm when there was a zone faulted. You had to bypass the faulted zones if you wanted to arm the panel. This functionality belongs in the panel, not at the central station. Besides that, the alarm control panel doesn't support central station monitoring.

nagyrobi, 2023-08-18:
@augustodinizl in ESPHome, this component implements an entire central, not just the input! Read the docs...

hwstar, 2023-08-19:
It replaces the firmware burglar alarm system panel (i.e. the circuit board which connects to all the wires sensors in your home), not a central station. Panels should offer the option to refuse to arm if there are faulted zones. (Unless you  specifically ask it to by bypassing the faulted zones) As I stated previously, the panel I replaced had this functionality, and the component written for esphome should consider implementing this functionality. 

It really doesn't matter what the input device is, it could be HomeAssistant, or it could be a wired keypad of some type (Whic I'm implementing on my system BTW). The decision to arm or disarm needs to be made in the firmware running in the panel, but it needs to be properly communicated back to the user via whatever input device is in use.

I see nothing in the docs which specifically says you can or cannot arm a system with faulted zones. It is silent on the matter. However,  IMHO allowing the arming of a system with faulted zones without at least warning the user that the zones are faulted will just lead to user frustration. 

This functionallity is missing in both ESPHome and the alarm panel in Home Assistant. Technically both need to be updated to support this, but this is an ESPHome forum.

Docs rechecked:
https://esphome.io/components/alarm_control_panel/index.html
https://esphome.io/components/alarm_control_panel/template.html

rarroyo6, 2023-08-20:
I agree. The panel should not arm unless all zones are closed or bypassed, this is standard in most commercial alarm systems.
I opened an issue #2364 to add a "Ready" or "Fault" attribute, this way we can at least show if any zones are opened before arming.

nagyrobi, 2023-08-20:
There seems to be a misunderstanding here. Please clarify what's your meaning for "central" office.

hwstar, 2023-08-20:
> There seems to be a misunderstanding here. Please clarify what's your meaning for "central" office.

In the United States, a "central station" is the alarm monitoring company responsible for monitoring an alarm system. The box with all of the "brains" where the sensors connect is called an "alarm panel".

nagyrobi, 2023-08-21:
OK.
In EU, the "central" part is the box with all of the "brains", whilst the "panel" is just the keyboard with the buttons and some visual feedback. These are all in the house, and the monitoring company is optional, the system can (and has to) work also completely offline from any third party.

A sysem can have several panels connnected to the same central brain.
That's why in this view, the zones fault should be assessed not in the "panels", but the "brains" part of the system. Panels are just like passive keyboards here.

:)

hwstar, 2023-08-21:
OK so we had some terminology which was different, and that was the source of the confusion. I think we are in agreement that the new functionality belongs in the "brain"/central/alarm panel, not the panel/keypads.

DanCech, 2024-03-31:
I just ran into exactly this issue when configuring an esphome alarm, it would be great to have some indication of whether the alarm is ready before arming it.

waynehaffenden, 2024-06-26:
+1 for this feature. @hwstar do you mind sharing the solution you use in the meantime? I appreciate it's only work around but better than driving away to have an internal door not quite closed and the siren starts going off!

mikec85, 2025-06-15:
+1 for this feature as well.