esphome/feature-requests #2345: Add support for new LilyGo T-Display-S3 AMOLED

julianrinaldi, 2023-08-11:
**Describe the problem you have/What new integration you would like**

I would like support for the new LilyGo T-Display-S3 AMOLED display added to ESPHome.

https://www.lilygo.cc/products/t-display-s3-amoled?variant=42859368677557

**Please describe your use case for this integration and alternatives you've tried:**

It's probably the nicest small screen released for ESP32, with a higher resolution (240x536), dpi, refresh rate and brighter screen. I'm sure it will be very popular for ESPHome users.

**Additional context**

Here are some links that might be useful:
https://github.com/Xinyuan-LilyGO/T-Display-S3-AMOLED
https://github.com/VolosR/TDisplayAmoled/tree/main/AmoledT-Display/HelloWorldAmoled

cosmicDustOfLightLength, 2023-09-10:
I have board file works for me very good, might be useful: 
[lilygo-t-amoled.txt](https://github.com/esphome/feature-requests/files/12568146/lilygo-t-amoled.txt)

bluechris, 2023-11-21:
> I have board file works for me very good, might be useful: [lilygo-t-amoled.txt](https://github.com/esphome/feature-requests/files/12568146/lilygo-t-amoled.txt)

I just got the board and i cannot in the life of me make the screen to light up no matter what i try. 
Your code is not ESPHome or something i dont understand correctly.

Either way, does anyone managed to make it work?

clydebarrow, 2023-12-13:
[image]

bluechris, 2023-12-16:
Dear Clydebarrow can you share the code of what you did?

clydebarrow, 2023-12-16:
There's some discussion (see link below)  that will help. I'm on the road so can only provide limited help right now, but if you clone my fork and check out the amoled branch you should be able to build from that. There is also a  PR for the touchscreen in esphome, #5941 IIRC.

The AMOLED driver is not in PR yet, needs the quad_spi PR merged first.

https://github.com/landonr/lilygo-tdisplays3-esphome/issues/25

See also the Display Features thread on Discord.

closed 2024-04-15 by clydebarrow