esphome/feature-requests #2349: Set a specific size (in pixels) for QR codes

Kernald, 2023-08-14:
**Describe the problem you have/What new integration you would like**

`Display::qr_code` takes a `scale` argument, and a position, but not a specific size. I'd like a way to scale it up or down to a specific size instead.

**Please describe your use case for this integration and alternatives you've tried:**

Using the scale argument works to an extent, but it's tricky to lay a QR code out nicely with other content - especially when the data length seems to change the size of the QR code.

**Additional context**

davini-dev, 2023-11-10:
Good

ssieb, 2023-11-11:
It's not really possible to set a specific size.  The qr code needs to have an integer number of pixels per block and the number of blocks depends on the data size and the redundancy.

nagyrobi, 2023-11-11:
In my experience, after generation, a lanczos resize (kinda dithering) to a not too far dimension is still perfectly acceptable and remains functional.

ssieb, 2023-11-11:
I would think at these display resolutions, any resizing is just going to make it blurry, unless it's big enough to have multiple pixels per block.

Kernald, 2023-11-11:
Couldn't we try to play with the redundancy and pixels per block until we find something that's close enough, maybe?

nagyrobi, 2024-04-03:
LVGL is using the same library and they found a way to set the size: 
https://github.com/lvgl/lv_lib_qrcode/blob/master/lv_qrcode.c

clydebarrow, 2025-01-21:
The `qrcode` widget in the LVGL component allows the size to be set.

closed 2025-01-21 by clydebarrow