esphome/feature-requests #2351: Dashboard: Better handling for user-editable yaml include files

hn, 2023-08-14:
**Describe the problem you have/What new integration you would like**

Improve handling for YAML include files (e.g. for packages, WiFi config, ...) within the ESPHome dashboard. These files should still be editable with the screen editor, but should be displayed somewhat stripped back (no buttons for "Update", "Logs", no "Update available" sign etc. and possibly a slightly different look overall).

Proposal: Files starting with an underscore ("_") are displayed and handled as editable include files.

Alternative or complementary proposal: Offer a switch "show/hide .dot files" within the dashboard.

**Please describe your use case for this integration and alternatives you've tried:**

I do not want to completely hide the files ([by prepending a dot or moving them into a subdir](https://esphome.io/guides/configuration-types.html#yaml-insertion-operator)) because I want to edit them from time to time with the screen editor.

**Additional context**

palves, 2023-08-18:
> Proposal: Files starting with an underscore ("_") are displayed and handled as editable include files.

And/or allow editing files in subdirectories (and such files in sub dirs would not be considered devices).  So you could put your include files in base/ or whatever subdirectory.  That's what I initially tried to do, but then quickly reverted to putting the shared files in the the top level config dir (named package-foo.yaml) so I could conveniently still edit them in the web ui.

hn, 2023-08-20:
Reconsidered the idea: it would be nice to have an ESPHome-specific list of prefixes and/or filenames that are treated as include (!=device) files:
- secrets.yaml
- package-*.yaml
- include-*.yaml
- (TBD)

These files could be made visible by a switch on the dashboard.

The advantage is that if you **don't** work with "." dot files, that they remain visible and 'non-special' in other systems (shell, git, ...), which is better here I think.

BugsBunny1403, 2024-11-14:
I agree with proposition

m1k3f15h, 2025-07-18:
Hi

Are there any news in this?
I want to create my first "Package" and i don't know how to name the file :)

Mike from Austria :)

BeatLink, 2025-07-20:
Eager for this feature as well

MatthiasNeuhaus, 2026-10-01:
## Proposal: a "Packages" view in the Device Builder dashboard — and I'd like to implement it

Picking this up again because it's still the main friction point when working with `packages:` / `!include` from the dashboard. I'd like to propose a concrete design and **I'm willing to implement it myself (backend + frontend PRs)** — but I'd rather get a maintainer 👍 on the direction first than show up with an unsolicited PR.

### The problem

Today there are two options for shared config files, and both have a drawback:

- **Top-level file** → shows up as a device that is permanently "offline".
- **Subfolder (e.g. `common/`)** or **dot-prefixed file** → hidden from the device list (good), but can't be opened in the dashboard editor at all, because `EditRequestHandler` only serves files whose parent is the config dir:

  ```python
  if filename.resolve().parent != settings.absolute_config_dir:
      self.send_error(404)
  ```

So people who use packages the recommended way (a subfolder) have to leave the dashboard to edit them. On HA OS that means the VS Code add-on, and on plain Docker setups a separate editor or SSH. There's a second, less obvious problem: when you change a shared file, **nothing tells you which devices are affected and need to be reinstalled.**

### Proposed design

**1. "Packages" button in the header**, next to "Secrets":

```
  UPDATE ALL   CLEAN ALL FILES   🔒 SECRETS   📦 PACKAGES   🔍  ⋮
```

**2. Package list**: YAML files in subfolders of the config dir, grouped by folder, laid out like the device list. Each entry shows which devices use it. You can view and edit a file, and create a new one.

```
 common/
 ┌──────────────────────────────────────────────┐
 │ base-settings.yaml        used by 3 devices  │
 │ EDIT                                      ⋮  │
 └──────────────────────────────────────────────┘
 ┌──────────────────────────────────────────────┐
 │ shelly-plus-1.yaml        used by 1 device   │
 │ EDIT                                      ⋮  │
 └──────────────────────────────────────────────┘
                                   [ + NEW PACKAGE ]
```

**3. A small counter on the device card**, e.g. `📦 4`, not the full list (that gets cluttered quickly). Clicking it shows the included files, and each one opens in the editor.

**4. Includes in the editor are clickable.** `!include common/base-settings.yaml` (and local `packages:` entries) can be clicked and open that file.

**5. Reinstall affected devices after saving.** When a package is saved, offer to install the devices that use it, along the lines of: *"3 devices use this file — install them now?"* This is the part I think adds the most value; today you have to keep track of it yourself.

**6. (Optional, follow-up)** A badge on the device card, *"included file changed since last install"*, when an included file is newer than the last build. It works alongside the existing "update available" indicator.

### Design decisions / safety

- **No fixed folder name.** Any subfolder of the config dir counts, so there's no new convention to learn. `.esphome/`, hidden folders and `secrets.yaml` stay excluded.
- **Path safety.** We'd relax the editor check from "parent == config dir" to "resolves inside config dir". The path is still resolved first, symlinks are only followed if the target stays inside the config dir, and `..` / absolute paths are rejected, with tests covering traversal cases.
- **Device detection doesn't change.** `list_yaml_files` / `filter_yaml_files` stay exactly as they are, so no existing setup suddenly shows new or missing devices.
- **Include detection** uses a lightweight pass over the YAML that collects `!include` targets and local `packages:` file references, without fully validating or compiling the config. Remote (git) packages are shown as remote and not resolved. If some files can't be resolved, the counter shows what was found; it should never break the dashboard.

### Implementation plan

Split into small, independently reviewable steps:

1. **Backend (`esphome/esphome`)**: allow editing YAML files in subfolders (with tests), plus an endpoint that lists package files with their "used by" devices.
2. **Frontend (`esphome/device-builder-frontend`)**: "Packages" header button and list, create new package, open in the existing editor.
3. **Backend + frontend**: include counter on device cards, clickable includes in the editor.
4. **Backend + frontend**: "install affected devices" after save, and optionally the "changed since last install" badge.

Step 1 on its own already fixes the most common complaint in this thread.

### Questions for maintainers

- Is this direction acceptable in principle? If you'd prefer only part of it (e.g. steps 1+2), that works for us too.
- Any preference on how include detection should be done, e.g. reusing parts of `yaml_util` vs. a separate lightweight scanner?
- Should the "Packages" view use the same component as the device list, or would you prefer a simpler file list?

If this sounds reasonable, I'll start with step 1 and open a draft PR so we can discuss details on actual code.

sebo-b, 2026-10-28:
@MatthiasNeuhaus, thanks for the detailed proposal and for offering your time to build it. I needed to manage several shared packages myself, so I went ahead and implemented it along the lines discussed here, for the new Device Builder:

- Backend: esphome/device-builder#2991 (`files/*` commands; merge first)
- Frontend: esphome/device-builder-frontend#2049

Any YAML in a subfolder of the config dir is a package. Top-level YAMLs stay devices, so existing setups don't change. You can list, create, edit, rename and delete packages. "Show usage" lists the devices that use a file, and after a save the dashboard offers to install them. Ctrl/Cmd+click on an `!include` opens the target. Renaming stays within the package's folder and rewrites the `!include` references to it.

I hope this covers your use case too. Feedback on the PRs is very welcome.

Two points need a maintainer decision:

1. **A "Packages" entry in the header menu**, next to Secrets.
2. **How the package list stays in sync.** It's a directory listing loaded when the page opens and again on reconnect, like `devices/list_archived`, rather than the snapshot + events pattern `ARCHITECTURE.md` asks for. A package watcher in the scanner could be a follow-up.

ssieb, 2026-10-28:
Please come discuss this in the device-builder channel on the esphome discord server.

MatthiasNeuhaus, 2026-10-29:
@sebo-b, thank you, that's great to see! This covers the core of what I had in mind, and the rename with reference rewriting and reusing `Device.dependencies` instead of a separate scanner are nice improvements over my sketch.

I'll go through both PRs and leave my feedback there as a review.

hn, 2026-10-29:
Thanks to everyone who has contributed to this discussion, @sebo-b!

I strongly vote for supporting a filename prefix (e.g. `package-*`, `include-*`) in addition to subdirectories to indicate a package or include file, as described [here](https://github.com/esphome/feature-requests/issues/2351#issuecomment-1685251958). It's simpler and more lightweight, especially for new users.

sebo-b, 2026-10-29:
> 13\. Thanks to everyone who has contributed to this discussion, [Sebastian Baberowski (@sebo-b)](https://github.com/sebo-b)!
>      I strongly vote for supporting a filename prefix (e.g. `package-*`, `include-*`) in addition to subdirectories to indicate a package or include file, as described [here](https://github.com/esphome/feature-requests/issues/2351#issuecomment-1685251958). It's simpler and more lightweight, especially for new users.

Hey @hn,

Thanks for the feedback! I did consider a filename convention (`package-*.yaml`, `include-*.yaml`, `*.template.yaml`, …), but I went with subfolders for a few reasons:

- It works with the existing tooling as is. ESPHome already ignores YAML files in subfolders when it lists devices. A prefix convention would need changes in esphome itself. Otherwise Device Builder, the CLI and the legacy dashboard would disagree about what counts as a device.
- IMHO it's easier for beginners. A hardcoded naming pattern is something you have to know about. If you get it slightly wrong (`packages-foo.yaml`, `pkg-foo.yaml`), the file quietly becomes a "device" again. With a folder, the rule is simply that anything in a subfolder is a package.
- It's one click in the UI. "New package" creates the file in a folder for you, so new users never have to learn the convention.

If there's still strong demand for prefixes later, they could be added on top of subfolders without breaking anything. I'd just rather start with the approach that needs no new rules.

What do you think?

MatthiasNeuhaus, 2026-10-29:
I'd go with subfolders too. The point that the CLI, the legacy dashboard and Device Builder agree on what counts as a device without changing esphome itself convinces me most. A typo in a prefix silently turning a package back into a device would be confusing, especially for beginners. My own setup (`common/` for packages, `bridge/` for migration firmwares) worked without renaming a single file. Prefixes on top can still come later if there's demand.

hn, 2026-10-30:
There's no right or wrong here, but in this case, from my experience, I'd vote for naming by purpose, not (only) location, especially for new users. Personally I'm totally fine with subfolders, but `package-wifi.yaml` tells a beginner *what* the file is, wherever it shows up. With subfolders, the meaning lives in the path, so on its own `wifi.yaml` could just as well be a device, and folder names like `base/` are a personal convention you have to know.

Home Assistant works the same way: `automations.yaml`, `scripts.yaml` or `secrets.yaml` are self-explanatory solely by file name. ESPHome already does this with `secrets.yaml`, too.

Both approaches could coexist very well, though.

Much of Home Assistant's success comes from being approachable for newcomers, and I think ESPHome benefits from the same. That's why I keep the beginner's perspective in mind here.

ssieb, 2026-10-30:
It's not like something is creating those files for the user.  It's the user that's creating them and will know what they are, so I don't see a naming issue.  Putting them in a directory is understandable.