esphome/feature-requests #2360: Addition of Daikin IR remote ARC466*** into the IR Remote Climate component

syed82ab, 2023-08-19:
**Describe the problem you have/What new integration you would like**

Addition of support for Daikin ARC466*** IR remote control.  

**Please describe your use case for this integration and alternatives you've tried:**

I would like to integrate IRremoteESP8266 implementation of IR remote controlling the Daikin ARC466*** into the ESPHome framework. I guess I can make a [https://esphome.io/components/climate/custom](custom climate) but it will be nicer if it's already in the default IR Remote Climate integration library.

**Additional context**

In the [https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Daikin.h](IRremoteESP8266) library, this is the "DaikinESPProtocol".

lrodorigo, 2024-07-20:
Hi, 
any update? Did you finally succeed?

I'm also intrested in configuring IR for DAIKIN ARC466.

davidandreoletti, 2026-11-03:
@syed82ab  @lrodorigo Did you make progress on this by any chance ?