esphome/feature-requests #2365: Use ids rather than names to identify entities to HA via api

clydebarrow, 2023-08-21:
**Describe the problem you have/What new integration you would like**

(I'm posting this in Feature Requests because although it's aimed at fixing a problem, the problem is one of design not implementation, and thus not strictly a bug.)

I recently encountered a problem with HA documented here: https://github.com/home-assistant/core/issues/97726

After some digging it appears the core of the problem is that ESPHome constructs a "unique" id to identify an entity to HA by mangling the entity name and the device name. Changing either name makes it seem that the entity has disappeared and a new one created. So:

1. I propose that the id used within the API should be based on the ESPHome id if provided (if not provided, I guess falling back to name-mangling is the only good option.) This would ensure that entity name changes do not change the identity of the entity.
2. There is no good reason that I can see for including the device name in the unique ID. HA is already able to maintain the device identity across device name changes, and only needs to uniquely identify an entity within the namespace for a given device.

These are obviously significant changes, but in the long term I believe it would make the API more robust. Implementation will have some challenges:

1. Breaking existing devices - clearly it would be unacceptable to have existing devices change their entitities' identities on an upgrade. This should be able to be managed with API versioning. 
2. Changes are required on the HA and ESPHome side and will need to be coordinated. 

The actual implementation on the ESPHome side shouldn't be too hard, probably more work on the HA side.  The biggest obstacle to getting this done will be convincing the respective dev teams that the gain is worth the pain, and that requires some discussion, and devising a workable implementation strategy.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

clydebarrow, 2023-09-08:
Noting that this issue also affects the [Webserver REST API](https://esphome.io/web-api/#api-rest) where the `id` required in a REST call is the mangled name including the device name, not the `id:` specified in the YAML.

ssieb, 2023-09-08:
The `id` is often not specified by the user and will get a somewhat random value.  It's literally the C++ variable name used when generating the `main.cpp` source code.

clydebarrow, 2023-09-08:
If the user *does* specify it then it seems logical to me that it should be used. It's already the case that the user has to specify an ID if they want to refer to it elsewhere in the YAML, so it seems reasonable to me that the same should apply for use in the REST API.

clydebarrow, 2023-09-08:
I'm applying the *principle of least astonishment* here - requiring an ID in order to refer to something seems less surprising than having to read and understand a paragraph explaining how infer the ID (or turn on VERY VERBOSE logging just to find out what it is.)

I'm ignoring the isse of backwards compatibility of course.

ssieb, 2023-09-08:
Except that's the opposite.  The id is specifically internal only, it's not supposed to be visible outside of the ESP.
Are you going to update all the docs to say that you need to specify an id if you want a useful REST target and HA entity id?

clydebarrow, 2023-09-08:
That would be the idea. The alternative is to leave things in the current rather mangled state. I didn't say it would be an inconsequential change.

Dilbert66, 2023-11-03:
Adding my 2 cents here.  I do feel that storing the id as part of the esphome::EntityBase class as well as using it as the instance variable name would be beneficial.  For example, I have a custom component that needs to know that a specific templated sensor exists in the yaml configuration using c++.  I can't look by the id as that disappears in the generated code.  Only the name is stored with the entity.  My work around is to have the user add the unique id i brackets (idxx) at the end of name where I can do a lookup for it in the entities structure after the fact.  If the id would be stored , I could then search for that id instead of having the name field complicated with the addition of that id.  Of course that would require a bit of code to implement but would be useful.  It would not be mandatory so would not break anything just an added usefull piece of info.