esphome/feature-requests #2367: Medisana BS410/BS430/BS440/BS444 BLE scale integration

fischmuc, 2023-08-22:
**Describe the problem you have/What new integration you would like**
Integration of scales from Medisana BS410/BS430/BS440/BS444 based on existing Phyton code from @keptenkurk https://github.com/keptenkurk/BS440

**Please describe your use case for this integration and alternatives you've tried:**
Expand the scale product range. Currently only "Miscale" is supported

**Additional context**

bwynants, 2023-12-19:
will this work for you?
https://github.com/bwynants/weegschaal

jondiez, 2024-01-14:
For me the above does not work, I have a Bs430 and it only tells me the characteristics of the people recorded on the scale.  There is a colleague on that github who must have achieved it with esp32-ble-arduino, but no idea how to do it.  It would be nice if it could be integrated by choosing the scale model, because it is surely something small to modify in the code.  Thank you!!!