esphome/feature-requests #2369: Arduino as ESP-IDF component

olicooper, 2023-08-23:
**Describe the problem you have/What new integration you would like**

Arduino can be loaded as a component over ESP-IDF as described here: https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html and here: https://github.com/platformio/platform-espressif32/issues/24

It would be great if ESPHome would allow this configuration:
```yaml
esp32:
  board: esp32dev
  framework:
    type: arduino, esp-idf
```

**Please describe your use case for this integration and alternatives you've tried:**

I would like to customise the `sdkconfig` compilation flags for ESP-IDF. It would also be useful for people who are migrating their code to pure ESP-IDF.

**Additional context**

I am trying to set custom pins for the PSRAM built in to the Sonoff NSPanel. It has 16MB PSRAM (see [this link](https://blakadder.com/nspanel-teardown/) for more info), but the pins are non standard, so I need to set these flags:
```
CONFIG_D0WD_PSRAM_CLK_IO=5 # default: 17
CONFIG_D0WD_PSRAM_CS_IO=18 # default: 18
```
I have tested this using pure esp-idf and I can detect the PSRAM, but Arduino doesn't allow setting these options due to the bootloader being precompiled. The only alternative it to create a custom bootloader using https://github.com/espressif/esp32-arduino-lib-builder but I think esphome wouldn't be able to include this.

olicooper, 2023-08-30:
I noticed work being done on [PSRAM support here](https://github.com/esphome/esphome/pull/5298), this issue seems related (needing to change the PSRAM pins e.g. CONFIG_D0WD_PSRAM_CLK_IO)

alfwro13, 2023-10-13:
if I understand this correctly this would be a great workaround for esphome features that currently only run on arduino framework (i.e. media_player component)

Bascht74, 2024-03-26:
Hi @alfwro13, 

could you share the yaml-configuration, 
that you used trying out the esp-idf framework with the nspanel?

Thx, 

Sebastian