esphome/feature-requests #2371: HA status sensor/attribute for Esphome devices.

luixal, 2023-08-25:
**Describe the problem you have/What new integration you would like**

I want to have a sensor in HA that represents esphome device state: online u offline, the same way it is shown in ESPHome dashboard. Or just an attribute in the entity.

**Please describe your use case for this integration and alternatives you've tried:**

Showing current sensor state. In my case, I need to put sensors to sleep before cutting power off but, if the sensor is offline for some reason, scripts used can cause errors. If I just have this sensor, I can handle theses cases.

Only alternative I've found is guessing the on/offline state from the last update field in HA sensor. And it has to be tweaked for each sensor.

**Additional context**

Well... device state is already being tracked in the dashboard... why not having this as a sensor?

slimcdk, 2023-08-28:
The dashboard simply sends a ping to the device host/ip. You can inspect it from the development tools in most browsers. You can create a similar sensor with https://www.home-assistant.io/integrations/ping/

luixal, 2023-09-08:
Great info, thanks!

closed 2023-09-08 by luixal