esphome/feature-requests #2375: Please add st7789v option for mirroring the display output

niahane, 2023-08-29:
**Describe the problem you have/What new integration you would like**

I use this display platform with a prism, so the image is In reverse.
It seems that no options is avaible for obtain rendering reversed.
[image]

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

niahane, 2023-08-31:
I have modified the source st7789v.cpp and it seems to work perfectly:
void HOT ST7789V::draw_absolute_pixel_internal(int x, int y, Color color) {
  if (x >= this->get_width_internal() || x < 0 || y >= this->get_height_internal() || y < 0)
    return;

  int inverted_x = this->get_width_internal() - 1 - x;

  if (this->eightbitcolor_) {
    auto color332 = display::ColorUtil::color_to_332(color);
    uint32_t pos = (inverted_x + y * this->get_width_internal());
    this->buffer_[pos] = color332;
  } else {
    auto color565 = display::ColorUtil::color_to_565(color);
    uint32_t pos = (inverted_x + y * this->get_width_internal()) * 2;
    this->buffer_[pos++] = (color565 >> 8) & 0xff;
    this->buffer_[pos] = color565 & 0xff;
  }
}

Unfortunately for now i am not able to add an option for enable/disable it in yaml file :(

Deepintheeast, 2025-12-24:
Is there a solution for this yet?