esphome/feature-requests #2376: Please, add support to enable /disable LD2410 sensor

Ahmed-farag36, 2023-08-31:
**Describe the problem you have/What new integration you would like**

I want to enable/ disable the sensor through UART 

**Please describe your use case for this integration and alternatives you've tried:**

I am using a custom component to add LD2410 sensor to home assistant, recently I noticed the official integration which is great but I miss this ability to enable/ disable the sensor
**Additional context**

If somebody know how but don't have the spare time, guide me and I might add this ability.

Ahmed-farag36, 2023-09-04:
Sorry for being late
In fact it is showing sensor data or not
**Why do I need it ?**
Right now I am connecting the sensor with esp-12f through UART and at the same time using the UART for flashing the node for first time serially but it shows error despite flashing correctly, I need to stop sensor sending data till flashing done.
Like this library *I don't remember the author now*
https://github.com/Ahmed-farag36/esphome-project-template/blob/4544bf9fc80e2d5b494d659e96c83a716fc92c8d/ld2410.h#L90
**And if you wondering why not use esp32 and ease it on myself

ssieb, 2023-09-04:
If the sensor is connected to the uart pins and has power, it's going to interfere regardless of whether or not it's sending data.  Either disconnect the sensor power or disconnect the sensor TX pin.
But how is adding this support going to help anyway if you can't flash it?

Ahmed-farag36, 2023-09-04:
No, I am able to flash the esp
To get the full picture, I am using a jumper to temporarily connect the USB / UART chip for initial flash, then I remove the jumper.
By default the sensor is permanently connected to esp.
I am flashing the esp through esp web tools (serial improv) after reaching 100% it throws an error and doesn't complete the on boarding process (wifi configuration).
It has flashed successfully and I can connect to its access point and configure the wifi credentials.
A method I have tried is to stop showing sensor readings till this initial flash complete and it succeeded 
Now that esphome has an official integration for ld2410 I want to use it instead

ssieb, 2023-09-04:
Oh, so the problem is the serial improv.  You have no reason to use that.  You said you connected and configured the wifi already, although you should have done that in the config anyway.  The initial flash is complete and successful.

closed 2023-09-04 by ssieb