esphome/feature-requests #2378: Filtering and configurable filter buttons

rubydesign, 2023-09-02:
**Describe the problem you have/What new integration you would like**

I'm getting to 200 devices and even with clear naming conventions, finding and checking devices is starting to be a problem.
Filtering on status, as has been proposed, would help. But i  am also suggesting configurable filter buttons.

**Please describe your use case for this integration and alternatives you've tried:**
When managing many devices, scrolling and searching are not ideal tools. 
Scrolling for say offline devices just takes a long time.
And search resets after an action (ie install) has been taken, thus one has to start from the beginning.
The search also takes three clicks instead of one (because the box is hidden), so that aggravates the situation (but that is not part of this request)

My suggestion is to have a configuration option that would add "quick search/filter" buttons to the menu (ie top bar).
I would like configure a button eg for "offline" devices. Or another one the searches for "sensor".
That way i can easily configure the sets that are relevant for me, while not cluttering the ui with things that are not relevant for people with just 20 devices.

**Additional context**

There has been previous talk of filtering https://github.com/esphome/feature-requests/issues/751
but that never went anywhere.

Also there was talk during the original search PR https://github.com/esphome/dashboard/pull/359
of adding filtering by status, but that also did not happen.

The above configurable suggestion is meant to have no change for most users, while providing significant benefit for the people with many devices.

I could knock up yaml examples or screenshopts to clarify if i wasn't clear enough.

hn, 2023-09-04:
I opened a related issue some weeks ago that might be relevant here: #2351 "Dashboard: Better handling for user-editable yaml include files"

coenier, 2024-01-24:
Would like to see some filtering on dashboard as well. 

Some filtering ideas

* Adding tags to yaml to allow for tagging of devices (relays, sensor etc), I would also tag them with things like room etc so that I can create my own quick filtering setup (just an idea)
* Online and off line status
* Devices that has updates and ones that has none