esphome/feature-requests #2382: Add support "publish state" for fan component

Ranchuking, 2023-09-07:
Hello, like the switch component, I hope we can add the support of "publish state" function for the fan component. I add a template output to a speed fan, then if I receive a rf433 code, the state of the fan should be changed.  
**Describe the problem you have/What new integration you would like**

I add a template output to a speed fan, then if I receive a rf433 code, the state of the fan should be changed. But now we only support "turn on" and "turn off". If I want to change the state, I need to use "turn on" or "turn off". and the fan component will do an action too, not just change the state of the speed fan. Thank you.
**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

ssieb, 2023-09-07:
What do you mean it will do an action and how is it different from a switch?

Ranchuking, 2023-09-07:
> What do you mean it will do an action and how is it different from a switch?
Well, in the template switch, we can publish the state to make it change the state, just change the state but not make an action "switch on" or "switch off". in the fan component, we can't only change its state. right?

ssieb, 2023-09-07:
Oh, you're referring to the template switch.  It depends on what action you're referring to.  Changing the state will trigger the `on_turn_on` and `on_turn_off` triggers, same as the ones on the fan.