esphome/feature-requests #2385: Add support aditional funcion on sensor SPS30

tiimsvk, 2023-09-11:
**Describe the problem you have/What new integration you would like**
Hello, could someone look at the other functions in the datasheet for the SPS30 dust particle sensor.
- switching on and off auto clean
- Fan speed warning
- Laser failure
- Fan failure

These functions would help in diagnosing the fault if the sensor is installed in an inaccessible place.
well thank you

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

tiimsvk, 2023-09-12:
Solution:

It is necessary to create an external component and define it in yaml:

```
external_components:
   - source: custom_component
```

Create the sps30 directory in the custom_component folder and copy the entire original sps30 integration into it

```
- automation.h
- sps30.cpp
- sps30.h
- __init__.py
- sensor.py
```

Further editing of files:
sps30.h:
after line 31 (bool start_fan_cleaning();) add the following:

```
   bool stop_measure();
   bool start_measure();
```

sps.cpp:
add the following at the end:

```
bool SPS30Component::start_measure() {
   uint8_t data[4];
   data[0] = SPS30_CMD_START_CONTINUOUS_MEASUREMENTS & 0xFF;
   data[1] = 0x03;
   data[2] = 0x00;
   data[3] = sht_crc_(0x03, 0x00);
   if (!this->write_command(SPS30_CMD_START_CONTINUOUS_MEASUREMENTS, SPS30_CMD_START_CONTINUOUS_MEASUREMENTS_ARG)) {
     ESP_LOGE(TAG, "Error initiating measurements");
     return false;
   }
   return true;
}

bool SPS30Component::stop_measure() {
   if (!write_command(SPS30_CMD_STOP_MEASUREMENTS)) {
     ESP_LOGD(TAG, "Stop measurement");
   }
   return true;
}
```

Subsequently, it is sufficient to invoke the action using the button template:

```
buttons:
   - platform: template
     name: "Stop measure"
     on_press:
       then:
         - lambda: id(my_sps30).stop_measure();

   - platform: template
     name: "Start measure"
     on_press:
       then:
         - lambda: id(my_sps30).start_measure();
```

There is one catch when you press stop, if you have set an update interval for some time, then every 5 sensor reading failures, the sensor is reset and continues measuring again.

This setting is defined in sps30.cpp

`static const uint8_t MAX_SKIPPED_DATA_CYCLES_BEFORE_ERROR = 5;`

[custom_component.zip](https://github.com/esphome/feature-requests/files/12583075/custom_component.zip)

DR-PLANTECHSTEIN, 2024-04-28:
why not merge this?