esphome/feature-requests #2387: Add support for W32-SC01 Plus

broyuken, 2023-09-15:
**Describe the problem you have/What new integration you would like**

Support for the W32-SC01 Plus dev board

**Please describe your use case for this integration and alternatives you've tried:**

I am trying to build a device that will show a clock, current audio track and have a speaker that I can play music from Home Assistant on. I have tried OpenHasp but it doesn't have audio support, and esp squeezelite doesn't have great display abilities. 

**Additional context**

jd1, 2024-04-04:
I got the board working with this component https://github.com/strange-v/ha_deck but the configuration is limited and I had some issues with hiding UI components.
I would love to see native support for the W32-SC01 Plus!

nagyrobi, 2024-04-04:
@nielsnl68 is working on it.