esphome/feature-requests #2388: Desktop/native platform support

robryk, 2023-09-15:
**Describe the problem you have/What new integration you would like**
I would like to be able to compile a ESPHome configuration to a binary that can be run on a desktop. Obviously, such a configuration could not e.g. refer to any pins, nor any actual hardware. 

**Please describe your use case for this integration and alternatives you've tried:**

This could be a valuable way to:
 - run tests (both manual and CI),
 - more easily move pieces of logic between an ESP device and a more standard computer in a HA-less environment (esp. with #2081 implemented).

Obvious alternative for the latter is actually running HA or writing the on-desktop piece of logic in Python, so that it uses the native API client library. I don't see any obvious alternatives for CI-style testing.

**Additional context**

Note that platformio has a ["native" platform](https://docs.platformio.org/en/latest/platforms/native.html) that could be useful here.

Unfocused, 2024-01-27:
ESPHome actually already has this, as of 2023.5.0 - but it's intentionally not documented anywhere yet since there's not much that works, and it's currently primarily for internal development. Although, there's been some recent activity in improving (& adding) things from @clydebarrow.

Just use the `host` platform component. eg:
```yaml
esphome:
  name: my-fake-device

host:
```

clydebarrow, 2024-01-27:
Yes, I'm using this to test LVGL integration, simulating a display with VNC.

```
external_components:
  - source: github://clydebarrow/esphome@lvgl
    refresh: 10min
    components: [ lvgl, font, vnc, host, sntp, time]

host:
  mac_addr: "62:23:45:AF:B3:DD"

touchscreen:
  - platform: vnc
    id: vnc_touchscreen
    display: vnc_display

display:
  - platform: vnc
    id: vnc_display
    update_interval: never
    auto_clear_enabled: false
    dimensions:
      width: 800
      height: 800

lvgl:
  touchscreens:
    - touchscreen_id: vnc_touchscreen
      long_press_time: 800ms
      long_press_repeat_time: 400ms
  color_depth: 16
  byte_order: big_endian
  displays:
    - display_id: vnc_display
   ```

agillis, 2024-05-30:
That seems really usefile. CAn you give me a full working example. I tried it but I get an error.   `KeyError: 'target_platform'`

```
esphome:
  name: "my-fake-device"

host:
  mac_addr: "62:23:45:AF:B3:DD"

external_components:
 - source: github://clydebarrow/esphome@lvgl_original
   refresh: 10min
   components: [ lvgl, font, vnc, host, sntp, time] 

touchscreen:
  - platform: vnc
    id: vnc_touchscreen
    display: vnc_display

display:
  - platform: vnc
    id: vnc_display
    update_interval: never
    auto_clear_enabled: false
    dimensions:
      width: 800
      height: 800

lvgl:
  touchscreens:
    - touchscreen_id: vnc_touchscreen
      long_press_time: 800ms
      long_press_repeat_time: 400ms
  color_depth: 16
  byte_order: big_endian
  displays:
    - display_id: vnc_display
    ```

clydebarrow, 2024-05-30:
> Can you give me a full working example

I updated the sample above. See also https://github.com/esphome/esphome/pull/6825

closed 2024-06-01 by clydebarrow