esphome/feature-requests #2389: Add support for Raspberry PI 4

halogen28372, 2023-09-16:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

Add support for Raspberry PI 4. This would allow me to have the PIs act as hubs for sensors around the house. ESP32s are more flaky and lose a lot of features if hardwired.

michd, 2023-09-21:
I think [Home Assisstant](https://www.home-assistant.io/) is what you're looking for then. :)

halogen28372, 2023-09-21:
Nope, that still wouldn't give me native control of the gpio.

matthew73210, 2023-12-11:
How would having a RPI4 running esphome have any advantage over an ESP32 variant?

halogen28372, 2023-12-12:
Mainly Ethernet and POE support. Secondarily hardware quality is generally better.

nagyrobi, 2023-12-12:
The whole point of microcontrollers like ESP32 is that they run the binary directly and don't need an operating system to do that.

A full blown computer architecture (be it even RPI) requires a disk-based operating system and process management etc. introducing a whole lot of overhead causing energy waste, a lot of error sources, needs maintenance etc.

A microcontroller is kinda set it and forget it solution. Once installed, it will work for ages.

matthew73210, 2023-12-12:
> Mainly Ethernet and POE support. Secondarily hardware quality is generally better. 

Lilygo do a poe microcontroller. And the RPi has less stable gpio support as it can hang from user input, kernel panic etc..
If you want to use an RPi, node-red can do what you want.

matthew73210, 2023-12-12:
> The whole point of microcontrollers like ESP32 is that they run the binary directly and don't need an operating system to do that.
>
>
>
> A full blown computer architecture (be it even RPI) requires a [...]

I concur, total waste.

nagyrobi, 2023-12-12:
ESPHome supports ESP32 with Ethernet very well, and there are many PoE versions too: https://esphome.io/components/ethernet#configuration-examples

closed 2023-12-12 by nagyrobi