esphome/feature-requests #2392: Alarm Control Panel - integrated notifications

MayorBug, 2023-09-17:
**Describe the problem you have/What new integration you would like**

I would like Alarm panel integration to have HA notification support baked in. 

**Please describe your use case for this integration and alternatives you've tried:**

Currently: It is quite simple to get a simple notification when alarm is using the on triggered: actions

```
on_triggered:
    then:
       send notification to HA
```

But if some additional context is wanted, like which sensor trigged the alarm then it required custom lambdas and that makes the setup much more complicated. 

Expectation: - Lets say there would be "notify: True" flag then the integration would send notifications directly using HA API and would include additional info like which sensor triggered and how many times.  Additionally it would be beneficial to have to have some config options for the notifications like "TTS: True" -  then it would send TTS voice through companion app.

[https://companion.home-assistant.io/docs/notifications/notifications-basic#text-to-speech-notifications](url)

```
action:
      - service: notify.mobile_app_<your_device_id_here>
        data:
          message: "TTS"
          data:
            tts_text: "Alarm has been triggered"
            media_stream: "alarm_stream_max"
```

**Additional context**

ShadyMCShadesen, 2023-09-17:
I think Alamo has the feature you want
https://github.com/nielsfaber/alarmo

MayorBug, 2023-09-20:
I am coming from Alarmo, but after I found out I can run the alarm logic directly on the esp I switched. I prefer running the alarm system directly on the esp since I am using wired sensors hooked to it. Hence why I was asking for this feature to be baked into Esphome