esphome/feature-requests #2393: Add support for ESP32-Wrover-IB Flash 16Mb, PSRAM 8Mb

19leva89, 2023-09-19:
**Describe the problem you have/What new integration you would like**

I want to use ESP32-Wrover-IB Flash 16Mb, PSRAM 8Mb in my project, please make integration for this.

**Please describe your use case for this integration and alternatives you've tried:**

I want to integrate ESP32-Wrover-IB into HA and link it with Zigbee module RF-BM-2652P2

**Additional context**

[https://robotools.in/wp-content/uploads/2022/09/Espressif-ESP32-WROVER-IB-Flash-WiFi-Bluetooth-Module-600x600.jpg](url)

ssieb, 2023-09-19:
What's not working?

19leva89, 2023-09-19:
esp32:
  board: I don't know what board to indicate here
  framework:
    type: arduino

randybb, 2023-09-19:
Just use the default one. platformio templates for most of the boards are exactly the same, so there is no point of selecting a different one.

19leva89, 2023-09-19:
If I choose board: esp32dev -> HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash, but I need HARDWARE: ESP32 240MHz, 8MB RAM, 16MB Flash

randybb, 2023-09-19:
RAM != PSRAM
PSRAM will be used automatically for display components, you can add `psram:` (without any params) to see it in the log.

For the flash you can change that with platformio options, but ESPHome will use 4 MB (2 MB for running, 2 MB for OTA) anyway, so this doesn't matter... 

```yaml
esphome:
  platformio_options:
    board_build.flash_mode: dio
    board_build.flash_size: 16MB
    board_upload.flash_size: 16MB
    board_upload.maximum_size: 16777216
```

19leva89, 2023-09-19:
Thanks for the tip, now my config looks like:
```
  platformio_options:
    board_build.flash_mode: dio
    board_build.flash_size: 16MB
    board_upload.flash_size: 16MB
    board_upload.maximum_size: 16777216
    board_upload.maximum_ram_size: 8388608
    board_build.partitions: "../../../custom_partitions_16mb.csv"
```