esphome/feature-requests #2395: Add support for CST816D touch panel

zagnuts, 2023-09-21:
**Describe the problem you have/What new integration you would like**

I was wondering if there was someone out there who was able to translate the existing Arduino CST816D drivers to a format that esphome can accept, or point me towards something that might help me work out how to do it myself? The device has a number of example files, including files required for Arduino IDE. 

**Please describe your use case for this integration and alternatives you've tried:**

You can now get a GC9A01 TFT screen with an integrated ESP32-2424S012 and CST816D touch screen. It is inexpensive, and can get it bare or with a black or white case so other than installing esp home it is plug and play. I believe that this touch panel is also used in the Lilygo T-Display-S3 Touch module.  Once we have the touch panel working, this small device would make an excellent and inexpensive control panel for thermostats, lights or other devices - eg tapping it to toggle a light on/off, swiping up/down to brighten/dim the lights, swiping left or right to view other screens.... 

**Additional context**

We’ve managed to get this running ([thread here](https://community.home-assistant.io/t/esp32-c3-with-integrated-gc9a01-cheap-touch-controller/610314)) so it displays info on the screen, and can even add the touch panel driver via the "include" option, but we are currently struggling trying to get the touch panel to actually work..

zagnuts, 2023-09-21:
[1_1_Factory_samples.zip](https://github.com/esphome/feature-requests/files/12682494/1_1_Factory_samples.zip)
FWIW Here are is the sample code

dvbit, 2023-09-21:
Upvote

bearpawmaxim, 2023-12-26:
@zagnuts in progress.

clydebarrow, 2023-12-27:
https://github.com/esphome/esphome/pull/5941