esphome/feature-requests #2396: Fan Control: Burst Mode

abnormisdictum, 2023-09-21:
**Describe the problem you have/What new integration you would like**
Unlike purely resistive loads (like bulbs) motors and fans cannot be speed controlled using simple phase firing AC Dimmer circuits. Having to create a VFD (variable frequency drive) style system is both tedious and risky (310VDC rectification). I was looking at burst firing.

Burst Firing is when you turn on the output (SSRs) for a certain number of cycles and off for a certain number of cycles, making sure to turn off the output at the zero crossing. So assuming that you have a 60Hz line, the fan would be on for 20cycles and off for 40 cycles would yield you a fan/motor speed of 33% (Ideally, doesn't have to be precise).

I have an input (which I used for the Phase Trigger Control Dimming application) where I use a HA11AA1 Optocoupler to sense both positive and negative cycles and can use this for zero-crossing detection. We would need to attach a Interrupt to the Zero-Crossing input and basically count the number of times zero crossing is detected (Double the number in case of bi-directional opto-couplers like the HA11AA1, because you would get two triggers per AC cycle) and turn on or off the output accordingly making sure to reset the counters.

**Please describe your use case for this integration and alternatives you've tried:**
This can be used to control fans and motors without causing transients that will cause humming or motor over heating associated with using Phase firing AC dimmers.

I had a look at the ESPHome repo to understand where I could make this modification, but I quickly got lost. I am an industrial programmer, and though I have some experience with C++ and Java, I am not very adapt at it.

**Additional context**
If any one could guide me how and were to make this modification, I would be very grateful.

Warm Regards.

nagyrobi, 2023-09-21:
What a great idea!

abnormisdictum, 2023-11-01:
Okay, so I have solved this issue. Sort of.

Instead of this whole convoluted system that I was thinking up, it is easier to use a simple PWM signal of 1Hz (yes, one hertz) on a Zero-Crossing SSR/Triac. When you control the "width" of this PWM signal, you are essentially allowing x number of half-cycles (because zero crossing will turn on/off only at each half cycle). The greater the width of the PWM the more number of half-cycles hence greater the speed. I have tried it with 1Hz, 2Hz, 5Hz, 6Hz, etc and found that 1Hz gave the best performance.

The reason I say that I sort of found the solution is because it does introduce a jerk in the ON/OFF cycles. This won't be noticeable in a small well mounted fan (e.g an exhaust fan) but it will be quite pronounced in ceiling fans which have simple hook-and-bolt mountings.

My search for a simpler alternative to the tri-capacitor regulator continues.

closed 2023-11-01 by abnormisdictum