esphome/feature-requests #2397: Improve OTA for devices with small flash memory size

tr4nt0r, 2023-09-21:
**Describe the problem you have/What new integration you would like**

I want to suggest some improvements to the way ESPHome handles OTA Updates, especially for devices with small flash memory sizes like the esp8266. I understand that for OTA to work it is required that there is enough space free for the new firmware, old and new firmware need to fit on the flash memory together. 

Currently you don't get any warning if the firmware is too big and would prevent future OTA updates. Also when uploading a new firmware with the Home Assistant Addon and there is not enough space it doesn't tell you this.

My suggestions are:

1. When uploading a firmware through the ESPHome Addon and it fails because there wasn't enough space, the error message should also tell you that
2. Issue a warning and add a prompt before uploading a firmware that would occupy more than 50% of the available flash memory of the target device.
3. To better handle OTA on devices with small memory, the OTA component could have an option to upload an intermediate firmware only with the components required, similar to an OTA safe mode. This way it would free up space to upload the actual firmware afterwards. (This would also increase the above mentioned threshold to more than 50%)

**Please describe your use case for this integration and alternatives you've tried:**

I had the problem, that after updating to a newer ESPHome Version the Firmware for one of my ESP8266-based devices (Sonoff S20) grew to about 536kb, which is more than 50% of the 1MB flash this device has. The next time i tried to upload a new firmware the upload failed and the error message was unclear about what happened. After some tries i downloaded the bin file and uploaded it to the device through the web server component i had on this device. There i finally received a meaningful error message (not enough space).

After i figured out that the firmware was too big for OTA i tried to reduce the size by removing some components (like the web server component) but only managed to reduce it to slightly less than 50% but was still bigger than the space left on the device because of the previous firmware.

To avoid disassembling the device for programming I tried building a very minimalistic firmware and removed almost everything unnecessary except whats needed for the OTA component. This way i managed to upload a new much smaller firmware to the device and then i had enough space for the new firmware.

**Additional context**

netweaver1970, 2023-09-22:
I recently had the same issue, but luckily the latest ESPhome version did manage to flash an unmodified config/firmware. It seems that version created a more optimised/compressed binary. 
Maybe I was just lucky, I was already giving up on being able to flash it ever again, apart from trying your minimalistic manual approach. 

Actually, to be honest, most of the time there's no real need to reflash/upgrade once the device is deployed and working fine. But every time we do get the messages and for cleanliness, I do try to update all for every major release.

But your suggestions are very valid, it would be more like a fully automated Tasmota-minimal approach, still supporting smaller devices that way. Sounds very appealing...
+1 !

etamminga, 2023-11-17:
Same issue here. A simple Sonoff relay with nothing more than 1 binary_sensor, 1 switch and 1 status_led already fills the flash with 483kb. I was trying to add the webserver but that caused the file to grow >500Kb and failed OTA.

How can we minimize the amount of code compiled/included?

nagyrobi, 2023-11-17:
gzip the binary, and flash it like that

wrmulf, 2024-02-05:
I wander why ESPHome does not use gzip by default, nor as option in the upload process.

jonzyk, 2026-01-21:
+1
I am playig with an esp32-c6, bluetooth + zigbee are big ones, even 4MB can be small.