esphome/feature-requests #2402: Ethernet RMII PHY Fiber mode

DjordjeMandic, 2023-09-26:
**Describe the problem you have/What new integration you would like**

Considering that most cheap ethernet phys which are already supported by esphome support fiber mode(RTL8201, IP101G, and etc) it would be nice to support fiber mode in esphome(i guess it can be read from phy via MDIO or set).

**Please describe your use case for this integration and alternatives you've tried:**

Allowing devices to directly connect to fiber without media converters.

**Additional context**

1x9 or spf can be used, no need for magnetics. I know that 100M fiber is dying out but sometimes when devices are far fiber cabling is much cheaper.

USB to 100Base-FX card (based on cheap ic found in usb-rj45 100Mbit/s converters)
https://fabiobaltieri.com/2013/05/26/usb-to-100base-fx-optical-fiber-network-card/

nagyrobi, 2023-09-26:
Is there any hardware ready to test?

DjordjeMandic, 2023-09-27:
No but it can be made. Currently I'm leaving this just as suggestion. When I get free time I will try to see if this can first be implemented in ESP-IDF for existing supported ethernet phy transceivers as it will make it easier to later implement it in ESPHome.  

Idea is to make few boards for every supported phy with header for Broadcom/AVAGO AFBR-5803 or SFP card edge conector

nagyrobi, 2023-09-27:
This is a wonderful idea.