esphome/feature-requests #2405: Add support for single pair ethernet 10BASE-T1S, 10BASE-T1L

1965z6987, 2023-09-27:
**In new project line we will use 10BSAE-T1S on LAN8670, LAN8671, LAN8672 chips.**

Single-pair Ethernet is becoming increasingly popular. This standard provides great flexibility in connecting smart devices. The main advantage is the presence of an IP-stack, which allows you to do without additional gateways, modbus-ethernet, can-ethernet.  A wired connection is much more reliable than wireless, especially in areas with a large number of smart devices operating in the 2.4 GHz band.

**At my company, we test 10BASE-T1S on Linux. The results are very impressive. Devices can be connected like in RS-485 on one bus, while the speed of 9 Mbit/s is always available.** 
But not all devices need Linux.

**Please consider adding support for these chips: LAN8670, LAN8671, LAN8672**

nagyrobi, 2023-09-27:
Does this need customized hardware?

nagyrobi, 2023-10-02:
Duplicate of https://github.com/esphome/feature-requests/issues/2126

closed 2023-10-02 by nagyrobi