esphome/feature-requests #2410: Add support for ST7701S display

RynoM, 2023-09-29:
**Describe the problem you have/What new integration you would like**
I want to be able to use this display with [Rotary Switch Screen](https://www.tinytronics.nl/shop/en/development-boards/microcontroller-boards/with-wi-fi/smart-panlee-rotary-switch-screen-2.1-inch-ips-display-rotary-encoder-esp32-s3), to turn it into a ESPhome powered thermostat. 

**Please describe your use case for this integration and alternatives you've tried:**
I've tried naively implementing a display component based on another similar screen type like in [this project](https://github.com/velijv/nesp/blob/main/nesp.yaml), where I got the backlight to turn on but not much else.

**Additional context**
Found some interest in support for this display and someone stating that Arduino_GFX library already supports it [link](https://community.home-assistant.io/t/imagine-getting-esp-home-on-this-device/516921/5)

ShurikenGitHub, 2023-10-03:
Hi that's me that said that.

I use this code to get the display working:
```
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
480 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */,
bus, GFX_NOT_DEFINED /* RST */, st7701_type7_init_operations, sizeof(st7701_type7_init_operations));
```

So specifically "st7701 type 7" should be supported. (whatever that means)

RynoM, 2023-10-10:
Small world! I'm not very familiar with adruino though, but I want to run ESPHome firmware on the device and not adruino. Is there any way to use the fact that it can work in arduino to get it to work with ESPHome?

SamsHomeAutomation, 2023-11-04:
Hello,
Has anyone got going developing an external component yet?

ShurikenGitHub, 2023-11-05:
Not really, i can probably come up with something but i'm not a display driver developer or esphome developer, so it will take some time. If there is anyone with more experience go ahead!

SamsHomeAutomation, 2023-11-05:
Thanks for the reply. Don't worry about coding this. Was just hoping that someone may have already started developing something. Its a neat little display and would like to get it up and running someday.

fborangic, 2024-02-05:
I've found an [excellent rotary switch](https://www.makerfabs.com/matouch-esp32-s3-rotary-ips-display-with-touch-2-1-st7701.html) screen from Makerfabs using the same ST7701S display. Seems extremely polished and I would love to use this in Home Assistant with ESPhome. A nice presentation [here](https://youtu.be/S43u-U4ds9I?si=x9xtrZ5LurLbWkd7). The clicks it makes on the rotary encoder are so satisfying.