esphome/feature-requests #2412: Restart/Reinitialise SPI action

samuelolteanu, 2023-09-29:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

I'm using a max7219 7 digit display together with a relay and a triac (servo controlled) after that relay for limiting the power of an electric water boiler on the same esp for solar pv maching reasons. The moment relay is closed, the triac will make some noise/spikes/whatever they do and the display sometimes will show nonsense, dimm, not show anything or anything in between. Tried a power supply with two outputs, same thing, and then i gave up and made an HA automation to restart the esp after each time relay is opening the circuit.
Is it possible to restart only the SPI component without wifi disconnect on esphome?
**Additional context**

xdecock, 2023-09-29:
Would it be possible to test something like that?
```
# Script to restart max7219
script:
  - id: restart_max7219
    then:
      - lambda: id(id_of_max7219).setup()

... on_xxxx:
      then:
        script.execute: restart_max7219
```

From what i read i don't see any obvious memory leak in the setup

samuelolteanu, 2023-09-29:
That whas quick... It works. This is my full yaml setup for anyone intersted.

```
external_components:
  source: github://pr#5446
  components: [ spi ]

esphome:
  name: esp32s2-3
  friendly_name: esp32s2-3

esp32:
  board: esp32-s2-saola-1
  framework:
    type: arduino

logger:
  level: warn

api:

ota:

mdns:

time:
  - platform: sntp
    id: sntp_time

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
        static_ip: 192.168.0.205
        gateway: 192.168.0.1
        subnet: 255.255.255.0
        dns1: 1.1.1.1
        dns2: 8.8.8.8
  ap:

captive_portal:

button:
  - platform: safe_mode
    name: "esp32s2-3 Restart (Safe Mode)"

binary_sensor:
  - platform: status
    name: "Connected Status"
    id: ha_connected
    on_release:
      then:
        - switch.turn_off: template_switch

  - platform: homeassistant
    name: "Releu Pompa"
    entity_id: switch.releu_pompa_apa_rece_real
    id: pump_relay

switch:
  - platform: restart
    name: "Restart"

  - platform: gpio
    pin: 8 #25 albs str
    id: ntc_vcc

  - platform: gpio
    pin: 9 # portocaliu
    id: ntc_vcc2
[... 446 lines cut ...]
#     std::string result = buf;
#     it.print(result.c_str());
#     if (display_state == 0) {
#       display_state = 1;
#       delay(500); // Display water pressure for x second
#     } else {
#       display_state = 0;
#       delay(2000); // Display temperature for x second
#     }

#   lambda: |-
#    float first = id (electric_boiler_temp).state;
#    float second = id (solar_boiler_temp).state;
#    char buf [20];
#    sprintf (buf, "%.1fC%.1fC", first, second);
#    std::string result = buf;
#    it.print (result.c_str ());

# Script to restart max7219
script:
  - id: restart_max7219
    then:
      - lambda: id(max7219_aidi).setup();

```

closed 2023-09-29 by samuelolteanu