esphome/feature-requests #2416: Add http_request support to LibreTiny platform

ghost58400, 2023-10-03:
**Describe the problem you have/What new integration you would like**

Add http_request support to LibreTiny platform, in the same way as esp8266

**Please describe your use case for this integration and alternatives you've tried:**

Make any http request with a LibreTiny device

**Additional context**

Since LibreTiny support was added to ESPHome on 2023.9, I would like to use the http_request component on a LibreTiny device.
The HTTPClient library is implemented on LibreTiny : https://docs.libretiny.eu/ltapi/class_h_t_t_p_client/
Since the arduino esp32 implementation of http_request uses the Arduino HTTPClient library too, is it as simple as adding the LibreTiny platform to init.py ?

Lalessus, 2023-10-19:
Seria interessante usar http_request no libretiny.

dsultanr, 2024-11-05:
@tiltec could you please share your config forcing to use the patched code?

tiltec, 2024-11-16:
@dsultanr I'll have a look at it, don't have a working config at the moment.

tiltec, 2025-11-04:
Not really updates, you still need to apply local patches to esphome and libretiny to make it work. I'm not sure how high the chance is to get them merged upstream...