esphome/feature-requests #2417: Add support for GC9503V LCD Driver (Panlee and WirelessTag 86 Box)

tomayr, 2023-10-03:
**Describe the problem you have/What new integration you would like**

Would like to add support for the GC9503V LCD Driver so ESPHome can be installed on all current generation Panlee and Wireless Tag ESP32 3.95" Displays. These look identical to the Sonoff NSpanel Pro.

**Please describe your use case for this integration and alternatives you've tried:**

Would like to use ESPHome as a base for a home automation thermostat with this Panlee panel.

**Additional context**

netweaver1970, 2023-10-27:
+1. I bought 2 of them, by accident. INstead of gathering dust in a box, it would be nice to have it supported.
Thanks.

RcRaCk2k, 2023-12-28:
Yeah i also bought **4inch ESP32 S3 Touchable Screen Durable WiFi 86 Central Control Panel Widely Use** on eBay. This device also have the Display GC9503V on board.

https://www.ebay.de/itm/355184665406

Looking forward for support of that device.

dmit2k, 2024-10-24:
Hello,

any progress with gc9503v support so far? This chip becomes pretty popular driver for inexpensive boards and more of them appear on the market.

It already has Arduino support which can be used as a reference design point as I understand.

clydebarrow, 2024-10-31:
The st7701s driver should work for it, probably just need a custom init sequence.

elik745i, 2025-01-10:
any progress? I got it working but over MQTT through custom sketch in Arduino IDE
https://www.youtube.com/watch?v=e3j2usPWanU

clydebarrow, 2025-01-10:
> through custom sketch in Arduino

So you must have the init sequence - you can configure that in YAML with the ST7701S driver.

GiorgioAresu, 2025-01-16:
Can the init sequences (or the ST7701S component) be debugged in some way or is it just trial and error? This is my configuration, mainly referenced on openhasp as that works great (but the poor homeassistant integration and old lvgl makes it a pain to work with in my setup):
```yaml
display:
  - platform: st7701s
    show_test_card: true
    id: GC9503V
    color_order: RGB
    dimensions:
      width: 480
      height: 480
    reset_pin:
      number: 41 # LCD autoreset, use RGB_VS (TFT_VSYNC) signal to keep working
      allow_other_uses: true
    de_pin: 40
    hsync_pin:
      number: 42
      inverted: false
    vsync_pin:
      number: 41
      inverted: false
      allow_other_uses: true
    pclk_pin: 39
    pclk_inverted: true
    pclk_frequency: 15MHz
    data_pins:
      red:
        - 18         # R1
        - 8          # R2
        - 3          # R3
        - 46         # R4
        - 10         # R5
      green:
        - 14         # G0
        - 13         # G1
        - 12         # G2
        - 11         # G3
        - 16         # G4
        - 17         # G5 
      blue:
        - number: 45 # B1
          allow_other_uses: true
        - number: 48 # B2
          allow_other_uses: true
        - 47         # B3
        - 0          # B4
        - 21         # B5
    # SPI
    cs_pin: 38
    data_rate: 10MHz ## should be 15, not supported
    spi_mode: 0

    hsync_pulse_width: 10
    hsync_front_porch: 8
    hsync_back_porch: 40
    vsync_pulse_width: 10
    vsync_front_porch: 8
    vsync_back_porch: 40
    init_sequence:
      - [ 0xF0, 0x58, 0x05, 0x00 ]
      - [ 0xF1, 0x00, 0x00, 0x00 ]
      - [ 0xF2, 0x00, 0x00, 0x00 ]
      - [ 0xF6, 0x00, 0x20, 0x00 ]
      - [ 0xF9, 0x44, 0x3F, 0x00 ]
      - [ 0xF8, 0x22, 0x22, 0x22 ]
      - [ 0xF3, 0x00, 0x00, 0x00 ]
      - [ 0xF4, 0x00, 0x00, 0x00 ]
      - [ 0xF5, 0x00, 0x00, 0x00 ]
      - [ 0x71, 0x55 ]
      - [ 0x6F, 0x00 ]
      - [ 0x62, 0x01 ]
[... 82 lines cut ...]
      # - [ 0x60, 0x38, 0x08, 0x7A, 0x7A, 0x38, 0x09, 0x7A, 0x7A ]
      # - [ 0x63, 0x31, 0xE4, 0x7A, 0x7A, 0x31, 0xE5, 0x7A, 0x7A ]
      # - [ 0x69, 0x04, 0x22, 0x14, 0x22, 0x14, 0x22, 0x08 ]
      # - [ 0x6B, 0x07 ]
      # - [ 0x7A, 0x08, 0x13 ]
      # - [ 0x7B, 0x08, 0x13 ]
      # - [ 0xD1, 0x00, 0x00, 0x00, 0x04, 0x00, 0x12, 0x00, 0x18, 0x00, 0x21, 0x00, 0x2A, 0x00, 0x35, 0x00, 0x47, 0x00, 0x56, 0x00, 0x90, 0x00, 0xE5, 0x01, 0x68, 0x01, 0xD5, 0x01, 0xD7, 0x02, 0x36, 0x02, 0xA6, 0x02, 0xEE, 0x03, 0x48, 0x03, 0xA0, 0x03, 0xBA, 0x03, 0xC5, 0x03, 0xD0, 0x03, 0xE0, 0x03, 0xEA, 0x03, 0xFA, 0x03, 0xFF ]
      # - [ 0xD2, 0x00, 0x00, 0x00, 0x04, 0x00, 0x12, 0x00, 0x18, 0x00, 0x21, 0x00, 0x2A, 0x00, 0x35, 0x00, 0x47, 0x00, 0x56, 0x00, 0x90, 0x00, 0xE5, 0x01, 0x68, 0x01, 0xD5, 0x01, 0xD7, 0x02, 0x36, 0x02, 0xA6, 0x02, 0xEE, 0x03, 0x48, 0x03, 0xA0, 0x03, 0xBA, 0x03, 0xC5, 0x03, 0xD0, 0x03, 0xE0, 0x03, 0xEA, 0x03, 0xFA, 0x03, 0xFF ]
      # - [ 0xD3, 0x00, 0x00, 0x00, 0x04, 0x00, 0x12, 0x00, 0x18, 0x00, 0x21, 0x00, 0x2A, 0x00, 0x35, 0x00, 0x47, 0x00, 0x56, 0x00, 0x90, 0x00, 0xE5, 0x01, 0x68, 0x01, 0xD5, 0x01, 0xD7, 0x02, 0x36, 0x02, 0xA6, 0x02, 0xEE, 0x03, 0x48, 0x03, 0xA0, 0x03, 0xBA, 0x03, 0xC5, 0x03, 0xD0, 0x03, 0xE0, 0x03, 0xEA, 0x03, 0xFA, 0x03, 0xFF ]
      # - [ 0xD4, 0x00, 0x00, 0x00, 0x04, 0x00, 0x12, 0x00, 0x18, 0x00, 0x21, 0x00, 0x2A, 0x00, 0x35, 0x00, 0x47, 0x00, 0x56, 0x00, 0x90, 0x00, 0xE5, 0x01, 0x68, 0x01, 0xD5, 0x01, 0xD7, 0x02, 0x36, 0x02, 0xA6, 0x02, 0xEE, 0x03, 0x48, 0x03, 0xA0, 0x03, 0xBA, 0x03, 0xC5, 0x03, 0xD0, 0x03, 0xE0, 0x03, 0xEA, 0x03, 0xFA, 0x03, 0xFF ]
      # - [ 0xD5, 0x00, 0x00, 0x00, 0x04, 0x00, 0x12, 0x00, 0x18, 0x00, 0x21, 0x00, 0x2A, 0x00, 0x35, 0x00, 0x47, 0x00, 0x56, 0x00, 0x90, 0x00, 0xE5, 0x01, 0x68, 0x01, 0xD5, 0x01, 0xD7, 0x02, 0x36, 0x02, 0xA6, 0x02, 0xEE, 0x03, 0x48, 0x03, 0xA0, 0x03, 0xBA, 0x03, 0xC5, 0x03, 0xD0, 0x03, 0xE0, 0x03, 0xEA, 0x03, 0xFA, 0x03, 0xFF ]
      # - [ 0xD6, 0x00, 0x00, 0x00, 0x04, 0x00, 0x12, 0x00, 0x18, 0x00, 0x21, 0x00, 0x2A, 0x00, 0x35, 0x00, 0x47, 0x00, 0x56, 0x00, 0x90, 0x00, 0xE5, 0x01, 0x68, 0x01, 0xD5, 0x01, 0xD7, 0x02, 0x36, 0x02, 0xA6, 0x02, 0xEE, 0x03, 0x48, 0x03, 0xA0, 0x03, 0xBA, 0x03, 0xC5, 0x03, 0xD0, 0x03, 0xE0, 0x03, 0xEA, 0x03, 0xFA, 0x03, 0xFF ]
      # - [ 0x3A, 0x66 ]
      # - [ 0x11 ]
      # - delay 200ms
      # - [ 0x29 ]

spi:
  - id: lcd_spi
    clk_pin:
      number: 45
      allow_other_uses: true
    mosi_pin: 
      number: 48
      allow_other_uses: true```

clydebarrow, 2025-01-16:
> is it just trial and error?

Pretty much. First make sure the backlight is on. Leave the SPI data rate at the default (1MHz) - it's only for setup so it doesn't have to be fast. Try SPI mode 3.

GiorgioAresu, 2025-01-18:
That didn´t work either. I made another few tries but without luck. Then I found this project [https://github.com/strange-v/ha_deck](https://github.com/strange-v/ha_deck) which is a custom component that adds another layer on top of the displays themselves but supports this exact model (WT32S3-86S/ZX3D95CE01S) and it works. My understanding, looking a bit at the code, is that the changes required to get it working with the ST7701S are more than just an init sequence (which I also tried). I was trying to make some sort of custom component Frankenstein but couldn´t make it work, I'm too rusty with C and making a display driver as the first custom component seems a bit too much

clydebarrow, 2025-01-19:
Well, if you want any more help, you'll need to show what you've tried so far. I'd suggest providing links to the device documentation, i.e. the schematic if available, the demo code where the GPIO assignments and init code are specified, and the yaml that you have tried so far. Links to any other drivers for the display chip will also be helpful.

GiorgioAresu, 2025-01-19:
Sure, thanks! Here's everything I found so far:

- [Board datasheet](https://github.com/user-attachments/files/18468175/WT32S3-86S.ZX3D95CE01S-TR-4848.Datasheet-V1.5EN.pdf). Note that LCD_SCLK and LCD_MOSI share the same GPIOs as RGB_D0 and RGB_D1 on the LCD.
- [GC9503V datasheet](https://github.com/user-attachments/files/18468172/IC-GC9503V-Specification.pdf)
- ESP-IDF firmware from wireless-tag.com, along with some more info, that I haven´t been able to build:
  - sdkconfig: https://github.com/wireless-tag-com/ZX3D95CE01S-TR-4848/blob/master/sdkconfig
  - pins: https://github.com/wireless-tag-com/ZX3D95CE01S-TR-4848/blob/master/main/board.h
- Panlee docs, which interestingly are missing this exact model: https://doc.panel-tag.com/ESP32-S3/board.html
- OpenHASP discussion on these models, in case it's useful: https://github.com/HASwitchPlate/openHASP/discussions/435. Interesting notes:
  - -AR, -UR, -TR variants should only differ in the additional hardware other than RS485 (but some discussion says otherwise, see the other points): speaker/mic, usb, SHT20 respectively. SHT20 already works in ESPHome
  - WT32-S3-WROVER-N16R8（16MB Flash/8Line 8M die inside)
  - 480 * 480 high-resolution IPS RGB display with capacitive touch. Touch already works with ft63x6 in ESPHome
  - Subthread in the same issue about specifics for the TR, it looks like there are discrepancies in the datasheet vs actual board https://github.com/HASwitchPlate/openHASP/discussions/435#discussioncomment-6612848
- Tasmota issue on the GC9503V with this exact board, they were able to make it work, I've been able to build it but not to get anything from the screen, I'm probably doing something wrong: https://github.com/arendst/Tasmota/issues/19489
- Arduino_GFX GC9503V init sequence: https://github.com/moononournation/Arduino_GFX/blob/9e12ba7f984348d87c698721aa6bbe7074b35f8a/src/display/Arduino_RGB_Display.h#L11
- HA_Deck for this model: https://github.com/strange-v/ha_deck/tree/main/components/hd_device_wt32s3_86s
  - This, I think, should be the most promising, considering it's already based on esphome, and it works with these bits https://github.com/strange-v/ha_deck/blob/main/examples/ha_deck1.yaml#L37-L303 on top of my base device configuration, just changing `hd_device_sc01_plus` to `hd_device_wt32s3_86s` even without the partition scheme.
  - Uses https://github.com/lovyan03/LovyanGFX, init sequence and setup here https://github.com/lovyan03/LovyanGFX/blob/55a0f66d9278faa596c8d51a8e8a3e537dd8f44f/src/lgfx/v1/platforms/esp32s3/Panel_RGB.cpp#L400. My conversion to esphome init sequence is:
```yaml
    # Rest of the yaml from the comment above
    data_rate: 1MHz
    init_sequence:
      #HA_DECK
      - [ 0xF0, 0x55, 0xAA, 0x52, 0x08, 0x00 ]
      - [ 0xF6, 0x5A, 0x87 ]
      - [ 0xC1, 0x3F ]
      - [ 0xC2, 0x0E ]
      - [ 0xC6, 0xF8 ]
      - [ 0xC9, 0x10 ]
      - [ 0xCD, 0x25 ]
      - [ 0xF8, 0x8A ]
      - [ 0xAC, 0x45 ]
      - [ 0xA0, 0xDD ]
      - [ 0xA7, 0x47 ]
      - [ 0xFA, 0x00, 0x00, 0x00, 0x04 ]
      - [ 0xA3, 0xEE ]
      - [ 0xFD, 0x28, 0x28, 0x00 ]
      - [ 0x71, 0x48 ]
      - [ 0x72, 0x48 ]
      - [ 0x73, 0x00, 0x44 ]
      - [ 0x97, 0xEE ]
      - [ 0x83, 0x93 ]
      - [ 0x9A, 0x72 ]
      - [ 0x9B, 0x5a ]
      - [ 0x82, 0x2c, 0x2c ]
      - [ 0xB1, 0x10 ]
      - [ 0x6D, 0x00, 0x1F, 0x19, 0x1A, 0x10, 0x0e, 0x0c, 0x0a, 0x02, 0x07, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x08, 0x01, 0x09, 0x0b, 0x0D, 0x0F, 0x1a, 0x19, 0x1f, 0x00 ]
      - [ 0x64, 0x38, 0x05, 0x01, 0xdb, 0x03, 0x03, 0x38, 0x04, 0x01, 0xdc, 0x03, 0x03, 0x7A, 0x7A, 0x7A, 0x7A ]
      - [ 0x65, 0x38, 0x03, 0x01, 0xdd, 0x03, 0x03, 0x38, 0x02, 0x01, 0xde, 0x03, 0x03, 0x7A, 0x7A, 0x7A, 0x7A ]
      - [ 0x66, 0x38, 0x01, 0x01, 0xdf, 0x03, 0x03, 0x38, 0x00, 0x01, 0xe0, 0x03, 0x03, 0x7A, 0x7A, 0x7A, 0x7A ]
      - [ 0x67, 0x30, 0x01, 0x01, 0xe1, 0x03, 0x03, 0x30, 0x02, 0x01, 0xe2, 0x03, 0x03, 0x7A, 0x7A, 0x7A, 0x7A ]
      - [ 0x68, 0x00, 0x08, 0x15, 0x08, 0x15, 0x7A, 0x7A, 0x08, 0x15, 0x08, 0x15, 0x7A, 0x7A ]
      - [ 0x60, 0x38, 0x08, 0x7A, 0x7A, 0x38, 0x09, 0x7A, 0x7A ]
      - [ 0x63, 0x31, 0xe4, 0x7A, 0x7A, 0x31, 0xe5, 0x7A, 0x7A ]
      - [ 0x6B, 0x07 ]
      - [ 0x7A, 0x08, 0x13 ]
      - [ 0x7B, 0x08, 0x13 ]
      - [ 0xD1, 0x00, 0x00, 0x00, 0x04, 0x00, 0x12, 0x00, 0x18, 0x00, 0x21, 0x00, 0x2a, 0x00, 0x35, 0x00, 0x47, 0x00, 0x56, 0x00, 0x90, 0x00, 0xe5, 0x01, 0x68, 0x01, 0xd5, 0x01, 0xd7, 0x02, 0x36, 0x02, 0xa6, 0x02, 0xee, 0x03, 0x48, 0x03, 0xa0, 0x03, 0xba, 0x03, 0xc5, 0x03, 0xd0, 0x03, 0xE0, 0x03, 0xea, 0x03, 0xFa, 0x03, 0xFF ]
      - [ 0xD2, 0x00, 0x00, 0x00, 0x04, 0x00, 0x12, 0x00, 0x18, 0x00, 0x21, 0x00, 0x2a, 0x00, 0x35, 0x00, 0x47, 0x00, 0x56, 0x00, 0x90, 0x00, 0xe5, 0x01, 0x68, 0x01, 0xd5, 0x01, 0xd7, 0x02, 0x36, 0x02, 0xa6, 0x02, 0xee, 0x03, 0x48, 0x03, 0xa0, 0x03, 0xba, 0x03, 0xc5, 0x03, 0xd0, 0x03, 0xE0, 0x03, 0xea, 0x03, 0xFa, 0x03, 0xFF ]
      - [ 0xD3, 0x00, 0x00, 0x00, 0x04, 0x00, 0x12, 0x00, 0x18, 0x00, 0x21, 0x00, 0x2a, 0x00, 0x35, 0x00, 0x47, 0x00, 0x56, 0x00, 0x90, 0x00, 0xe5, 0x01, 0x68, 0x01, 0xd5, 0x01, 0xd7, 0x02, 0x36, 0x02, 0xa6, 0x02, 0xee, 0x03, 0x48, 0x03, 0xa0, 0x03, 0xba, 0x03, 0xc5, 0x03, 0xd0, 0x03, 0xE0, 0x03, 0xea, 0x03, 0xFa, 0x03, 0xFF ]
      - [ 0xD4, 0x00, 0x00, 0x00, 0x04, 0x00, 0x12, 0x00, 0x18, 0x00, 0x21, 0x00, 0x2a, 0x00, 0x35, 0x00, 0x47, 0x00, 0x56, 0x00, 0x90, 0x00, 0xe5, 0x01, 0x68, 0x01, 0xd5, 0x01, 0xd7, 0x02, 0x36, 0x02, 0xa6, 0x02, 0xee, 0x03, 0x48,  [... cut ...]
      - [ 0xFF ]
```
- Here's an issue on the ESP-IDF repo where a guy was able to get a minimal example of it working https://github.com/espressif/esp-bsp/issues/345 (https://sourceforge.net/p/esp32-s3-86-box-project-base/code/HEAD/tree/trunk/touchscreen/)

A picture of my board:
[image]

I'm basically overwhelmed with all of this and not really sure on how to piece all of this together with the esphome way of doing the external components (or anything really, again way too rusty on my basic C skills)

clydebarrow, 2025-01-19:
```
      # - [ 0x11, CMD_INIT_DELAY, 200 ] // Sleep Out . CMD_INIT_DELAY=0x80, so this should translate to the next line?
      - [ 0x11, 0x80, 200 ]
      - [ 0x29 ]
      # - [ 0xFF, 0xFF ] // This is the edge case where the length would be 0xFF, so is it enough to just send a single 0xFF?
      - [ 0xFF ]

```
Only looked quickly, but this is wrong, it should just be:

```
      - [ 0x11]
      - delay 200
 ```
The 0x29 is not needed, the driver does that. And the FF FF is just an end marker for the table, it's not to be sent. The 0x11 and delay may not be necessary but shouldn't do any harm.

xiasi0, 2025-01-19:
```- [ 0xF0, 0x08, 0x55, 0xAA, 0x52, 0x08, 0x00 ]
- [ 0xF6, 0x5A, 0x87 ]
- [ 0xC1, 0x3F ]
- [ 0xC2, 0x0E ]
- [ 0xC6, 0xF8 ]
- [ 0xC9, 0x10 ]
- [ 0xCD, 0x25 ]
- [ 0xF8, 0x8A ]
- [ 0xAC, 0x45 ]
- [ 0xA0, 0xDD ]
- [ 0xA7, 0x47 ]
- [ 0xFA, 0x00, 0x00, 0x00, 0x04 ]
- [ 0x86, 0x99, 0xa3, 0xa3, 0x51 ]
- [ 0xA3, 0xEE ]
- [ 0xFD, 0x28, 0x28, 0x00 ]
- [ 0x71, 0x48 ]
- [ 0x72, 0x48 ]
- [ 0x73, 0x00, 0x44 ]
- [ 0x97, 0xEE ]
- [ 0x83, 0x93 ]
- [ 0x9A, 0x72 ]
- [ 0x9B, 0x5a ]
- [ 0x82, 0x2c, 0x2c ]
- [ 0xB1, 0x10 ]
- [ 0x6D, 0x00, 0x1F, 0x19, 0x1A, 0x10, 0x0E, 0x0C, 0x0A, 0x02, 0x07, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x08, 0x01, 0x09, 0x0B, 0x0D, 0x0F, 0x1A, 0x19, 0x1F, 0x00 ]
- [ 0x64, 0x38, 0x05, 0x01, 0xDB, 0x03, 0x03, 0x38, 0x04, 0x01, 0xDC, 0x03, 0x03, 0x7A, 0x7A, 0x7A, 0x7A ]
- [ 0x65, 0x38, 0x03, 0x01, 0xDD, 0x03, 0x03, 0x38, 0x02, 0x01, 0xDE, 0x03, 0x03, 0x7A, 0x7A, 0x7A, 0x7A ]
- [ 0x66, 0x38, 0x01, 0x01, 0xDF, 0x03, 0x03, 0x38, 0x00, 0x01, 0xE0, 0x03, 0x03, 0x7A, 0x7A, 0x7A, 0x7A ]
- [ 0x67, 0x30, 0x01, 0x01, 0xE1, 0x03, 0x03, 0x30, 0x02, 0x01, 0xE2, 0x03, 0x03, 0x7A, 0x7A, 0x7A, 0x7A ]
- [ 0x68, 0x00, 0x08, 0x15, 0x08, 0x15, 0x7A, 0x7A, 0x08, 0x15, 0x08, 0x15, 0x7A, 0x7A ]
- [ 0x60, 0x38, 0x08, 0x7A, 0x7A, 0x38, 0x09, 0x7A, 0x7A ]
- [ 0x63, 0x31, 0xE4, 0x7A, 0x7A, 0x31, 0xE5, 0x7A, 0x7A ]
- [ 0x69, 0x04, 0x22, 0x14, 0x22, 0x14, 0x22, 0x08 ]
- [ 0x6B, 0x07 ]
- [ 0x7A, 0x08, 0x13 ]
- [ 0x7B, 0x08, 0x13 ]
- [ 0xD1, 0x00, 0x00, 0x00, 0x04, 0x00, 0x12, 0x00, 0x18, 0x00, 0x21, 0x00, 0x2A, 0x00, 0x35, 0x00, 0x47, 0x00, 0x56, 0x00, 0x90, 0x00, 0xE5, 0x01, 0x68, 0x01, 0xD5, 0x01, 0xD7, 0x02, 0x36, 0x02, 0xA6, 0x02, 0xEE, 0x03, 0x48, 0x03, 0xA0, 0x03, 0xBA, 0x03, 0xC5, 0x03, 0xD0, 0x03, 0xE0, 0x03, 0xEA, 0x03, 0xFA, 0x03, 0xFF ]
- [ 0xD2, 0x00, 0x00, 0x00, 0x04, 0x00, 0x12, 0x00, 0x18, 0x00, 0x21, 0x00, 0x2A, 0x00, 0x35, 0x00, 0x47, 0x00, 0x56, 0x00, 0x90, 0x00, 0xE5, 0x01, 0x68, 0x01, 0xD5, 0x01, 0xD7, 0x02, 0x36, 0x02, 0xA6, 0x02, 0xEE, 0x03, 0x48, 0x03, 0xA0, 0x03, 0xBA, 0x03, 0xC5, 0x03, 0xD0, 0x03, 0xE0, 0x03, 0xEA, 0x03, 0xFA, 0x03, 0xFF ]
- [ 0xD3, 0x00, 0x00, 0x00, 0x04, 0x00, 0x12, 0x00, 0x18, 0x00, 0x21, 0x00, 0x2A, 0x00, 0x35, 0x00, 0x47, 0x00, 0x56, 0x00, 0x90, 0x00, 0xE5, 0x01, 0x68, 0x01, 0xD5, 0x01, 0xD7, 0x02, 0x36, 0x02, 0xA6, 0x02, 0xEE, 0x03, 0x48, 0x03, 0xA0, 0x03, 0xBA, 0x03, 0xC5, 0x03, 0xD0, 0x03, 0xE0, 0x03, 0xEA, 0x03, 0xFA, 0x03, 0xFF ]
- [ 0xD4, 0x00, 0x00, 0x00, 0x04, 0x00, 0x12, 0x00, 0x18, 0x00, 0x21, 0x00, 0x2A, 0x00, 0x35, 0x00, 0x47, 0x00, 0x56, 0x00, 0x90, 0x00, 0xE5, 0x01, 0x68, 0x01, 0xD5, 0x01, 0xD7, 0x02, 0x36, 0x02, 0xA6, 0x02, 0xEE, 0x03, 0x48, 0x03, 0xA0, 0x03, 0xBA, 0x03, 0xC5, 0x03, 0xD0, 0x03, 0xE0, 0x03, 0xEA, 0x03, 0xFA, 0x03, 0xFF ]
- [ 0xD5, 0x00, 0x00, 0x00, 0x04, 0x00, 0x12, 0x00, 0x18, 0x00, 0x21, 0x00, 0x2A, 0x00, 0x35, 0x00, 0x47, 0x00, 0x56, 0x00, 0x90, 0x00, 0xE5, 0x01, 0x68, 0x01, 0xD5, 0x01, 0xD7, 0x02, 0x36, 0x02, 0xA6, 0x02, 0xEE, 0x03, 0x48, 0x03, 0xA0, 0x03, 0xBA, 0x03, 0xC5, 0x03, 0xD0, 0x03, 0xE0, 0x03, 0xEA, 0x03, 0xFA, 0x03, 0xFF ]
- [ 0xD6, 0x00, 0x00, 0x00, 0x04, 0x00, 0x12, 0x00, 0x18, 0x00, 0x21, 0x00, 0x2A, 0x00, 0x35, 0x00, 0x47, 0x00, 0x56, 0x00, 0x90, 0x00, 0xE5, 0x01, 0x68, 0x01, 0xD5, 0x01, 0xD7, 0x02, 0x36, 0x02, 0xA6, 0x02, 0xEE, 0x03, 0x48, 0x03, 0xA0, 0x03, 0xBA, 0x03, 0xC5, 0x03, 0xD0, 0x03, 0xE0, 0x03, 0xEA, 0x03, 0xFA, 0x03, 0xFF ]
- [ 0x3a, 0x66 ]
- [ 0x11, 0x29 ]
```
hi Clyde This is the init_sequence I tried, and there is no output on the screen. It's worth noting that the GPIO is shared.

xiasi0, 2025-01-19:
```output:
  - platform: ledc
    pin: GPIO05
    id: backlight
    frequency: 2000Hz
light:
  - platform: monochromatic
    output: backlight
    name: Backlight
    restore_mode: ALWAYS_ON
i2c:
  - id: bus_a
    sda: GPIO15
    scl: GPIO06
touchscreen:
  - platform: ft63x6
    interrupt_pin: GPIO4
    id: my_touchscreen
    display: my_display
    on_touch:
    - logger.log:
        format: Touch at (%d, %d)
        args: [touch.x, touch.y]
    - lambda: |-
        ESP_LOGI("cal", "x=%d, y=%d, x_raw=%d, y_raw=%0d",
            touch.x,
            touch.y,
            touch.x_raw,
            touch.y_raw
            );
spi:
  - id: lcd_spi
    clk_pin: 
      number: 45
      allow_other_uses: true
    mosi_pin: 
      number: 48
      allow_other_uses: true
display:
  - platform: st7701s
    id: my_display
    update_interval: never
    auto_clear_enabled: true
    spi_mode: MODE3
    data_rate: 1MHz
    color_order: RGB
    invert_colors: False
    show_test_card: true
    dimensions:
      width: 480
      height: 480
    cs_pin: GPIO 38
    de_pin: GPIO 40
    hsync_pin: GPIO 42
    vsync_pin: GPIO 41
    pclk_pin: GPIO 39
    # pclk_frequency: 12MHz
    # pclk_inverted: True
    # hsync_pulse_width: 8
    # hsync_front_porch: 10
    # hsync_back_porch: 20
    # vsync_pulse_width: 8
    # vsync_front_porch: 10
    # vsync_back_porch: 10
    init_sequence:
      - [ 0xF0, 0x55, 0xAA, 0x52, 0x08, 0x00 ]
      - [ 0xF6, 0x5A, 0x87 ]
      - [ 0xC1, 0x3F ]
      - [ 0xC2, 0x0E ]
      - [ 0xC6, 0xF8 ]
      - [ 0xC9, 0x10 ]
      - [ 0xCD, 0x25 ]
      - [ 0xF8, 0x8A ]
      - [ 0xAC, 0x45 ]
      - [ 0xA0, 0xDD ]
      - [ 0xA7, 0x47 ]
      - [ 0xFA, 0x00, 0x00, 0x00, 0x04 ]
      - [ 0xA3, 0xEE ]
      - [ 0xFD, 0x28, 0x28, 0x00 ]
      - [ 0x71, 0x48 ]
      - [ 0x72, 0x48 ]
      - [ 0x73, 0x00, 0x44 ]
      - [ 0x97, 0xEE ]
      - [ 0x83, 0x93 ]
      - [ 0x9A, 0x72 ]
      - [ 0x9B, 0x5a ]
      - [ 0x82, 0x2c, 0x2c ]
      - [ 0xB1, 0x10 ]
      - [ 0x6D, 0x00, 0x1F, 0x19, 0x1A, 0x10, 0x0e, 0x0c, 0x0a, 0x02, 0x07, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x08, 0x01, 0x09, 0x0b, 0x0D, 0x0F, 0x1a, 0x19, 0x1f, 0x00 ]
      - [ 0x64, 0x38, 0x05, 0x01, 0xdb, 0x03, 0x03, 0x38, 0x04, 0x01, 0xdc, 0x03, 0x03, 0x7A, 0x7A, 0x7A, 0x7A ]
      - [ 0x65, 0x38, 0x03, 0x01, 0xdd, 0x03, 0x03, 0x38, 0x02, 0x01, 0xde, 0x03, 0x03, 0x7A, 0x7A, 0x7A, 0x7A ]
      - [ 0x66, 0x38, 0x01, 0x01, 0xdf, 0x03, 0x03, 0x38, 0x00, 0x01, 0xe0, 0x03, 0x03, 0x7A, 0x7A, 0x7A, 0x7A ]
      - [ 0x67, 0x30, 0x01, 0x01, 0xe1, 0x03, 0x03, 0x30, 0x02, 0x01, 0xe2, 0x03, 0x03, 0x7A, 0x7A, 0x7A, 0x7A ]
      - [ 0x68, 0x00, 0x08, 0x15, 0x08, 0x15, 0x7A, 0x7A, 0x08, 0x15, 0x08, 0x15, 0x7A, 0x7A ]
      - [ 0x60, 0x38, 0x08, 0x7A, 0x7A, 0x38, 0x09, 0x7A, 0x7A ]
      - [ 0x63, 0x31, 0xe4, 0x7A, 0x7A, 0x31, 0xe5, 0x7A, 0x7A ]
      - [ 0x6B, 0x07 ]
      - [ 0x7A, 0x08, 0x13 ]
      - [ 0x7B, 0x08, 0x13 ]
      - [ 0xD1, 0x00, 0x00, 0x00, 0x04, 0x00, 0x12, 0x00, 0x18, 0x00, 0x21, 0x00, 0x2a, 0x00, 0x35, 0x00, 0x47, 0x00, 0x56, 0x00, 0x90, 0x00, 0xe5, 0x01, 0x68, 0x01, 0xd5, 0x01, 0xd7, 0x02, 0x36, 0x02, 0xa6, 0x02, 0xee, 0x03, 0x48, 0x03, 0xa0, 0x03, 0xba, 0x03, 0xc5, 0x03, 0xd0, 0x03, 0xE0, 0x03, 0xea, 0x03, 0xFa, 0x03, 0xFF ]
      - [ 0xD2, 0x00, 0x00, 0x00, 0x04, 0x00, 0x12, 0x00, 0x18, 0x00, 0x21, 0x00, 0x2a, 0x00, 0x35, 0x00, 0x47, 0x00, 0x56, 0x00, 0x90, 0x00, 0xe5, 0x01, 0x68, 0x01, 0xd5, 0x01, 0xd7, 0x02, 0x36, 0x02, 0xa6, 0x02, 0xee, 0x03, 0x48, 0x03, 0xa0, 0x03, 0xba, 0x03, 0xc5, 0x03, 0xd0, 0x03, 0xE0, 0x03, 0xea, 0x03, 0xFa, 0x03, 0xFF ]
      - [ 0xD3, 0x00, 0x00, 0x00, 0x04, 0x00, 0x12, 0x00, 0x18, 0x00, 0x21, 0x00, 0x2a, 0x00, 0x35, 0x00, 0x47, 0x00, 0x56, 0x00, 0x90, 0x00, 0xe5, 0x01, 0x68, 0x01, 0xd5, 0x01, 0xd7, 0x02, 0x36, 0x02, 0xa6, 0x02, 0xee, 0x03, 0x48, 0x03, 0xa0, 0x03, 0xba, 0x03, 0xc5, 0x03, 0xd0, 0x03, 0xE0, 0x03, 0xea, 0x03, 0xFa, 0x03, 0xFF ]
      - [ 0xD4, 0x00, 0x00, 0x00, 0x04, 0x00, 0x12, 0x00, 0x18, 0x00, 0x21, 0x00, 0x2a, 0x00, 0x35, 0x00, 0x47, 0x00, 0x56, 0x00, 0x90, 0x00, 0xe5, 0x01, 0x68, 0x01, 0xd5, 0x01, 0xd7, 0x02, 0x36, 0x02, 0xa6, 0x02, 0xee, 0x03, 0x48, 0x03, 0xa0, 0x03, 0xba, 0x03, 0xc5, 0x03, 0xd0, 0x03, 0xE0, 0x03, 0xea, 0x03, 0xFa, 0x03, 0xFF ]
      - [ 0xD5, 0x00, 0x00, 0x00, 0x04, 0x00, 0x12, 0x00, 0x18, 0x00, 0x21, 0x00, 0x2a, 0x00, 0x35, 0x00, 0x47, 0x00, 0x56, 0x00, 0x90, 0x00, 0xe5, 0x01, 0x68, 0x01, 0xd5, 0x01, 0xd7, 0x02, 0x36, 0x02, 0xa6, 0x02, 0xee, 0x03, 0x48, 0x03, 0xa0, 0x03, 0xba, 0x03, 0xc5, 0x03, 0xd0, 0x03, 0xE0, 0x03, 0xea, 0x03, 0xFa, 0x03, 0xFF ]
      - [ 0xD6, 0x00, 0x00, 0x00, 0x04, 0x00, 0x12, 0x00, 0x18, 0x00, 0x21, 0x00, 0x2a, 0x00, 0x35, 0x00, 0x47, 0x00, 0x56, 0x00, 0x90, 0x00, 0xe5, 0x01, 0x68, 0x01, 0xd5, 0x01, 0xd7, 0x02, 0x36, 0x02, 0xa6, 0x02, 0xee, 0x03, 0x48, 0x03, 0xa0, 0x03, 0xba, 0x03, 0xc5, 0x03, 0xd0, 0x03, 0xE0, 0x03, 0xea, 0x03, 0xFa, 0x03, 0xFF ]
      - [ 0x3a, 0x66 ]
    data_pins:
      red:
        - 18
        - 8
        - 3
        - 46
        - 10
      green:
        - 14
        - 13
        - 12
        - 11
        - 16
        - 17
      blue:
        - number: 45
          allow_other_uses: true
        - number: 48
          allow_other_uses: true
        - 47 
        - 0 
        - 21
```
 YAML，The backlight and touch are working properly.

GiorgioAresu, 2025-01-19:
This is the yaml I'm using without success, with the correct init sequence. I checked again all the pin definitions for what should be inverted and what not. I added a reference to where I got that, hopefully I didn´t do exactly the opposite, but I think I might have tried all the combinations by now 😆 
```yaml
esphome:
  platformio_options:
    board_build.arduino.memory_type: qio_opi
    board_build.arduino.psram_type: opi
    board_build.f_flash: 80000000L

psram:
  mode: octal
  speed: 80MHz

logger:
  baud_rate: 115200
  level: VERBOSE
  tx_buffer_size: 1024
  hardware_uart: UART0

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    # type: arduino
    type: esp-idf
    sdkconfig_options:
      # From: https://github.com/wireless-tag-com/ZX3D95CE01S-TR-4848/blob/master/sdkconfig
      CONFIG_SOC_SPI_MEM_SUPPORT_OPI_MODE: "y"
      CONFIG_ESPTOOLPY_FLASH_MODE_AUTO_DETECT: "y"
      CONFIG_ESPTOOLPY_FLASHMODE_QIO: "y"
  flash_size: 16MB

output:
  - platform: ledc
    pin: 5
    id: backlight_led
    frequency: 1kHz

light:
  - platform: monochromatic
    output: backlight_led
    name: Backlight
    id: backlight
    restore_mode: ALWAYS_ON
    default_transition_length: 200ms

sensor:
  - platform: htu21d
    model: sht21
    temperature:
      name: "Temperature"
    humidity:
      name: "Humidity"
    heater:
      name: "Heater"

i2c:
  sda: 15
  scl: 6
  frequency: 400kHz

touchscreen:
  - platform: ft63x6
    interrupt_pin: 4

spi:
  - id: lcd_spi
    clk_pin:
      number: 45
      allow_other_uses: true
    mosi_pin: 
      number: 48
      allow_other_uses: true

color:
  - id: COLOR_CSS_TOMATO
    red:   1.0000
    green: 0.3882
    blue:  0.2784
    white: 0.0000
display:
  - platform: st7701s
    # update_interval: 5s
    show_test_card: true
    lambda: |-
      it.fill(id(COLOR_CSS_TOMATO));
    id: GC9503V
    color_order: RGB
    dimensions:
      width: 480
      height: 480
    reset_pin:
      number: 41 # LCD autoreset, use RGB_VS (TFT_VSYNC) signal to keep working
      inverted: true
      allow_other_uses: true
    de_pin:
      number: 40
      inverted: false # https://github.com/strange-v/ha_deck/blob/main/components/hd_device_wt32s3_86s/LGFX.hpp#L115
    hsync_pin:
      number: 42
      inverted: true # hsync_polarity = 1; Means idle high. https://github.com/strange-v/ha_deck/blob/main/components/hd_device_wt32s3_86s/LGFX.hpp#L106
    vsync_pin:
      number: 41
      inverted: true # vsync_polarity = 1; Means idle high. https://github.com/strange-v/ha_deck/blob/main/components/hd_device_wt32s3_86s/LGFX.hpp#L110
      allow_other_uses: true
    pclk_pin: 39
    pclk_inverted: false # https://github.com/strange-v/ha_deck/blob/main/components/hd_device_wt32s3_86s/LGFX.hpp#L116
    pclk_frequency: 15MHz
    data_pins:
      red:
        - 18         # R1
        - 8          # R2
        - 3          # R3
        - 46         # R4
        - 10         # R5
      green:
        - 14         # G0
        - 13         # G1
        - 12         # G2
        - 11         # G3
        - 16         # G4
        - 17         # G5 
      blue:
        - number: 45 # B1
          allow_other_uses: true
        - number: 48 # B2
          allow_other_uses: true
        - 47         # B3
        - 0          # B4
        - 21         # B5

    # SPI
    cs_pin:
      number: 38
      inverted: true # Active low? https://github.com/lovyan03/LovyanGFX/blob/55a0f66d9278faa596c8d51a8e8a3e537dd8f44f/src/lgfx/v1/platforms/esp32s3/Panel_RGB.cpp#L541-L548
    data_rate: 1MHz # Could be up to 15
    spi_mode: 0 # Idle low. https://github.com/strange-v/ha_deck/blob/main/components/hd_device_wt32s3_86s/LGFX.hpp#L114

    hsync_pulse_width: 10
    hsync_front_porch: 8
    hsync_back_porch: 40
    vsync_pulse_width: 10
    vsync_front_porch: 8
    vsync_back_porch: 40
    init_sequence:
      #HA_DECK
      - [ 0xF0, 0x55, 0xAA, 0x52, 0x08, 0x00 ]
      - [ 0xF6, 0x5A, 0x87 ]
      - [ 0xC1, 0x3F ]
      - [ 0xC2, 0x0E ]
      - [ 0xC6, 0xF8 ]
      - [ 0xC9, 0x10 ]
      - [ 0xCD, 0x25 ]
      - [ 0xF8, 0x8A ]
      - [ 0xAC, 0x45 ]
      - [ 0xA0, 0xDD ]
      - [ 0xA7, 0x47 ]
      - [ 0xFA, 0x00, 0x00, 0x00, 0x04 ]
      - [ 0xA3, 0xEE ]
      - [ 0xFD, 0x28, 0x28, 0x00 ]
      - [ 0x71, 0x48 ]
      - [ 0x72, 0x48 ]
      - [ 0x73, 0x00, 0x44 ]
      - [ 0x97, 0xEE ]
      - [ 0x83, 0x93 ]
      - [ 0x9A, 0x72 ]
      - [ 0x9B, 0x5a ]
      - [ 0x82, 0x2c, 0x2c ]
      - [ 0xB1, 0x10 ]
      - [ 0x6D, 0x00, 0x1F, 0x19, 0x1A, 0x10, 0x0e, 0x0c, 0x0a, 0x02, 0x07, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x08, 0x01, 0x09, 0x0b, 0x0D, 0x0F, 0x1a, 0x19, 0x1f, 0x00 ]
      - [ 0x64, 0x38, 0x05, 0x01, 0xdb, 0x03, 0x03, 0x38, 0x04, 0x01, 0xdc, 0x03, 0x03, 0x7A, 0x7A, 0x7A, 0x7A ]
      - [ 0x65, 0x38, 0x03, 0x01, 0xdd, 0x03, 0x03, 0x38, 0x02, 0x01, 0xde, 0x03, 0x03, 0x7A, 0x7A, 0x7A, 0x7A ]
      - [ 0x66, 0x38, 0x01, 0x01, 0xdf, 0x03, 0x03, 0x38, 0x00, 0x01, 0xe0, 0x03, 0x03, 0x7A, 0x7A, 0x7A, 0x7A ]
      - [ 0x67, 0x30, 0x01, 0x01, 0xe1, 0x03, 0x03, 0x30, 0x02, 0x01, 0xe2, 0x03, 0x03, 0x7A, 0x7A, 0x7A, 0x7A ]
      - [ 0x68, 0x00, 0x08, 0x15, 0x08, 0x15, 0x7A, 0x7A, 0x08, 0x15, 0x08, 0x15, 0x7A, 0x7A ]
      - [ 0x60, 0x38, 0x08, 0x7A, 0x7A, 0x38, 0x09, 0x7A, 0x7A ]
      - [ 0x63, 0x31, 0xe4, 0x7A, 0x7A, 0x31, 0xe5, 0x7A, 0x7A ]
      - [ 0x6B, 0x07 ]
```
Isn´t the ST7701S driver adding more stuff to the init sequence that shouldn't be there for the GC9503V? https://github.com/esphome/esphome/blob/dev/esphome/components/st7701s/st7701s.cpp#L163-L177
(but obviously it's not that easy, still doesn't work removing those lines)

clydebarrow, 2025-01-20:
Again, only a quick look, but remove the inverted GPIO pin specifications. The only one that may be inverted is PCLK and if so you specify that differently. CS is definitely not inverted (it's active low, but the SPI driver handles that.)

Remove the reset pin from the display - it's not connected.

clydebarrow, 2025-01-20:
Add this to the init sequence

```
     - [0xB0, 0]
 ```

That sets the signal polarities. It should be the power on default, but won't hurt.

GiorgioAresu, 2025-01-20:
```yaml
display:
  - platform: st7701s
    # update_interval: 5s
    show_test_card: true
    lambda: |-
      it.fill(id(COLOR_CSS_TOMATO));
    id: GC9503V
    color_order: RGB
    dimensions:
      width: 480
      height: 480
    # reset_pin:
    #   number: 41 # LCD autoreset, use RGB_VS (TFT_VSYNC) signal to keep working
    #   # inverted: true
    #   allow_other_uses: true
    de_pin:
      number: 40
      # inverted: false # https://github.com/strange-v/ha_deck/blob/main/components/hd_device_wt32s3_86s/LGFX.hpp#L115
    hsync_pin:
      number: 42
      # inverted: true # hsync_polarity = 1; Means idle high. https://github.com/strange-v/ha_deck/blob/main/components/hd_device_wt32s3_86s/LGFX.hpp#L106
    vsync_pin:
      number: 41
      # inverted: true # vsync_polarity = 1; Means idle high. https://github.com/strange-v/ha_deck/blob/main/components/hd_device_wt32s3_86s/LGFX.hpp#L110
      # allow_other_uses: true
    pclk_pin: 39
    pclk_inverted: false # https://github.com/strange-v/ha_deck/blob/main/components/hd_device_wt32s3_86s/LGFX.hpp#L116
    pclk_frequency: 15MHz
    data_pins:
      red:
        - 18         # R1
        - 8          # R2
        - 3          # R3
        - 46         # R4
        - 10         # R5
      green:
        - 14         # G0
        - 13         # G1
        - 12         # G2
        - 11         # G3
        - 16         # G4
        - 17         # G5 
      blue:
        - number: 45 # B1
          allow_other_uses: true
        - number: 48 # B2
          allow_other_uses: true
        - 47         # B3
        - 0          # B4
        - 21         # B5

    # SPI
    cs_pin:
      number: 38
      # inverted: true # Active low? https://github.com/lovyan03/LovyanGFX/blob/55a0f66d9278faa596c8d51a8e8a3e537dd8f44f/src/lgfx/v1/platforms/esp32s3/Panel_RGB.cpp#L541-L548
    data_rate: 1MHz ## should be 15, not supported
    spi_mode: 0 # Idle low. https://github.com/strange-v/ha_deck/blob/main/components/hd_device_wt32s3_86s/LGFX.hpp#L114

    hsync_pulse_width: 10
    hsync_front_porch: 8
    hsync_back_porch: 40
    vsync_pulse_width: 10
    vsync_front_porch: 8
    vsync_back_porch: 40
    init_sequence:
      #HA_DECK
      - [ 0xF0, 0x55, 0xAA, 0x52, 0x08, 0x00 ]
      - [ 0xF6, 0x5A, 0x87 ]
      - [ 0xC1, 0x3F ]
[... 43 lines cut ...]
This is the log output for the display section. I see the data pins order is mixed up by the st7701s driver, does the gc9503v need the same treatment?
```
[08:08:15][C][:183]: ST7701S RGB LCD
[08:08:15][C][display.st7701s:184]:   Height: 480
[08:08:15][C][display.st7701s:185]:   Width: 480
[08:08:15][C][display.st7701s:186]:   CS Pin: GPIO38
[08:08:15][C][display.st7701s:188]:   DE Pin: GPIO40
[08:08:15][C][display.st7701s:192]:   Data pin 0: GPIO11
[08:08:15][C][display.st7701s:192]:   Data pin 1: GPIO16
[08:08:15][C][display.st7701s:192]:   Data pin 2: GPIO17
[08:08:15][C][display.st7701s:192]:   Data pin 3: GPIO18
[08:08:15][C][display.st7701s:192]:   Data pin 4: GPIO8
[08:08:15][C][display.st7701s:192]:   Data pin 5: GPIO3
[08:08:15][C][display.st7701s:192]:   Data pin 6: GPIO46
[08:08:15][C][display.st7701s:192]:   Data pin 7: GPIO10
[08:08:15][C][display.st7701s:192]:   Data pin 8: GPIO45
[08:08:15][C][display.st7701s:192]:   Data pin 9: GPIO48
[08:08:15][C][display.st7701s:192]:   Data pin 10: GPIO47
[08:08:15][C][display.st7701s:192]:   Data pin 11: GPIO0
[08:08:15][C][display.st7701s:192]:   Data pin 12: GPIO21
[08:08:15][C][display.st7701s:192]:   Data pin 13: GPIO14
[08:08:15][C][display.st7701s:192]:   Data pin 14: GPIO13
[08:08:15][C][display.st7701s:192]:   Data pin 15: GPIO12
[08:08:15][C][display.st7701s:193]:   SPI Data rate: 1MHz
```

clydebarrow, 2025-01-20:
> I see the data pins order is mixed up by the st7701s driver,

I suspect you have the data pins in the wrong order, but that will only affect the color, you should still see an image of some sort. though it could be very dark.

clydebarrow, 2025-01-20:
Another point - I believe there are multiple hardware variants of the board. It would pay to double check that you are using the right data for the board you have. Also have you actually compiled any other firmware and got a display?

GiorgioAresu, 2025-01-20:
The backlight is at full brightness, so I think even if dark, I should be able to see something.
Yes, the ones that work are HA_Deck (I'll copy from above so you don´t have to hunt):
- https://github.com/strange-v/ha_deck/tree/main/components/hd_device_wt32s3_86s
- Uses https://github.com/lovyan03/LovyanGFX, init sequence and setup here https://github.com/lovyan03/LovyanGFX/blob/55a0f66d9278faa596c8d51a8e8a3e537dd8f44f/src/lgfx/v1/platforms/esp32s3/Panel_RGB.cpp#L400.
And OpenHASP:
- Board definition https://github.com/HASwitchPlate/openHASP/blob/master/user_setups/esp32s3/zw3d95ce01s_ar.ini#L135
- TFT library instantiation https://github.com/HASwitchPlate/openHASP/blob/74063e285feec73adda373f4be0edbc366e123f7/src/drv/tft/tft_driver_arduinogfx.cpp#L86

creeloper27, 2025-01-26:
To add to @GiorgioAresu working firmware list, I am currently experimenting with using this display using Tasmota firmware (using this already previously mentioned Issue as a how to https://github.com/arendst/Tasmota/issues/19489) and it works almost perfectly, there are some weird wobbly-like artifacts around the image though that I am trying to understand the cause of.

sEbola76, 2025-01-27:
After a few days of struggling with the code, I managed to achieve a small success.
[image]
However, the method on the ST7701s component did not work.
I created an external component based on the source code from: https://github.com/esphome/esphome/tree/dev/esphome/components/rpi_dpi_rgb and started testing various initialization codes sent via SPI.
The one from: https://github.com/wireless-tag-com/ZX3D95CE01S-TR-4848/blob/master/main/screen.c works.
The another one from @GiorgioAresu  works after changing the last value from 0xB0 to 0x29.
But... esphome lvgl still doesn't work (black screen, only for a while after the reset I see boot image), I suspect that the problem is in the control of the LCD_CS (38) [active low] pin and LCD_RST [active low - must be high to work, the high state is obtained from RGB_VS (41) through the RC filter].

GiorgioAresu, 2025-01-28:
That's a good step forward!
Did you try setting the `inverted` properties where appropriate, like in https://github.com/esphome/feature-requests/issues/2417#issuecomment-2601053105?

sEbola76, 2025-01-29:
> That's a good step forward! Did you try setting the `inverted` properties where appropriate, like in [#2417 (comment)](https://github.com/esphome/feature-requests/issues/2417#issuecomment-2601053105)?

Check my code: https://github.com/sEbola76/gc9503v
I also have tested on second fresh module  from Panlee, and everything is working.

[image]

clydebarrow, 2025-01-29:
> esphome lvgl still doesn't work (black screen, only for a while after the reset I see boot image),

This turns out to be caused by a recent change that was meant to automate the setting of `auto_clear_enabled` but also had the effect that explicitly setting it to `false` resulted in it being `true`. Remove that setting and it should solve the issue. I have raised a PR to fix it.

klattimer, 2025-02-09:
Bump: These viewesmart screen dial knobs use gc9503cv - https://github.com/VIEWESMART/UEDX48480021-MD80ESP32_2.1inch-Knob

sergeynik77, 2025-03-04:
https://github.com/VIEWESMART/UEDX48480021-MD80ESP32_2.1inch-Knob
https://www.youtube.com/shorts/-kkFuuyLGPM

reading this thread, launched the 2.1 knob display

sergeynik77, 2025-03-05:
edit   https://github.com/sEbola76/gc9503v

void GC9503V::rgb_driver_init()
{
	__spi_send_cmd (0xF0); __spi_send_data (0x55); __spi_send_data (0xAA); __spi_send_data (0x52); __spi_send_data (0x08); __spi_send_data (0x00);
	__spi_send_cmd (0xF6); __spi_send_data (0x5A); __spi_send_data (0x87);
        __spi_send_cmd (0xC1); __spi_send_data (0x3F);
//	__spi_send_cmd (0xC2); __spi_send_data (0x0E);
//	__spi_send_cmd (0xC6); __spi_send_data (0xF8);
	__spi_send_cmd (0xC9); __spi_send_data (0x10);
	__spi_send_cmd (0xCD); __spi_send_data (0x25);
	__spi_send_cmd (0xF8); __spi_send_data (0x8A);
	__spi_send_cmd (0xAC); __spi_send_data (0x45);
	__spi_send_cmd (0xA7); __spi_send_data (0x47);
	__spi_send_cmd (0xA0); __spi_send_data (0xDD);

	__spi_send_cmd (0x87); __spi_send_data (0x04); __spi_send_data (0x03); __spi_send_data (0x66);
	__spi_send_cmd (0x86); __spi_send_data (0x99); __spi_send_data (0xa3); __spi_send_data (0xa3); __spi_send_data (0x51);
	__spi_send_cmd (0xFA); __spi_send_data (0x00); __spi_send_data (0x00); __spi_send_data (0x00); __spi_send_data (0x04);

	__spi_send_cmd (0x9A); __spi_send_data (0x8a);
	__spi_send_cmd (0x9B); __spi_send_data (0x62);
	__spi_send_cmd (0x82); __spi_send_data (0x48); __spi_send_data (0x48);
	__spi_send_cmd (0xB1); __spi_send_data (0x10);
	__spi_send_cmd (0x7a); __spi_send_data (0x13); __spi_send_data (0x1a);
	__spi_send_cmd (0x7b); __spi_send_data (0x13); __spi_send_data (0x1a);

                                                                              	
	__spi_send_cmd (0x6D); __spi_send_data (0x1e); __spi_send_data (0x1e); __spi_send_data (0x1e); __spi_send_data (0x1e);
		__spi_send_data (0x1e); __spi_send_data (0x1e); __spi_send_data (0x02); __spi_send_data (0x0b);
		__spi_send_data (0x01); __spi_send_data (0x00); __spi_send_data (0x1f); __spi_send_data (0x1E);
		__spi_send_data (0x09); __spi_send_data (0x0f); __spi_send_data (0x1E); __spi_send_data (0x1E);
		__spi_send_data (0x1E); __spi_send_data (0x1E); __spi_send_data (0x10); __spi_send_data (0x0a);
		__spi_send_data (0x1E); __spi_send_data (0x1f); __spi_send_data (0x00); __spi_send_data (0x08);
		__spi_send_data (0x0b); __spi_send_data (0x02); __spi_send_data (0x1e); __spi_send_data (0x1e);
		__spi_send_data (0x1e); __spi_send_data (0x1e); __spi_send_data (0x1e); __spi_send_data (0x1e);
	
	__spi_send_cmd (0x64); __spi_send_data (0x18); __spi_send_data (0x07); __spi_send_data (0x01); __spi_send_data (0xe7);
		__spi_send_data (0x03); __spi_send_data (0x03); __spi_send_data (0x18); __spi_send_data (0x06);
		__spi_send_data (0x01); __spi_send_data (0xe6); __spi_send_data (0x03); __spi_send_data (0x03);
		__spi_send_data (0x7A); __spi_send_data (0x7A);	__spi_send_data (0x7A); __spi_send_data (0x7A);
	
	__spi_send_cmd (0x65); __spi_send_data (0x58); __spi_send_data (0x26); __spi_send_data (0x18); __spi_send_data (0x2c);
		__spi_send_data (0x03); __spi_send_data (0x03); __spi_send_data (0x58); __spi_send_data (0x26);
		__spi_send_data (0x18); __spi_send_data (0x2c); __spi_send_data (0x03); __spi_send_data (0x03);
		__spi_send_data (0x7A); __spi_send_data (0x7A); __spi_send_data (0x7A); __spi_send_data (0x7A);
	
	__spi_send_cmd (0x66); __spi_send_data (0x58); __spi_send_data (0x26); __spi_send_data (0x18); __spi_send_data (0x2c);
		__spi_send_data (0x03); __spi_send_data (0x03); __spi_send_data (0x58); __spi_send_data (0x26);
		__spi_send_data (0x18); __spi_send_data (0x2c); __spi_send_data (0x03); __spi_send_data (0x03);
		__spi_send_data (0x7A); __spi_send_data (0x7A); __spi_send_data (0x7A); __spi_send_data (0x7A);
	
	__spi_send_cmd (0x67); __spi_send_data (0x18); __spi_send_data (0x05); __spi_send_data (0x01); __spi_send_data (0xe5);
		__spi_send_data (0x03); __spi_send_data (0x03); __spi_send_data (0x18); __spi_send_data (0x04);
		__spi_send_data (0x01); __spi_send_data (0xe4); __spi_send_data (0x03); __spi_send_data (0x03);
		__spi_send_data (0x7A); __spi_send_data (0x7A); __spi_send_data (0x7A); __spi_send_data (0x7A);
	
//	__spi_send_cmd (0x68); __spi_send_data (0x00); __spi_send_data (0x08); __spi_send_data (0x15); __spi_send_data (0x08);
//		__spi_send_data (0x15); __spi_send_data (0x7A); __spi_send_data (0x7A); __spi_send_data (0x08);
//		__spi_send_data (0x15); __spi_send_data (0x08); __spi_send_data (0x15); __spi_send_data (0x7A);
//		__spi_send_data (0x7A);
	
	__spi_send_cmd (0x60); __spi_send_data (0x18); __spi_send_data (0x09); __spi_send_data (0x7A); __spi_send_data (0x7A);
		__spi_send_data (0x51); __spi_send_data (0xf1); __spi_send_data (0x7A); __spi_send_data (0x7A);
	__spi_send_cmd (0x63); __spi_send_data (0x51); __spi_send_data (0xf1); __spi_send_data (0x7A); __spi_send_data (0x7A);
		__spi_send_data (0x18); __spi_send_data (0x08); __spi_send_data (0x7A); __spi_send_data (0x7A);
//	__spi_send_cmd (0x69); __spi_send_data (0x04); __spi_send_data (0x22); __spi_send_data (0x14); __spi_send_data (0x22);
//		__spi_send_data (0x14); __spi_send_data (0x22); __spi_send_data (0x08);
	
//	__spi_send_cmd (0x6B); __spi_send_data (0x07);
[... 145 lines cut ...]
      green:
        - 21         # G0
        - 47         # G1
        - 48         # G2
        - 45         # G3
        - 38         # G4
        - 39         # G5 
      blue:
        - 10 # B1
        - 11 # B2
        - number: 12
          allow_other_uses: true
        - number: 13
          allow_other_uses: true              
        - 14         # B5

    enable_pin:  
      number: 18
      inverted: false
    sclk_pin:
      number: 13
      allow_other_uses: true
    mosi_pin: 
      number: 12
      allow_other_uses: true

klattimer, 2025-03-05:
Kinda hard to copy and paste all that but I think I managed it. Still not working for me, I get a few faded coloured stripes on the display and that's all.

clydebarrow, 2025-12-02:
Needs a PR raised to add this to the `mipi_rgb` component.

xiasi0, 2025-12-14:
Is this model available in esphome now?

noki88, 2026-01-12:
I am using 5 Smart Panlee WT32S3-86S (ZX3D95CE01S) displays. 
With this implementation, 2 out of 5 remain black, 2 work as expected, and 1 flickers.
Commenting out the following line makes all displays work correctly:
`config.timings.flags.pclk_active_neg = this->pclk_inverted_;`