esphome/feature-requests #2418: Disable/enable BLE

Rapsssito, 2023-10-03:
**Describe the problem you have/What new integration you would like**

I would like to have a `ble.enable` and `ble.disable` actions like the existing `wifi.enable` and `wifi.disable`. Along with this, I would like to have a `esp32_improv` parameter to select if the service should start at boot or not.

**Please describe your use case for this integration and alternatives you've tried:**

I am currently working on a project with very high energy limitations. I use `esp32_improv` for connecting the device to the WiFi network for the first time. However, the `esp32_improv` service starts at boot and does not stop (never). In my use case, the BLE server that `esp32_improv` creates becomes useless once the device has connected to a WiFi network. For this reason, I would like to be able to turn it off and on on demand with actions like `ble.enable` and `ble.disable`.

**Additional context**

I am using an ESP32-C3.

ChristophCaina, 2023-10-14:
I would request the same / similar thing ... in addition with the Bluetooth-Proxy...
It would be great, if you could either enable / disable bluetooth in general, or specific parts, like the BLE Tracker, or Bluetooth Proxy... etc.

I do have a device, which does only need to act as a BT Proxy for some amount of time... and then, for the rest of the year, I would turn that feature off, because I just don't need it until the next year.

closed 2023-11-06 by jesserockz

cyberhuman, 2023-11-08:
ref: esphome/issues#5087