esphome/feature-requests #2419: esphome commandline - update all

Stofakiller2, 2023-10-04:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

Stofakiller2, 2023-10-04:
Hi,
I run esphome as LXC, and i use the GUI everytime there is an update of espHome, i can see there is a commandline tool for updating a single esp, but through the GUI an "update all" function, but if my browser is closed, sllep etc, it stops compiling. I cant see there is a command line function for compiling all my esp's... Is that even posiible?

Regards,
Jan

ssieb, 2023-10-04:
If you have the command line, it would be simple to script running through all the yaml files and updating them.

Stofakiller2, 2023-10-04:
Yes i know, but then its up to me to maintain that file, and there is an
update all in the GUI, why not in the command line?

Den ons. 4. okt. 2023 kl. 08.40 skrev Samuel Sieb ***@***.***

nagyrobi, 2023-10-04:
Submit the file as PR, add some doc, and community will take over!