esphome/feature-requests #2420: ESPHome multiple devices

bzumik1, 2023-10-05:
**Describe the problem you have/What new integration you would like**
 
I would like to ask if it would be possible to implement option to show multiple devices in HA for one ESP board. This would be useful for ESPHome hubs, bus connected devices etc. One concrete example is [SAMSUNG AC unit controller](https://github.com/lanwin/esphome_samsung_ac). There ESP board is connected to communication bus and it reads data from indoor and outdoor units. As you can imagine it becomes messy when you have 6 indoor units. Then all the data are mixed under one device. It would be perfect if we could create device for each indoor unit.

**Please describe your use case for this integration and alternatives you've tried:**

I am not avare of any alternative how to not mix data from all the devices connected to hub/communication bus. I have seen this was already discussed [here](https://github.com/esphome/feature-requests/issues/1714) and closed as not usefull, but that was because previously it was for bluetooth devices and it was replaced with bluetooth proxy.

**Additional context**

bzumik1, 2023-10-23:
Hi, could some check this please?

mbo18, 2023-12-12:
Maybe see here: https://github.com/esphome/feature-requests/issues/1335

bzumik1, 2024-02-07:
Thanks for the link!

nagyrobi, 2024-04-02:
Duplicate of https://github.com/esphome/feature-requests/issues/1335

closed 2024-04-02 by nagyrobi