esphome/feature-requests #2421: home assistant addon multihomed network with dedicated wifi network support

mcarbonneaux, 2023-10-08:
**Describe the problem you have/What new integration you would like**

on home assitant OS you can have multiple network configured. but when y a add specifique wifi network to home assistant os to be abel to communique to my esphome device with home-assistant-addon, esphome ha addons are not abel to see my device...

**Please describe your use case for this integration and alternatives you've tried:**

to be abel to use dedicated wifi network with esphome device with multihomed network (wifi or other physical network) homeassistant os. 

to avoid that esphome device can acces to my private network if is compromised...

**Additional context**

nagyrobi, 2023-10-08:
You can achieve what you want, by instead of connecting HA to multiple networks directly, put a proper routing platform between them.

Use things like OpenWRT, or PfSense.

mcarbonneaux, 2023-10-08:
> You can achieve what you want, by instead of connecting HA to multiple networks directly, put a proper routing platform between them.
> 
> Use things like OpenWRT, or PfSense.

i use asus router with mesh and is not abel to do advanced routing...
the only way of connecting my home assistant to the guest network is to connect my homme assistant to this wifi network.

nagyrobi, 2023-10-08:
Your Asus router's place is in the trashcan, if you want such features. Use a proper routing plaftorm with normal Access Points.

This is not something that should be resolved by ESPHome or HA, sorry.

closed 2023-10-08 by nagyrobi

mcarbonneaux, 2023-10-09:
is not only with asus router, but all home router...

the only need is to be abel to seed device not only on the principal network of ha (ha os support multiple network) but on all connected network. 

i think in esphome as somethink that force only the discovery on the principal network.

ssieb, 2023-10-09:
Can HA see the device using the name?
A simple workaround for the dashboard would be to add the `use_address:` option.

mcarbonneaux, 2023-10-11:
> Can HA see the device using the name?
i think is related to that...

> A simple workaround for the dashboard would be to add the `use_address:` option.
i've try to use ping in place of mdns but without success...
i while try that (useing `use_address:`) !
i think is just discovery problem...