esphome/feature-requests #2427: Add support for Davicom DM9051 SPI Ethernet Controller

tudstudent, 2023-10-12:
**Describe the problem you have/What new integration you would like**
Introduce support for the Davicom DM9051 SPI Ethernet Controller

**Please describe your use case for this integration and alternatives you've tried:**
The ETH01 Evo modules are equipped with this "new" ETH chip, enabling POE usage, however device not currently supported in ESPHome.

**Additional context**
[](url)
[Davicom Website - Product Page](https://www.davicom.com.tw/production-item.php)
[Espreff Description on how to implement](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_eth.html)

Unfortunately I am not a programmer, will be able to test, but not to make pull requests in this topic, therefore asking for implementation to this community.

tudstudent, 2023-10-26:
Additionally:
[Driver source code](https://github.com/espressif/esp-idf/blob/master/components/esp_eth/src/esp_eth_mac_dm9051.c)

cleymaba, 2023-11-03:
Support for DM9051 was added to Arduino ESP32 Core version 3.0.0

[Espressif Announcement](https://blog.espressif.com/announcing-the-arduino-esp32-core-version-3-0-0-3bf9f24e20d4)

khabbar-a, 2023-12-12:
I bought 5 units on aliexpres, thinking they were compatible with esphome, someone found a way to make them work

khabbar-a, 2023-12-12:
I bought 5 eth01-evo units on Aliexpress that mount this chip thinking they were compatible with esphome but I wasn't able to make them work, does anyone have a solution thanks

tudstudent, 2023-12-17:
The EVO work perfectly, EXCEPT the physical ETH01 port.
Unfortunately as you have seen my request this is not (yet) supported in ESPHome.

All other functionalities work fine.

I suppose you have them working by WiFi?

d3rsch0rsch, 2023-12-25:
+1 for this requests, as the eth01-evo ist by far the cheapest solution for PoE enabled ESP32.

skinkie, 2023-12-28:
I have added an issue to the eth01-evo repo with full documentation **and** an example to get wired ethernet working.
https://github.com/AI6YP/eth01-evo/issues/2

tube0013, 2024-04-16:
With the W5500 support added which is also based on an SPI implementation, should this be too hard to add now?

Imanol82, 2024-05-21:
Any update?
Thanks!

sabaatworld, 2024-05-22:
I purchased a bunch of these boards, just to realize that it's not supported. It would be great if this board could be added to ESPHome.

bmork, 2024-06-04:
I started looking at this, with partial success.  WiP code (expect lots of force-pushing and buggy commits!) is here:
https://github.com/bmork/esphome/tree/dm9051

Not sure I like the current ethernet SPI solution.  Will probably try to change that into using the spi bus component.  Doesn't seem impossible.  I had to drop the only W5500 specific part (SPI framing) anyway, since the DM9051 uses different framing.  Both drivers will auto-configure the format anyway as long as it's unset, so why bother?  Letting the driver DTRT looks much better than moving that knowlegde into esphome.

But first task is to make RX more stable.  It's pretty unusable as-is.

bmork, 2024-06-06:
So I realized why the current code didn't use the esphome spi component: We need commit ecd1759a0ba6 ("feat(esp_eth): abstraction of SPI driver for SPI Ethernet modules") in esp-idf for that, and it was added as recently as v5.2

I did a quick hack, but it is not exactly ready to merge... It boots and sort of works. But it's easy to make it fall over with an
> spi_master: spi_device_acquire_bus(1031): Cannot acquire bus when a polling transaction is in progress.

error by simply flood pinging the device.  I will leave it there until we both have a use case and esp-idf v5.2 or higher is recommended.  The current PR is fine.

My experimental hack is here if anyone is interested: https://github.com/bmork/esphome/tree/dm9051-spidev-experimental

Imanol82, 2024-09-05:
Hello!
I was very excited to see that you had managed to get the ETH01-EVO working but I can't... Is it because it hasn't been officially uploaded yet or something like that?

[image]

But the result is this one:

[image]

What am I doing wrong?

Thank you very much!

randybb, 2024-09-05:
@Imanol82 It was not merged yet, this is the reason why it is not working. The PR was linked, but here is a direct link to the config https://github.com/esphome/esphome/pull/6861#issuecomment-2276027979

Imanol82, 2024-09-05:
Well, I'll have to wait for it to come out officially (because that doesn't work for me). 
It looks like they are making a lot of progress :)

Thank you!

randybb, 2024-09-05:
You are missing the external component part - which will pull the component from that PR.

Imanol82, 2024-09-05:
Sorry. 
I'm a bit beginner. I don't understand what you're telling me.

Thank you very much!!!

randybb, 2024-09-05:
This part :)
```yaml
external_components:
  - source: github://pr#6861
    components:
      - ethernet
```

Imanol82, 2024-09-05:
Yes, now it is compiling

Thank you very much!

Imanol82, 2024-09-05:
I win hahahaha

Imanol82, 2024-11-03:
Has something happened with the new ESPHome update?

I get this error when trying to update the ETH01-EVO:

[image]

Thank you!

bmork, 2024-11-04:
@Imanol82 - the PR was updated to apply on top of
https://github.com/esphome/esphome/commit/302ba2874e12960c76a624fa6066bf5df1ffac2e
which is in the dev branch now.  I'm afraid this temporarily breaks using the PR as an external component with any released esphome versions.

You might be able to use the previous version at
https://github.com/esphome/esphome/commit/d5bdf2575c151abfad39b6c24c3856395a7d0901
if necessary.

Or use the esphome dev branch instead of 2024.10.2

Imanol82, 2024-11-04:
If it is a problem that you already have under control and I just have to wait, I will wait.

As it is, the ESP32 now works fine for me, it just won't let me update it.

Thanks for your help!

taHC81, 2025-03-03:
Good job, subscribing to this request ;)

closed 2025-06-30 by jesserockz