esphome/feature-requests #2428: Growatt solar sensor data expansion with supporting documentation

ewanvt, 2023-10-13:
**Describe the problem you have/What new integration you would like**

Growatt solar integration supports sensors for PV string 1 and 2 only.
I would love if we can add full support for all PV strings available under the RTU2 protocol as outlined in the following document from growatt @ page 25 under PV3. Items 1-10 are already integrated but items 11-34 are not available at the moment
https://www.photovoltaicsolar.in/Growatt_Manual/MAX%20Series%20Modbus%20RTU%20Protocol.pdf

**Please describe your use case for this integration and alternatives you've tried:**

Use case is to get access to all PV string data availabe via the modbus.
These are clearly documented in the above reference document and should hopefully be a simple addition

**Additional context**

Due to recent breaking changed to the growatt home assitant integration i believe we should see a risinge demand for modbus access to the same sensor data however the current version does not support all PV string data

Thanks to all the devs past and current for your great work.

[image]

caladrien, 2024-07-17:
Love to see that. I will do an installation of 3 inverter in the next month. One will have a pv3 as well. Id like to do the whole communication via esphome.

nagyrobi, 2024-07-17:
In ESPHome any sensor can be read with ModBUS component universally. No need for dedicated components.
https://esphome.io/components/modbus_controller
https://esphome.io/components/sensor/modbus_controller

caladrien, 2024-07-23:
Yes I know. But since this is an out of the box solution it would be nice.

I try to implement it myself but since i firstly dealing with modbus I run in some problems or unanswered question. My @nagyrobi  you can help here?

My Code:
```
uart:
  id: mod_bus
  tx_pin: GPIO3
  rx_pin: GPIO1
  baud_rate: 9600
  #baud_rate: 115200
  
modbus:
  id: modbus1
  uart_id: mod_bus
  disable_crc: True #this is only to see if I receive anything
  send_wait_time: 1s #recommended by the growatt modbus manual
  
modbus_controller:
  - id: growatt
# the Modbus device addr
    address: 0x1
    modbus_id: modbus1
    setup_priority: -10 
    update_interval: 10s
    
    

sensor:
  - platform: modbus_controller
    name: "${device_name} DcPower"
    address: 5
    register_type: "read"
    unit_of_measurement: W
    device_class: power
    icon: mdi:flash
    value_type: U_DWORD
    accuracy_decimals: 1
    internal: false
    filters:
    - multiply: 0.1
    - 
..... 

```
Compiling works and code is running. But I do not receive anything.

Log saving:
`[14:56:50][D][modbus_controller:040]: Modbus command to device=1 register=0x05 countdown=0 no response received - removed from send queue`

I guess I have a wiring problem. The TX Led is blinking 4 times in a 1s interval on my RS485 module. That means its transmitting something.

Relevent quenstion would be:
- Do I HAVE to ground everything together? Inverter, Rs485 modul and ESP
- Do I need the 120Ohm bus termination?
- Is my Cable to long? 15-20m

[image]

Documentaion is relativly clear about where to connect. 3 and 4. GND would e 15, if relevant.

The code I have from here:
https://community.simon42.com/t/esphome-modbus-growatt/2015

nagyrobi, 2024-07-23:
If only the TX is blinking, and RX not, it means your slave device does not respond.
What's the slave address of the inverter? Are you sure it's `0x1`?
Is this the only device on the cable (bus)?
Check if 120Ohm bus termination resistors are present only at the ends of the bus. Some RS485 modules have it already on board, some not. Some slave devices have it configurable with a DIP switch or a jumper, or may not have it at all. Check the user manual.
The cable shouldn't be long.

caladrien, 2024-07-23:
Thank you very much for answering.

Yes, the adress is 0x01. Although in the inverter you do it decimal. So it is 001. That can set in the inverter LCD display. It is the only device on the bus. I hope here hex and decimal correspond. 

I was hoping that i do not have to enable the bus via the config page because that means that i have to leaf china in my network, at least for a little while. I hope RS485 is enabled by default. 

I let you know how it turns out.

caladrien, 2024-07-23:
I guess my mistake was had two devices connected. I left the usb ethernet adapter from the vendor at its port. As soon as i plug it out I received data. Without your comment I never would have guessed that this might be the problem. Thank you!

nagyrobi, 2024-07-23:
Maybe you could share your full working config, make a PR to the ESPHome Devices site.