esphome/feature-requests #2429: add esp-idf media_player support

alfwro13, 2023-10-13:
**Describe the problem you have/What new integration you would like**

Please add support for media_player component on esp-idf framework.  Many esphome features now recommend using esp-idf framework (i.e. bluetooth proxy) however media player still only work on arduino framework.

**Please describe your use case for this integration and alternatives you've tried:**

Currently media player is only available on arduino framework and when trying to compile this code:
````
i2s_audio:
  - id: i2s_out
    i2s_lrclk_pin: 44
    i2s_bclk_pin: 36

media_player:
  - platform: i2s_audio
    name: Media Player
    id: media_out
    i2s_audio_id: i2s_out
    dac_type: external
    i2s_dout_pin: 35
    mode: mono
````
I get: This feature is only available with frameworks [‘arduino’].

**Additional context**

amj3brg, 2023-10-14:
same problem too.

jarodwsams, 2023-11-04:
This would be very helpful, since it would allow the possibility to use your voice assistant device (i.e. M5Stack Atom Echo) as a TTS speaker for notifications from Home Assistant. Unless there's already another way to do that...

cociweb, 2023-11-09:
> I think this would be the same issue/feature-request [Add media player to echo voice assistant](https://github.com/esphome/firmware/issues/77)

hi, not really. the tts echo is bounded by that fact where esphome not able to play wav/raw sources. ESPHome can echo only from http (not https) mp3 tts streams. piper and other recommended tts are transfer wav objects. check it on: https://github.com/home-assistant/core/issues/92969

gnumpi, 2024-02-08:
I just uploaded a a first version of a custom component (https://github.com/gnumpi/esphome_audio) which aims to provide access to the Espressif Audio Development Framework (ADF) from ESPHome. It also implements a media_player which makes use of the adf pipeline concept. So far I tested it on a ESP32-S3-DevKitC-1 and on a MatrixIo-Voice. Is anyone willing to test it on their devices?
There is still some work to do for making the component more configurable. It is definitely not in a ready to use status yet, but I would greatly appreciate any feedback you can provide on this.

gnumpi, 2024-02-09:
Thanks @amj3brg, that would be awesome. 
I managed to run it on a matrixio-voice now, which only has an esp32-wroom. 
So, I guess an esp32 wrover should be fine.

amj3brg, 2024-02-09:
hello
this weekend i will prepare de code for my wrover that use this project with wrover.
https://www.youtube.com/watch?v=McE6lsnmvBQ
i invited you to join my group on facebook to share projects.
https://www.facebook.com/groups/3416472405264998
[cid:c79bf40b-08ba-4264-8975-31b0f64e772c]
[https://www.bing.com/th?id=OVP.KTSqlY-uEY5nf11fhX1L3QEsDh&pid=Api]<https://www.youtube.com/watch?v=McE6lsnmvBQ>
Um assistente de voz para o Home Assistant com integração local/Cloud - Made in Portugal<https://www.youtube.com/watch?v=McE6lsnmvBQ>
Tal como prometido aqui fica então vídeo do assistente de voz criado pelo José Amaral made in Portugal . Se és daqueles que vai seguindo o projeto Home Assistant sabes que o ano 2023 foi o ano dedicado pela equipa desenvolvimento desta plataforma a voz ( Voice). E este projeto é então um dispositivo que explora essa mesma funcionalidade ...
www.youtube.com

formatBCE, 2024-02-23:
Does it mean that we can replace ADF speaker with this media_player for S3-BOX based voice satellite?

gnumpi, 2024-02-23:
@formatBCE, yes, that's the general idea, there are still some issues to be resolved, but I'm on it...

formatBCE, 2024-02-24:
> @formatBCE, yes, that's the general idea, there are still some issues to be resolved, but I'm on it...  

Great to hear it! Awesome news!

formatBCE, 2024-02-24:
@amj3brg there's code for Box and Box-3, which works good. But it's using adf directly - which, I believe, just a wrap on top of idf framework, and could be replaced easily. But I'm not good with idf - so I'm just carefully waiting for adf Media player to come. :)

tiimsvk, 2024-03-06:
any news? 
I use this type of display ESP32-4848S040 (mcu:ESP32-S3 display:ST7701 touch:GT911 and speaker NS4168), which also has an integrated speaker. The whole device is not officially supported in esphome because the display is ST7701 and works only on an unofficial component(https://github.com/esphome/esphome/pull/5872) under ESP-IDF, so it would be nice to integrate media_player into ESP-IDF as well.

closed 2025-02-10 by jesserockz