esphome/feature-requests #2431: BME680 BSEC support framework esp-idf

tiimsvk, 2023-10-15:
**Describe the problem you have/What new integration you would like**
is it possible to add support for this sensor via BSEC using ESP-IDF?
well thank you

```
bme680_bsec:

esp32:
  board: esp32dev
  framework:
    type: esp-idf
```

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

LuukML, 2023-10-28:
Hi, 

I would also really like to have  this one in suport! i know there is a new librery but im not capabul of making this work. I can help out if any one is willing to colaberate, i can send a pcb with the BME680 mounted to it . for now im using the sensor just for temp and basic info. 

hoope any one will join this one!

Cheers Luuk

LuukML, 2023-10-28:
I think this might help out to make it work.

https://community.bosch-sensortec.com/t5/MEMS-sensors-forum/Implementing-BSEC-2-4-0-0-on-ESP-IDF-5-0-with-ESP32-S1-and-S2/m-p/74787

joshuaboniface, 2023-12-16:
Seconding this. There seems to be a bunch of examples of compiling the bsec library against ESP-IDF, so that part should work.

MStapelfeldt, 2024-07-26:
I also hich this could be done soon. I have few devices using bme680 and Voice assist. To be able to use on device WW, I need to switch to esp-idf, but then bme680 can't be used.

KevinGaudry, 2025-02-28:
Following up :)

MaikoVoigt, 2025-03-14:
I have created a working example for the BSEC2 library based on [luar123's example](https://github.com/luar123/Bosch-BSEC2-Library/tree/master/examples/generic_examples/basic_idf) of using the library with PlatformIO. I have adapted the example so that it also works without PlatformIO.

For this, it is necessary to use customized versions of the BSEC2 and BME68x libraries with the Arduino dependencies removed and the corresponding “CMakeLists.txt” files added.

https://github.com/MaikoVoigt/ESP-IDF-Bosch-BSEC2-example

PatrikZuffa, 2025-06-18:
Following up too.

PatrikZuffa, 2025-06-18:
Maybe a hint: The BME280 works with ESP-IDF framework.

kbx81, 2025-06-19:
https://esphome.io/components/sensor/bme68x_bsec2

PatrikZuffa, 2025-06-27:
@kbx81 Thanks for link, works great!