esphome/feature-requests #2433: Added support for Home Assistant event entity

jkolo, 2023-10-16:
**Describe the problem you have/What new integration you would like**

The Event entity was introduced in Home Assistant 2023.8. ESPhome should introduce support for this type of entity

**Please describe your use case for this integration and alternatives you've tried:**

The doorbell is a good example. I am currently operating it as a binary_sensor with the PRESENCE class, and the event is dedicated to this type of device.

**Additional context**

Home Assistant documentation: 
- Integration: https://www.home-assistant.io/integrations/event/
- Developer: https://developers.home-assistant.io/docs/core/entity/event/

nagyrobi, 2023-10-16:
Thinking along with https://github.com/esphome/feature-requests/issues/2282, all components supporting interrupts, should support configuring event entities.

luca-iodice, 2024-02-01:
Would be useful also to handle multiclicks event in general.

Robbe-B, 2024-02-27:
No button events in ESP Home yet ? 
I would like to fire some events on a button press like e.g 'Single Click' 'Double Click' 'Long Click'
This could become very handy since my entire house uses ESP Home to read all buttons for the lights.
Currently I have some magic in a blueprint in order to determine the click type (single double, ....) of a binary sensor.
But since this feels rather trickery, it would be well received to have these events implemented. 
Kind regards,
Robbe

luca-iodice, 2024-02-27:
For now the only known workaround to me is to use binary sensors. Obviously you create a lot of redundant yaml code and chattery in the event bus.

Globolo001, 2024-04-09:
Yup, I completely agree that is is needed. Currently the only events that are natively sent are "pressed" and "released" for each binary sensor (as far as I understand the problem)

It would be so useful to be able to neatly send "double_clicked" and others into HA's events-list without created lots of other bloating binary sensors.

[image]

Robbe-B, 2024-04-10:
Can't believe this is actually happening => https://github.com/esphome/esphome/pull/6451
@nohat, thank you for all these efforts, I hope this can get merged soon.

michaelarnauts, 2024-05-16:
I've moved my project over from binary sensors to events, but they somehow aren't picked up in Home Assistant.

I still have my old `homeassistant.event` triggers, and I can create automations based on the state of the binary sensors, but I would assume the new events would also be listed in Home Assistant?

Am I missing something?

```yaml
substitutions:
  device_name: woonkamer
  short_min:   10ms
  short_max:  500ms
  long_min:   500ms
  long_max:  2000ms

event:
  - platform: template
    id: button_1
    name: Button 1
    device_class: button
    event_types:
      - pressed_short
      - pressed_long

binary_sensor:
  - platform: gpio
    name: "D1"
#    internal: true
    pin:
      number: D1
      inverted: true
      mode:
        input: true
        pullup: true
    on_click:
      - min_length: ${short_min}
        max_length: ${short_max}
        then:
          - event.trigger:
              id: button_1
              event_type: pressed_short
          - homeassistant.event:
              event: esphome.button_pressed
              data:
                device: ${device_name}
                button: d1
                type: short
      - min_length: ${long_min}
        max_length: ${long_max}
        then:
          - event.trigger:
              id: button_1
              event_type: pressed_long
          - homeassistant.event:
              event: esphome.button_pressed
              data:
                device: ${device_name}
                button: d1
                type: long
```

[image]

Robbe-B, 2024-05-16:
@michaelarnauts,
You can do something like this to automate on: 
[image]

Julian-1-2-3-4-5, 2026-03-15:
> Can't believe this is actually happening => [esphome/esphome#6451](https://github.com/esphome/esphome/pull/6451) [@nohat](https://github.com/nohat), thank you for all these efforts, I hope this can get merged soon.

does this help us as someone earlier mentioned with #2972 and #2282 ?