esphome/feature-requests #2443: Globals trigger

matlar83, 2023-10-25:
**Describe the problem you have/What new integration you would like**

I think a trigger on global vars changes would be useful. I have some computed vars that depend on global vars that need to be evaluated every time a global var change.
I would have a on_value trigger on global variables to execute a script that re-evaluated the computed values.

**Please describe your use case for this integration and alternatives you've tried:**

As a workaround, at least for boolean computed values, I use template binary sensors that are updated every loop. This does not seems to be very efficient and is valid only for boolean values.
For example, if I would like to have a "status" text sensor that depends on different variables, I would not be able to use the workaround.

**Additional context**

KRobert74, 2023-11-22:
I have the same wish.
I want do update template sensors if a global variable was changed.

vgomenyuk, 2024-01-16:
+1! Need to execute some script when the global variable is changed to some exact value.

GraffJosh, 2024-01-24:
+1 I like using variables to manage state.