esphome/feature-requests #2447: ST7735 - I2C support - T-Dongle S3

netmindz, 2023-10-29:
**Describe the problem you have/What new integration you would like**

There is support for the ST7735, but this is for the SPI varient. The Lilygo T-Dongle S3 uses the I2C

**Please describe your use case for this integration and alternatives you've tried:**

I've tried the existing ST7735 driver

**Additional context**

https://github.com/Xinyuan-LilyGO/T-Dongle-S3

randybb, 2023-10-29:
nope, it is SPI - SDA is MOSI, SCL is SCK. [ST7735 ](https://www.displayfuture.com/Display/datasheet/controller/ST7735.pdf) has SPI or parallel (8080, which is not supported by esphome) interface.

netmindz, 2023-10-30:
Thanks for the tip @randybb 

The full working example is

```
spi:
  clk_pin: 5
  mosi_pin: 3

display:
  - platform: st7735
    model: "INITR_MINI160X80"
    reset_pin: 1
    cs_pin: 4
    dc_pin: 2
    rotation: 90
    device_width: 80
    device_height: 160
    col_start: 0
    row_start: 0
    eight_bit_color: false
    invert_colors: true
    use_bgr: true
    update_interval: 1s
    lambda: |-
     it.printf(10, 10, id(font1),Color(255, 0, 0), "hello");
     
font:
  - file: "gfonts://Roboto"
    id: font1
    size: 30
```

closed 2023-10-30 by netmindz

netmindz, 2023-10-30:
Example with working status LED too

```
spi:
  - id: spi_display
    clk_pin: 5
    mosi_pin: 3
    interface: hardware

  - id: spi_led
    clk_pin: 39
    mosi_pin: 40
    interface: software

display:
  - platform: st7735
    spi_id: spi_display
    model: "INITR_MINI160X80"
    reset_pin: 1
    cs_pin: 4
    dc_pin: 2
    rotation: 90
    device_width: 80
    device_height: 160
    col_start: 0
    row_start: 0
    eight_bit_color: false
    invert_colors: true
    use_bgr: true
    update_interval: 1s
    lambda: |-
     it.printf(10, 10, id(font1),Color(255, 0, 0), "Hello sexy boy :)");

font:
  - file: "gfonts://Roboto"
    id: font1
    size: 20

light:
  - platform: spi_led_strip
    spi_id: spi_led
    name: "Light"
```