esphome/feature-requests #2448: USB_CDC logger support for arduino framework

Tecnologic, 2023-10-30:
**Describe the problem you have/What new integration you would like**

USB_CDC should be a valid config for hardware_uart of the logger component when framework is arduino.

For ESP-IDF this works but i would like to have it working for Arduino as well.
**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

This [https://github.com/esphome/esphome/pull/4225](https://github.com/esphome/esphome/pull/4225) Pull request seems to add this feature but is stale.

rjsachse, 2023-12-02:
I am looking for this feature too, arduino platform has more components so don't want to change framework

tomaszduda23, 2024-10-13:
It seems to be already supported https://github.com/esphome/esphome/tree/dev/tests/components/logger